meta-rust/meta-rust #251: merge meta-rust to oe-core

rwmacleod, 2019-09-17:
The rust recipes and patches in the meta-rust layer should be integrated into oe-core.
Is this a goal that developers and users of meta-rust support? 

I can start to work on the merging the code unless someone with more experience with meta-rust wants to take the lead. I'd like to have an initial version in Oct/Nov 2019 for oe-core-3.1-M1/M2. librsvg should be update to use rust as part of that work.

Perhaps this issue could be used to collect a list of problems that need to be resolved before oe-core will accept the merger. I'm not aware of all the issues but here are a coupule:

1. good use of sstate-cache, 
2. ability to work offline after initially collecting source from dependencies  
3. build from C rather than a rust bootstrap compiler (https://guix.gnu.org/blog/2018/bootstrapping-rust/ )

janderholm, 2019-11-02:
2. I believe this is already possible but you need to properly vendor your package. I.e. use cargo vendor or cargo bitbake. rustc, libstd and cargo should all use the vendored sources bundled in the downloaded rust source.
3. I don't think this is feasible. mrustc is only able to build rustc 1.29. This means to build 1.38 we would need to build rustc 11 times. This would take a hideous amount of time.

otavio, 2019-11-02:
I think it'd be more realistic if we rely on Rust official binaries to build the cross ones.

weltling, 2020-08-19:
May I ask about the progress on this intention?

Thansk.

weltling, 2020-08-27:
Thanks for the updates. Please count be as an interested party as well. Is there a branch accessible already?

Thanks.

rwmacleod, 2021-02-17:
Finally pushed a new poky-contrib to:
http://git.yoctoproject.org/cgit/cgit.cgi/poky-contrib/log/?h=rmacleod/rust-wip-2021-02-16
   9cd171b77b0..e6fb0ba0a4a  rmacleod/rust-wip-2021-02-16 -> rmacleod/rust-wip-2021-02-16

I have an update to librsvg in another branch but it still doesn't build using autotools unfortunately. I can push that to another branch if someone wants it.

rwmacleod, 2021-02-24:
The oe-core team is interested in getting rust merged. I've squash history and sent the first pull request:
https://lists.openembedded.org/g/openembedded-core/message/148563?p=,,,20,0,0,0::Created,,,20,2,0,80869121,d=2&d=2
https://lists.openembedded.org/g/openembedded-core/topic/80869129?p=,,,20,0,0,0::Created,,,20,2,20,80869129,d=2,ct=1&d=2&ct=1

rwmacleod, 2021-03-05:
Rust won't be merged for the 3.3 release of oe-core, as explained here:
https://lists.openembedded.org/g/openembedded-core/message/148921?p=,,,20,0,0,0::Created,,Merge+meta-rust+to+oe-core,20,2,0,80893251

rwmacleod, 2021-08-10:
Starting work on the merge to oe-core again. See:
http://git.yoctoproject.org/cgit/cgit.cgi/poky-contrib/log/?h=rmacleod/rust-redo-aug-10-2021-a

rwmacleod, 2021-08-23:
Richard merged the meta-rust PR to master-next: http://git.yoctoproject.org/cgit/cgit.cgi/poky/log/?h=master-next
but there are some problems that we'd love help with still so that we get this into oe-core/master for this release.

We've fixed a few problems but there are still potential blockers for merge:

1.  bitbake rust-native on an older system (ubun-16.04, debian-8?) with no sstate present and uninative enabled https://autobuilder.yoctoproject.org/typhoon/#/builders/116/builds/581/steps/13/logs/stdio    This builds for me on ubu-18.04.3 so I'll have to try a docker image.

Richard said, "If I don't merge it now, I will not be merging it until some of these issues get fixed though, it won't just 'merge after we branch'"

Running more tests on the autobuilder.

We could also use help updating librsvg to work with epiphany and gstreamer.
Please come talk on #yocto or send an email to the list. -- Randy

weltling, 2021-08-25:
That's a great news, thanks for the update!

It also looks like there are some patches flew upstream on top of the initial merge, should those be probably backported over here, too? Another question probably to be discussed separately for the `meta-rust` roadmap, as it sure still makes sense to have it as a discord solution for a flexible Rust version handling, etc. But, there will be probably some conflicting namespaces or other stuff to care about so some approach might need to be figured out.

Thanks

rwmacleod, 2021-08-25:
Let's see what happens this week as far as merging to oe-core. Last I heard,  someone from the AWS dev team had offered to help debug the intermittent problems that we are seeing.

rwmacleod, 2021-11-24:
Rust is in oe-core and working well:
https://git.openembedded.org/openembedded-core/log/meta/recipes-devtools/rust

I have a PR ( https://github.com/meta-rust/meta-rust/pull/364 ) to deal with removing meta-rust content that is in oe-core but people like to use meta-rust/master with older yocto releases it seems. I'll keep that PR open for a while and hope that new branches are created to allow meta-rust to be useful to people who are on dunfell or other pre-honister branches. ( See https://wiki.yoctoproject.org/wiki/Releases for release names to numbers translation. )

closed 2021-11-24 by rwmacleod