meta-rust/meta-rust #254: Issues with 128 bit integers targeting ARM

markubiak, 2019-10-25:
## Version(s) of meta-rust
5cda04c7c2158fc982502ca1e7893f71e4ca1729

## Version(s) of poky and/or oe-core
warrior
poky: 1153a954e652304b6b5d287437817b7da891d491
oe-core: 3bdbf72e3a4bf18a4a2c7afbde4f7ab773aeded9

## Expected result
A working binary is generated for Librespot

## Actual result
Compilation fails on the num-bigint crate, relating to 128 bit integers in both places

## Steps to reproduce
Attempt to build librespot with attached .bb file targeting any ARM chip (tried A9 for DE1-SoC and A72 for Pi 4)

## Steps to resolve
I have resolved this before by manually editing num-bigint's source to remove any possibility of generating the '--cfg has_i128' flag seen at the end of the quoted error string. I was able to successfully get past this a month or so ago (I do have an old working binary), but was an idiot and forgot to document. I believe it was editing build.rs to remove everything related to i128 and changing the default feature from std to i128. I did just try this with no success, but I know that this was the resolution to the issue a month ago.

> |   process didn't exit successfully: `rustc --crate-name num_bigint /home/quartus/yocto/pi/build/tmp/work/aarch64-poky-linux/librespot/0.1.0.AUTOINC+551daadc44-r0/cargo_home/registry/src/github.com-1ecc6299db9ec823/num-bigint-0.2.2/src/lib.rs --color never --crate-type lib --emit=dep-info,metadata,link -C opt-level=3 --cfg 'feature="default"' --cfg 'feature="std"' -C metadata=e077619a5dbe2bb0 -C extra-filename=-e077619a5dbe2bb0 --out-dir /home/quartus/yocto/pi/build/tmp/work/aarch64-poky-linux/librespot/0.1.0.AUTOINC+551daadc44-r0/git/target/aarch64-poky-linux/release/deps --target aarch64-poky-linux -C linker=/home/quartus/yocto/pi/build/tmp/work/aarch64-poky-linux/librespot/0.1.0.AUTOINC+551daadc44-r0/wrapper/target-rust-ccld -L dependency=/home/quartus/yocto/pi/build/tmp/work/aarch64-poky-linux/librespot/0.1.0.AUTOINC+551daadc44-r0/git/target/aarch64-poky-linux/release/deps -L dependency=/home/quartus/yocto/pi/build/tmp/work/aarch64-poky-linux/librespot/0.1.0.AUTOINC+551daadc44-r0/git/target/release/deps --extern num_integer=/home/quartus/yocto/pi/build/tmp/work/aarch64-poky-linux/librespot/0.1.0.AUTOINC+551daadc44-r0/git/target/aarch64-poky-linux/release/deps/libnum_integer-fe330d5a5e9edd7a.rlib --extern num_traits=/home/quartus/yocto/pi/build/tmp/work/aarch64-poky-linux/librespot/0.1.0.AUTOINC+551daadc44-r0/git/target/aarch64-poky-linux/release/deps/libnum_traits-9dae7c3036eccd2e.rlib --cap-lints allow -L /home/quartus/yocto/pi/build/tmp/work/aarch64-poky-linux/librespot/0.1.0.AUTOINC+551daadc44-r0/recipe-sysroot/usr/lib/rust --remap-path-prefix=/home/quartus/yocto/pi/build/tmp/work/aarch64-poky-linux/librespot/0.1.0.AUTOINC+551daadc44-r0=/usr/src/debug/librespot/0.1.0.AUTOINC+551daadc44-r0 --cfg has_i128`

Normally I'd say this is num-bigint's fault for generating 128 bit integers on an embedded device -- however, outside of openembedded the binary for librespot buils just fine.

[librespot_0.1.0.zip](https://github.com/meta-rust/meta-rust/files/3773091/librespot_0.1.0.zip)

wdehoog, 2019-10-27:
I had the same problem. You need to update the autocfg crate. See https://users.rust-lang.org/t/error-when-cross-building-num-bigint/33814. I did it like this: https://github.com/wdehoog/meta-sfos-librespot/commit/61125da3b5eaab9b2984875552826deda9b36783

markubiak, 2019-11-16:
I apologise for the delay, but that was the fix! So the main issue is Librespot shipping with an old version of num-bigint?

closed 2019-11-16 by markubiak