meta-rust/meta-rust #256: Cannot build for armv7

MastaG, 2019-11-27:
## Version(s) of meta-rust
master
## Version(s) of poky and/or oe-core
master of everything
## Expected result
working rust compiler
## Actual result
process didn't exit successfully: `rustc --edition=2018 --crate-name core src/libcore/lib.rs --color never --crate-type lib --emit=dep-info,metadata,link -C opt-level=2 -C metadata=ff967dfcd444300c -C extra-filename=-ff967dfcd444300c --out-dir /home/mastag/src/odroid-oe-core/build/tmp/work/cortexa15t2hf-neon-vfpv4-oe-linux-gnueabi/libstd-rs/1.37.0-r0/rustc-1.37.0-src/src/libstd/arm-oe-linux-gnueabi/release/deps --target arm-oe-linux-gnueabi -C linker=/home/mastag/src/odroid-oe-core/build/tmp/work/cortexa15t2hf-neon-vfpv4-oe-linux-gnueabi/libstd-rs/1.37.0-r0/wrapper/target-rust-ccld -L dependency=/home/mastag/src/odroid-oe-core/build/tmp/work/cortexa15t2hf-neon-vfpv4-oe-linux-gnueabi/libstd-rs/1.37.0-r0/rustc-1.37.0-src/src/libstd/arm-oe-linux-gnueabi/release/deps -L dependency=/home/mastag/src/odroid-oe-core/build/tmp/work/cortexa15t2hf-neon-vfpv4-oe-linux-gnueabi/libstd-rs/1.37.0-r0/rustc-1.37.0-src/src/libstd/release/deps -L /home/mastag/src/odroid-oe-core/build/tmp/work/cortexa15t2hf-neon-vfpv4-oe-linux-gnueabi/libstd-rs/1.37.0-r0/recipe-sysroot/usr/lib/rust --remap-path-prefix=/home/mastag/src/odroid-oe-core/build/tmp/work/cortexa15t2hf-neon-vfpv4-oe-linux-gnueabi/libstd-rs/1.37.0-r0=/usr/src/debug/libstd-rs/1.37.0-r0 -L /home/mastag/src/odroid-oe-core/build/tmp/work/cortexa15t2hf-neon-vfpv4-oe-linux-gnueabi/libstd-rs/1.37.0-r0/recipe-sysroot/usr/lib -C link-arg=-Wl,-soname,libstd.so` (signal: 11, SIGSEGV: invalid memory reference)
## Steps to reproduce
Trying to build the firefox browser from OSSystems/meta-browser which requires meta-rust.
I'm building on a Fedora 31 x86_64 host for arm 32bit target.
Chromium builds fine but it doesn't require meta-rust.

cmeissl, 2019-12-03:
You can create an bbappend for rust and rust cross with:

```
# we add back the armv7a tune here which has been removed in meta-openenembedded
# the tune itself is not needed, but the rust.inc file uses the TUNE_FEATURES
# value for parsing compile parameters. Can be removed if fixed in meta-rust
TUNE_FEATURES_append = " armv7a"
```

MastaG, 2019-12-09:
Thanks it worked :+1: 
Perhaps you could add this by default so I won't have to bbappend it myself?

janderholm, 2019-12-12:
In my opinion the target generation needs to rewritten entirely since it doesn't work very well. It should be possible to map -mcpu and or -march to the proper CPU and features rather than depend on TUNE_FEATURES which tend to change.

cmeissl, 2020-02-14:
Issues should be resolved since https://github.com/meta-rust/meta-rust/pull/242 has been merged.

osenft, 2021-05-06:
This is again / still happening on the most recent commit (02db606a9678bddf81169a3f2e7c6346e0876839).

wltu, 2022-04-25:
can confirm that this is not working. :(

jmlemetayer, 2024-03-18:
I have got a similar issue (similar errors) which was due to the way I added the layer to the build.

As poky has already integrated a version of the `meta-rust` some files are overlapping.

The recipe files (`.bb`) seem ok because of the `BBFILE_PRIORITY` mechanism between the layers.

But classes (`.bbclass`) seems to be picked according to the order of the layers in the `BBLAYERS` variable (`bblayers.conf`).

Putting `meta-rust` before `poky` fix the issue:

```
BBLAYERS ?= " \
  ##OEROOT##/meta-rust \  <- first
  ##OEROOT##/meta \
  ##OEROOT##/meta-poky \
  ##OEROOT##/meta-yocto-bsp \
  ...
  "
```