meta-rust/meta-rust #258: Adding meta-rust layer causes bitbake-layers to fail

GeoffClements, 2020-01-04:
## Version(s) of meta-rust

$ git show
commit 0f950f5e333a1c8999320bf18232144f3dd9c80e (HEAD -> master, origin/master, origin/HEAD)
Merge: 10fb698 71895ec
Author: Steven Walter <stevenrwalter@gmail.com>
Date:   Sun Dec 15 18:23:50 2019 -0500

    Merge pull request #255 from rettichschnidi/update-to-rust-1.39.0
    
    Update to Rust 1.39.0

## Version(s) of poky and/or oe-core

zeus

## Expected result

Expected a list of layers

## Actual result

Got a python traceback see below

## Steps to reproduce

$ bitbake-layers show-layers
NOTE: Starting bitbake server...
layer                 path                                      priority
==========================================================================
meta                  /home/geoff/rpi/poky/meta                 5
meta-poky             /home/geoff/rpi/poky/meta-poky            5
meta-yocto-bsp        /home/geoff/rpi/poky/meta-yocto-bsp       5
meta-raspberrypi      /home/geoff/rpi/poky/meta-raspberrypi     9

$ bitbake-layers add-layer ../meta-rust
NOTE: Starting bitbake server...

$ bitbake-layers show-layers
NOTE: Starting bitbake server...
Traceback (most recent call last):
  File "/home/geoff/rpi/poky/bitbake/bin/bitbake-layers", line 93, in <module>
    ret = main()
  File "/home/geoff/rpi/poky/bitbake/bin/bitbake-layers", line 61, in main
    tinfoil.prepare(True)
  File "/home/geoff/rpi/poky/bitbake/lib/bb/tinfoil.py", line 408, in prepare
    self.run_command('parseConfiguration')
  File "/home/geoff/rpi/poky/bitbake/lib/bb/tinfoil.py", line 466, in run_command
    raise TinfoilCommandFailed(result[1])
bb.tinfoil.TinfoilCommandFailed: Traceback (most recent call last):
  File "/home/geoff/rpi/poky/bitbake/lib/bb/command.py", line 74, in runCommand
    result = command_method(self, commandline)
  File "/home/geoff/rpi/poky/bitbake/lib/bb/command.py", line 275, in parseConfiguration
    command.cooker.parseConfiguration()
  File "/home/geoff/rpi/poky/bitbake/lib/bb/cooker.py", line 437, in parseConfiguration
    self.handleCollections(self.data.getVar("BBFILE_COLLECTIONS"))
  File "/home/geoff/rpi/poky/bitbake/lib/bb/cooker.py", line 1229, in handleCollections
    raise CollectionError("Errors during parsing layer configuration")
bb.cooker.CollectionError: Errors during parsing layer configuration

$ cat conf/bblayers.conf 
# POKY_BBLAYERS_CONF_VERSION is increased each time build/conf/bblayers.conf
# changes incompatibly
POKY_BBLAYERS_CONF_VERSION = "2"

BBPATH = "${TOPDIR}"
BBFILES ?= ""

BBLAYERS ?= " \
  /home/geoff/rpi/poky/meta \
  /home/geoff/rpi/poky/meta-poky \
  /home/geoff/rpi/poky/meta-yocto-bsp \
  /home/geoff/rpi/poky/meta-raspberrypi \
  /home/geoff/rpi/poky/meta-rust \
  "

GeoffClements, 2020-01-05:
A little bit more information (sorry, I'm quite new to OE so I'm probably doing something stupid).

```
$ bitbake -p core-image-minimal
ERROR: Layer 'rust-layer' depends on layer 'openembedded-layer', but this layer is not enabled in your configuration

Summary: There was 1 ERROR message shown, returning a non-zero exit code.
```

The meta layer provides openembedded-core but what provides openembedded-layer?

bachp, 2020-01-05:
@GeoffClements Regarding your second issue. You need to add the corresponging layer from [meta-openembedded](https://github.com/openembedded/meta-openembedded) to your `BBLAYERS`.

Specifically the `openembedded-layer` refers to [meta-oe](https://github.com/openembedded/meta-openembedded/tree/master/meta-oe).

GeoffClements, 2020-01-05:
@bachp Thanks, that solved both problems, so closing.

I'm still getting used to the OE ecosystem and meta-openembedded seemed to cover similar ground to the meta layer from the yocto project and I was getting a little, confused.

closed 2020-01-05 by GeoffClements

bachp, 2020-01-05:
meta-openembedded is a single repo containing multiple layers. It contains additional packages that are not in OE-core (meta). There are also many more layers, for a list have a look at https://layers.openembedded.org/.

ptinstituteblr, 2020-07-25:
i got same error and solved with below methods

1) all layers should be on same branch. like meta-raspberripi , meta-openembedded , meta-poky should be on same zeus branch.
we can check we are on which branch by git branch.
we can mover to branch by git checkout zeus.

2) if you already got error their then open bblayers.conf and delete last added layers before adding new layers again (as suggested above).