meta-rust/meta-rust #261: Documentation meta-rust external rust-bin and nightly support

IniterWorker, 2020-01-30:
Hello meta-rust maintainers,

I have some questions about meta-rust usage.

With the current documentation, I don't know if it's possible to have a nightly build to use some nightly features of Rust and if it's possible to reduce the compile time with a pre-built binary of Rust.

- I didn't see anything correctly clear about nightly support or nightly integration example, like do something like that to have a nightly build. 
- I didn't see a simple example of the use of the external Rust bins.

Could you give me a hint?

posborne, 2020-01-30:
@IniterWorker meta-rust-bin is a separate layer that makes use of the prebuilt compiler and sysroots: https://github.com/rust-embedded/meta-rust-bin.  This has some advantages and limitations and only targets stable releases or rustc/cargo.

This layer certainly aligns more with what is generally considered to be the proper way to do things within Yocto and can make better use of caching, allows for sharing of std as a shared library, etc.  It does require the extra work of ensuring that your recipe and Cargo.toml are kept in sync (modeling crate deps within Yocto) by use of https://github.com/cardoe/cargo-bitbake

If you need to use nightly compiler features then meta-rust is probably the better base for making that work as it provides much greater control but this would be built from source (Yocto has its own very capable mechanisms that would allow for build reuse using downloads/sstate caches).  Depending on what you need, this might be able to be achieved by bbappending a release version in your layer (though I haven't tried this).

Making available nightly (pinned or otherwise) in meta-rust is likely possible, although it presents a serious amount of work from a maintenance perspective.  I believe most people using either of these layers are shipping product and, as such, are targeting stable versions of rust.

Hopefully this helps.

aswild, 2021-04-05:
@posborne, would you consider adding a note in meta-rust's readme linking to meta-rust-bin as an alternative? I wasn't aware of meta-rust-bin's existence until I saw this issue today, and it's a faster alternative for some use-cases.

I'm actually quite excited, I've been contemplating a way to use "rustup" prebuilts in Yocto rather than wait for llvm and rustc to bootstrap and now I don't have to do that work myself :)