meta-rust/meta-rust #264: Unable to build updated librsvg-2.46.4 recipe

rwmacleod, 2020-02-14:
Using meta-rust: d8d77be  Merge pull request #259 from ColinFinck/update-to-rust-1.40.0

With oe-core:
$ git log -2 --oneline
   69cc9c2405 (HEAD -> master) librsvg: Update from 2.40.20 to 2.46.4
   44a4ac2294 (origin/master, origin/HEAD, rust-wip) systemd: upgrade 243.4 -> 244.1
with the extra commit pasted below.

I expect to be able to build librsvg, just as I can build rust-hello-world or other packages such as ripgrep which I added elsewhere.

I see:

| && cd /buildarea/rmacleod/src/distro/yocto/b/librsvg/tmp-glibc/work/core2-64-oe-linux/librsvg/2.46.4-r0/build && ./x86_64-oe-linux-libtool  --tag=CC
   --mode=link x86_64-oe-linux-gcc  -m64 -march=core2 -mtune=core2 -msse3 -mfpmath=sse --sysroot=/buildarea/rmacleod/src/distro/yocto/b/librsvg/tmp-gl
ibc/work/core2-64-oe-linux/librsvg/2.46.4-r0/recipe-sysroot  -O2 -pipe -g -feliminate-unused-debug-types -fmacro-prefix-map=/buildarea/rmacleod/src/di
stro/yocto/b/librsvg/tmp-glibc/work/core2-64-oe-linux/librsvg/2.46.4-r0=/usr/src/debug/librsvg/2.46.4-r0                      -fdebug-prefix-map=/buil
darea/rmacleod/src/distro/yocto/b/librsvg/tmp-glibc/work/core2-64-oe-linux/librsvg/2.46.4-r0=/usr/src/debug/librsvg/2.46.4-r0                      -fd
ebug-prefix-map=/buildarea/rmacleod/src/distro/yocto/b/librsvg/tmp-glibc/work/core2-64-oe-linux/librsvg/2.46.4-r0/recipe-sysroot=                     
 -fdebug-prefix-map=/buildarea/rmacleod/src/distro/yocto/b/librsvg/tmp-glibc/work/core2-64-oe-linux/librsvg/2.46.4-r0/recipe-sysroot-native=   -Wl,-O1
 -Wl,--hash-style=gnu -Wl,--as-needed -o librsvg_c_api.la librsvg/dummy.lo && mv /buildarea/rmacleod/src/distro/yocto/b/librsvg/tmp-glibc/work/core2-6
4-oe-linux/librsvg/2.46.4-r0/build/target/x86_64-oe-linux-gnu/release/librsvg_c_api.a .libs/librsvg_c_api.a                                           
| error: failed to run `rustc` to learn about target-specific information                                                                             
|                                                                                                                                                     
| Caused by:                                                                                                                                          
|   process didn't exit successfully: `rustc - --crate-name ___ --print=file-names --crate-type bin --crate-type rlib --crate-type dylib --crate-type 
cdylib --crate-type staticlib --crate-type proc-macro --print=sysroot --print=cfg` (exit code: 1)                                                     
| --- stderr                                                                                                                                          
| error: Error loading host specification: Could not find specification for target "x86_64-linux"                                                     
|                                                                                                                                                     
|                                                                                                                                                     
| make[2]: *** [Makefile:1751: librsvg_c_api.la] Error 101   

To reproduce, build oe-core+meta-openembeded/meta-oe+meta-rust, git am <attached commit>
and bitbake librsvg

librsvg update commit
--------------------
From c48deb8b016a923d07c2a905835975bd002ecee7 Mon Sep 17 00:00:00 2001
From: Randy MacLeod <Randy.MacLeod@windriver.com>
Date: Mon, 10 Feb 2020 11:13:58 -0500
Subject: [PATCH] librsvg: Update from 2.40.20 to 2.46.4

The new version is written in Rust.
Remove the patches that were backported and drop the reproducibility
patch since that code was dropped in the Rust translation.

Update the license checksums since the license info was updated
to be more consistent in commit:
   bd8ebc43 Use the LGPL 2.1 consistently
      Fixes these:
      https://gitlab.gnome.org/GNOME/librsvg/issues/72
      https://gitlab.gnome.org/GNOME/librsvg/issues/432

Also inherit gettext to address:
   QA Issue: AM_GNU_GETTEXT used but no inherit gettext
and sort the inherit list for good measure.

---
 ...t-Bsymbolic-fixes-configure-on-macOS.patch | 35 -----------
 .../0001-Remove-non-reproducible-SRCDIR.patch | 30 ----------
 .../librsvg/librsvg/gtk-option.patch          | 60 -------------------
 .../{librsvg_2.40.20.bb => librsvg_2.46.4.bb} | 22 ++++---
 4 files changed, 10 insertions(+), 137 deletions(-)
 delete mode 100644 meta/recipes-gnome/librsvg/librsvg/0001-Auto-detect-Bsymbolic-fixes-configure-on-macOS.patch
 delete mode 100644 meta/recipes-gnome/librsvg/librsvg/0001-Remove-non-reproducible-SRCDIR.patch
 delete mode 100644 meta/recipes-gnome/librsvg/librsvg/gtk-option.patch
 rename meta/recipes-gnome/librsvg/{librsvg_2.40.20.bb => librsvg_2.46.4.bb} (60%)

diff --git a/meta/recipes-gnome/librsvg/librsvg/0001-Auto-detect-Bsymbolic-fixes-configure-on-macOS.patch b/meta/recipes-gnome/librsvg/librsvg/0001-Auto-detect-Bsymbolic-fixes-configure-on-macOS.patch
deleted file mode 100644
[... 169 lines cut ...]
-            file://0001-Remove-non-reproducible-SRCDIR.patch \
-"
+SRC_URI[archive.md5sum] = "aca4001b267de77924b786886a1ad64a"
+SRC_URI[archive.sha256sum] = "b45b9ee3b64c58baaf800bcdff5fcd04d79930dba4c56e46e0d3b0aead40cc29"
+SRC_URI[archive.sha1sum] = "7ad4b036296a48d3cb043fd1a23644ddf1b90269"
+SRC_URI[archive.sha384sum] = "0587784fe1b757983df1067cc18e6f7a37654cfae33bf4c5a4e249f22376dd66a756019dc2ba5761359cef782aa8a206"
+SRC_URI[archive.sha512sum] = "abb4d9eeef62b778092269446987f3c8205cd260c253d5e70a9e3c562b9b23a64c76562d7bb53eadc9c12cca8a1cb52009f9b9d78695cc6970fe7e14b9ba10a0"
 
-SRC_URI[archive.md5sum] = "4949d313b0c5d9161a5c259104af5568"
-SRC_URI[archive.sha256sum] = "cff4dd3c3b78bfe99d8fcfad3b8ba1eee3289a0823c0e118d78106be6b84c92b"
+CARGO_DISABLE_BITBAKE_VENDORING = "1"
 
 CACHED_CONFIGUREVARS = "ac_cv_path_GDK_PIXBUF_QUERYLOADERS=${STAGING_LIBDIR_NATIVE}/gdk-pixbuf-2.0/gdk-pixbuf-query-loaders"
 
@@ -32,7 +30,7 @@ PACKAGECONFIG ??= "gdkpixbuf"
 # The gdk-pixbuf loader
 PACKAGECONFIG[gdkpixbuf] = "--enable-pixbuf-loader,--disable-pixbuf-loader,gdk-pixbuf-native"
 # GTK+ test application (rsvg-view)
-PACKAGECONFIG[gtk] = "--with-gtk3,--without-gtk3,gtk+3"
+# PACKAGECONFIG[gtk] = "--with-gtk3,--without-gtk3,gtk+3"
 
 do_install_append() {
 	# Loadable modules don't need .a or .la on Linux
-- 
2.17.1

rwmacleod, 2020-02-16:
`From c48deb8b016a923d07c2a905835975bd002ecee7 Mon Sep 17 00:00:00 2001
From: Randy MacLeod <Randy.MacLeod@windriver.com>
Date: Mon, 10 Feb 2020 11:13:58 -0500
Subject: [PATCH] librsvg: Update from 2.40.20 to 2.46.4

The new version is written in Rust.
Remove the patches that were backported and drop the reproducibility
patch since that code was dropped in the Rust translation.

Update the license checksums since the license info was updated
to be more consistent in commit:
   bd8ebc43 Use the LGPL 2.1 consistently
      Fixes these:
      https://gitlab.gnome.org/GNOME/librsvg/issues/72
      https://gitlab.gnome.org/GNOME/librsvg/issues/432

Also inherit gettext to address:
   QA Issue: AM_GNU_GETTEXT used but no inherit gettext
and sort the inherit list for good measure.

---
 ...t-Bsymbolic-fixes-configure-on-macOS.patch | 35 -----------
 .../0001-Remove-non-reproducible-SRCDIR.patch | 30 ----------
 .../librsvg/librsvg/gtk-option.patch          | 60 -------------------
 .../{librsvg_2.40.20.bb => librsvg_2.46.4.bb} | 22 ++++---
 4 files changed, 10 insertions(+), 137 deletions(-)
 delete mode 100644 meta/recipes-gnome/librsvg/librsvg/0001-Auto-detect-Bsymbolic-fixes-configure-on-macOS.patch
 delete mode 100644 meta/recipes-gnome/librsvg/librsvg/0001-Remove-non-reproducible-SRCDIR.patch
 delete mode 100644 meta/recipes-gnome/librsvg/librsvg/gtk-option.patch
 rename meta/recipes-gnome/librsvg/{librsvg_2.40.20.bb => librsvg_2.46.4.bb} (60%)

diff --git a/meta/recipes-gnome/librsvg/librsvg/0001-Auto-detect-Bsymbolic-fixes-configure-on-macOS.patch b/meta/recipes-gnome/librsvg/librsvg/0001-Auto-detect-Bsymbolic-fixes-configure-on-macOS.patch
deleted file mode 100644
index 954bb60880..0000000000
--- a/meta/recipes-gnome/librsvg/librsvg/0001-Auto-detect-Bsymbolic-fixes-configure-on-macOS.patch
+++ /dev/null
@@ -1,35 +0,0 @@
-From b99891e31eb6ce550e7e1cb2ca592095b3050a93 Mon Sep 17 00:00:00 2001
-From: Brion Vibber <brion@pobox.com>
-Date: Sun, 25 Feb 2018 18:42:36 -0800
-Subject: Auto-detect -Bsymbolic, fixes configure on macOS
-
-The -Bsymbolic linker option is ELF-specific, and was breaking
-configure on macOS unless --disable-Bsymbolic was explicitly passed.
-
-Switching the behavior from requiring -Bsymbolic to be available
-by default to just warning and continuing on without.
-
-Fixes https://gitlab.gnome.org/GNOME/librsvg/issues/211
-
-Upstream-Status: Backport
-Signed-off-by: Adrian Bunk <bunk@stusta.de>
----
- configure.ac | 2 +-
- 1 file changed, 1 insertion(+), 1 deletion(-)
-
-diff --git a/configure.ac b/configure.ac
-index 15b26b2d..9f8dce29 100644
---- a/configure.ac
-+++ b/configure.ac
-@@ -216,7 +216,7 @@ AM_CONDITIONAL([ENABLE_PIXBUF_LOADER],[test "$enable_pixbuf_loader" = "yes"])
- AC_ARG_ENABLE([Bsymbolic],
-   [AS_HELP_STRING([--disable-Bsymbolic],
-                   [disable linking with -Bsymbolic])],
--  [],[enable_Bsymbolic=yes])
-+  [enable_Bsymbolic=no],[enable_Bsymbolic=auto])
- 
- BSYMBOLIC_LDFLAG=
- if test "$enable_Bsymbolic" != "no"; then
--- 
[... 134 lines cut ...]
+SRC_URI[archive.md5sum] = "aca4001b267de77924b786886a1ad64a"
+SRC_URI[archive.sha256sum] = "b45b9ee3b64c58baaf800bcdff5fcd04d79930dba4c56e46e0d3b0aead40cc29"
+SRC_URI[archive.sha1sum] = "7ad4b036296a48d3cb043fd1a23644ddf1b90269"
+SRC_URI[archive.sha384sum] = "0587784fe1b757983df1067cc18e6f7a37654cfae33bf4c5a4e249f22376dd66a756019dc2ba5761359cef782aa8a206"
+SRC_URI[archive.sha512sum] = "abb4d9eeef62b778092269446987f3c8205cd260c253d5e70a9e3c562b9b23a64c76562d7bb53eadc9c12cca8a1cb52009f9b9d78695cc6970fe7e14b9ba10a0"
 
-SRC_URI[archive.md5sum] = "4949d313b0c5d9161a5c259104af5568"
-SRC_URI[archive.sha256sum] = "cff4dd3c3b78bfe99d8fcfad3b8ba1eee3289a0823c0e118d78106be6b84c92b"
+CARGO_DISABLE_BITBAKE_VENDORING = "1"
 
 CACHED_CONFIGUREVARS = "ac_cv_path_GDK_PIXBUF_QUERYLOADERS=${STAGING_LIBDIR_NATIVE}/gdk-pixbuf-2.0/gdk-pixbuf-query-loaders"
 
@@ -32,7 +30,7 @@ PACKAGECONFIG ??= "gdkpixbuf"
 # The gdk-pixbuf loader
 PACKAGECONFIG[gdkpixbuf] = "--enable-pixbuf-loader,--disable-pixbuf-loader,gdk-pixbuf-native"
 # GTK+ test application (rsvg-view)
-PACKAGECONFIG[gtk] = "--with-gtk3,--without-gtk3,gtk+3"
+# PACKAGECONFIG[gtk] = "--with-gtk3,--without-gtk3,gtk+3"
 
 do_install_append() {
 	# Loadable modules don't need .a or .la on Linux
-- 
2.17.1

`

rwmacleod, 2020-02-16:
Ah, I saw that to build on Ubuntu-19.10, you need to:
$ GDK_PIXBUF_QUERYLOADERS=/usr/lib/x86_64-linux-gnu/gdk-pixbuf-2.0/gdk-pixbuf-query-loaders ./configure
so at least now I have a way to compare the yocto build to a working self-hosted configuration that works.

rwmacleod, 2020-02-26:
Busy with other work but, their is clearly a compiler configuration problem:
$ /ala-lpggp31/rmacleod/src/distro/yocto/b/rust-jan6/tmp-glibc\
> /work/core2-64-oe-linux/rust-hello-world/git-r0/recipe-sysroot-native/usr/bin/rustc --print cfg
error: Error loading host specification: Could not find specification for target "x86_64-linux"
$
I did spend some time trying to get rust-gdb goign to debug rustc. That will take some time given my limited knowledge of rust and rustc code.

srwalter, 2020-05-12:
I've only ever used this layer with poky.  Can you provide further details about what repos you're using and their revisions?  Step-by-step would be great

rwmacleod, 2020-05-15:
The problem has a work-around or maybe it's even a fix.
-inherit cargo gettext gnomebase gobject-introspection gtk-doc
pixbufcache upstream-version-is-even
+inherit gettext gnomebase gobject-introspection gtk-doc pixbufcache
upstream-version-is-even 
...
+ inherit cargo

This switches the librsvg configuration from autotools to cargo.

https://lists.openembedded.org/g/openembedded-core/message/137183?p=,,,20,0,0,0::Created,,librsvg,20,2,0,72806932

I'm updating the librsvg recipe to a newer version and adding more recent meta-rust changes to my merge-meta-rust-to-oe-core branch. 

Once I'm happy with the outcome, I'll close this issue.

srwalter, 2020-06-20:
@rwmacleod any update on this?

rwmacleod, 2021-02-05:
Sorry for the long delay.
I have a work-around that could be a fix if I could find the time to work on it.
At the shell doing:
```
$export RUST_TARGET_PATH="/.../tmp-glibc/work/core2-64-unknown-linux/librsvg/2.46.4-r0/recipe-sysroot-native/usr/lib/rustlib"

 $ ./tmp-glibc/work/core2-64-unknown-linux/librsvg/2.46.4-r0/recipe-sysroot-native/usr/bin/rustc --print cfg
debug_assertions
target_arch="x86_64"
...
```
so we need something like: https://github.com/meta-rust/meta-rust/commit/c83864ce19659c2f8e6a64c39ea4c28beda91bd5
but for when we are using rustc via autotools.

rwmacleod, 2021-11-24:
Fixed by: http://cgit.openembedded.org/openembedded-core/commit/meta/recipes-gnome/librsvg?id=e060ee5ea96f3c3903b8aa018a5a3af3c60e5ab3

closed 2021-11-24 by rwmacleod