meta-rust/meta-rust #265: Building libraries and executables that aren't published to crates.io

jdegges, 2020-03-09:
Is there a method to build rust libraries and executables that aren't yet published to crates.io with Bitbake?

For example, the source tree could look like:
```
/code/
|-- some-rust-lib-a
|-- some-rust-lib-b
|   |-- Cargo.toml:
|       [dependencies]
|       some-rust-lib-a = {path = "../some-rust-lib-a"}
|-- some-rust-bin
|   |-- Cargo.toml:
|       [dependencies]
|       some-rust-lib-b = {path = "../some-rust-lib-b"}
```

And the BitBake recipe tree could look something like:
```
/meta-project/recipes-project/
some_rust_lib_a_0.1.0.bb
some_rust_lib_b_0.1.0.bb
some_rust_bin_0.1.0.bb
```

Is there a meta-rust BitBake recipe syntax that would allow these projects to depend on each other correctly?

jmesmon, 2020-03-09:
Some background on meta-rust that might be informative here:

 - When initially using meta-rust, I didn't do cargo builds the "oe" way: I just inherited `cargo` and did no special fetching. This is (while not advisable), probably still functional. It will break some OE rules though, so I'd recommend avoiding it
 - We (for some time) had a number of crates included as separate bitbake packages/recipies. These were removed in  55e9c18a3775504798d44d267ac9fce38317bd0d. These _theoretically_ followed the model you're looking for, but there are probably some issues with that approach in the current code today. Most notably, from a rust perspective, it didn't allow linking to 2 different versions of a crate (which is common).
 - The current model uses (essentially) a "vendoring"-like approach: every package specifies all the _sources_ it needs, bitbake downloads these, and then cargo is configured to use that source of package sources. I'd recommend pursuing this method, though I don't think we have an example of using a non-crates.io package. From the OE side, you might represent intermediate crates as includes (for example, providing variables with URLs, etc), and only have a recipe for the top-level crate.

srwalter, 2020-05-12:
That pattern we use for this internally is to have a git superproject with submodules.  That way the bitbake git fetcher checks out all the code we need that's not on crates.io.  Then in our Cargo.toml we specify the dependencies with relative paths...

IniterWorker, 2022-10-20:
Our current problem seems related to [cargo paths-overrides policy](https://doc.rust-lang.org/cargo/reference/overriding-dependencies.html#paths-overrides).

> Note: using a local configuration to override paths will only work for crates that have been published to [crates.io](https://crates.io/). You cannot use this feature to tell Cargo how to find local unpublished crates.

I think we should be able to override packages that are not published (e.g private).

Anyone has an advise? Or should we go to the cargo team?