meta-rust/meta-rust #269: x86_64: can't find crate for `proc_macro`` when compiling proc-macro2

russell-islam, 2020-03-27:
## Version(s) of meta-rust
646fecee8afb8e878da28d4cff8ca617e162aec0
## Version(s) of poky and/or oe-core
8167800fc4cba6ac525ef4a13ea9353f5a591fa3
## Expected result
Able to build cargo
## Actual result
failed to build cargo
## Steps to reproduce
bitbake cargo

Detailed error:
Running `rustc --crate-name memchr /home/muislam/work/linux-yocto/image/tmp/work/core2-64-msft-linux/cargo/1.41.0-r0/rustc-1.41.0-src/vendor/memchr/src/lib.rs --error-format=json --json=diagnostic-rendered-ansi,artifacts --crate-type lib --emit=dep-info,metadata,link -C opt-level=2 --cfg 'feature="default"' --cfg 'feature="use_std"' -C metadata=a2ce0451de3001cb -C extra-filename=-a2ce0451de3001cb --out-dir /home/muislam/work/linux-yocto/image/tmp/work/core2-64-msft-linux/cargo/1.41.0-r0/rustc-1.41.0-src/target/x86_64-msft-linux/release/deps --target x86_64-msft-linux -C linker=/home/muislam/work/linux-yocto/image/tmp/work/core2-64-msft-linux/cargo/1.41.0-r0/wrapper/target-rust-ccld -L dependency=/home/muislam/work/linux-yocto/image/tmp/work/core2-64-msft-linux/cargo/1.41.0-r0/rustc-1.41.0-src/target/x86_64-msft-linux/release/deps -L dependency=/home/muislam/work/linux-yocto/image/tmp/work/core2-64-msft-linux/cargo/1.41.0-r0/rustc-1.41.0-src/target/release/deps --cap-lints allow -L /home/muislam/work/linux-yocto/image/tmp/work/core2-64-msft-linux/cargo/1.41.0-r0/recipe-sysroot/usr/lib/rust --remap-path-prefix=/home/muislam/work/linux-yocto/image/tmp/work/core2-64-msft-linux/cargo/1.41.0-r0=/usr/src/debug/cargo/1.41.0-r0 --cfg memchr_runtime_simd --cfg memchr_runtime_sse2 --cfg memchr_runtime_sse42 --cfg memchr_runtime_avx`
   Compiling crossbeam-channel v0.3.8
error[E0463]: can't find crate for `proc_macro`
  --> /usr/src/debug/cargo/1.41.0-r0/rustc-1.41.0-src/vendor/proc-macro2-0.4.30/src/lib.rs:87:1

closed 2020-04-07 by russell-islam

Asalle, 2021-02-19:
@russell-islam did you find a fix for this? I'm having the same issue

russell-islam, 2021-02-19:
> @russell-islam did you find a fix for this? I'm having the same issue

Cargo can be used to build rust project but not standalone. So I did not try to build cargo standalone.