meta-rust/meta-rust #271: Failed to build a package using a recipie generated by cargo-bitbake

russell-islam, 2020-04-09:
## Version(s) of meta-rust
646fecee8afb8e878da28d4cff8ca617e162aec0
## Version(s) of poky and/or oe-core
8167800fc4cba6ac525ef4a13ea9353f5a591fa3
## Expected result
Able to build the rust package
## Actual result
failed to build
## Steps to reproduce
I am using cargo-bitbake to create a recipe for the project cloud-hypervisor (https://github.com/cloud-hypervisor/cloud-hypervisor/)
After starting the build using meta-rust layer it got failed with the following error.

error: failed to select a version for `vmm`.
|     ... required by package `cloud-hypervisor v0.4.0 (/home/muislam/work/linux-yocto/image-x86-64/tmp/work/core2-64-msft-linux/cloud-hypervisor/git.AUTOINC+7cb61d3960-r0/git)`
| versions that meet the requirements `= 0.1.0` are: 0.1.0
|
| the package `cloud-hypervisor` depends on `vmm`, with features: `cmos, pci_support, acpi` but `vmm` does not have these features.

Generated recipe:

`
inherit cargo

SRC_URI += "git://github.com/cloud-hypervisor/cloud-hypervisor.git;protocol=https;nobranch=1"
SRCREV = "7cb61d396032aa982e974c36d205aecdd43d67a9"
S = "${WORKDIR}/git"
CARGO_SRC_DIR = ""
PV_append = ".AUTOINC+7cb61d3960"

SRC_URI += " \
    crate://crates.io/aho-corasick/0.6.10 \
    crate://crates.io/ansi_term/0.11.0 \
    crate://crates.io/anyhow/1.0.26 \
    crate://crates.io/arc-swap/0.4.4 \
    crate://crates.io/arrayref/0.3.6 \
    crate://crates.io/arrayvec/0.5.1 \
    crate://crates.io/atty/0.2.14 \
    crate://crates.io/autocfg/0.1.7 \
    crate://crates.io/autocfg/1.0.0 \
    crate://crates.io/backtrace-sys/0.1.32 \
    crate://crates.io/backtrace/0.3.43 \
    crate://crates.io/base64/0.11.0 \
    crate://crates.io/bitflags/0.5.0 \
    crate://crates.io/bitflags/1.2.1 \
    crate://crates.io/blake2b_simd/0.5.10 \
    crate://crates.io/byteorder/1.3.2 \
    crate://crates.io/c2-chacha/0.2.3 \
    crate://crates.io/cast/0.2.3 \
    crate://crates.io/cc/1.0.50 \
    crate://crates.io/cfg-if/0.1.10 \
    crate://crates.io/clap/2.33.0 \
    crate://crates.io/cloudabi/0.0.3 \
    crate://crates.io/constant_time_eq/0.1.5 \
    crate://crates.io/credibility/0.1.3 \
    crate://crates.io/crossbeam-utils/0.7.0 \
    crate://crates.io/dirs-sys/0.3.4 \
    crate://crates.io/dirs/2.0.2 \
    crate://crates.io/epoll/4.1.0 \
    crate://crates.io/failure/0.1.6 \
    crate://crates.io/failure_derive/0.1.6 \
    crate://crates.io/fuchsia-cprng/0.1.1 \
    crate://crates.io/getrandom/0.1.14 \
    crate://crates.io/glob/0.2.11 \
    crate://crates.io/hermit-abi/0.1.6 \
    crate://crates.io/ipnetwork/0.15.1 \
    crate://crates.io/itoa/0.4.5 \
    crate://crates.io/kernel32-sys/0.2.2 \
    crate://crates.io/kvm-bindings/0.2.0 \
    crate://crates.io/kvm-ioctls/0.5.0 \
    crate://crates.io/lazy_static/1.4.0 \
    crate://crates.io/libc/0.2.66 \
    crate://crates.io/libssh2-sys/0.2.14 \
    crate://crates.io/libz-sys/1.0.25 \
    crate://crates.io/lock_api/0.3.3 \
    crate://crates.io/log/0.3.9 \
    crate://crates.io/log/0.4.10 \
    crate://crates.io/memchr/2.3.0 \
    crate://crates.io/openssl-sys/0.9.54 \
    crate://crates.io/parking_lot/0.10.0 \
    crate://crates.io/parking_lot_core/0.7.0 \
    crate://crates.io/pkg-config/0.3.17 \
    crate://crates.io/pnet/0.25.0 \
    crate://crates.io/pnet_base/0.22.0 \
    crate://crates.io/pnet_datalink/0.25.0 \
    crate://crates.io/pnet_macros/0.25.0 \
    crate://crates.io/pnet_macros_support/0.25.0 \
    crate://crates.io/pnet_packet/0.25.0 \
    crate://crates.io/pnet_sys/0.25.0 \
    crate://crates.io/pnet_transport/0.25.0 \
    crate://crates.io/ppv-lite86/0.2.6 \
    crate://crates.io/proc-macro2/1.0.8 \
    crate://crates.io/quote/1.0.2 \
    crate://crates.io/rand/0.4.6 \
    crate://crates.io/rand/0.7.3 \
    crate://crates.io/rand_chacha/0.2.1 \
    crate://crates.io/rand_core/0.3.1 \
    crate://crates.io/rand_core/0.4.2 \
    crate://crates.io/rand_core/0.5.1 \
    crate://crates.io/rand_hc/0.2.0 \
    crate://crates.io/rdrand/0.4.0 \
    crate://crates.io/redox_syscall/0.1.56 \
    crate://crates.io/redox_users/0.3.4 \
    crate://crates.io/regex-syntax/0.6.14 \
    crate://crates.io/regex/1.0.6 \
    crate://crates.io/remain/0.2.1 \
    crate://crates.io/remove_dir_all/0.5.2 \
    crate://crates.io/rust-argon2/0.7.0 \
    crate://crates.io/rustc-demangle/0.1.16 \
    crate://crates.io/rustc-serialize/0.3.24 \
    crate://crates.io/rustc_version/0.2.3 \
    crate://crates.io/ryu/1.0.2 \
    crate://crates.io/scopeguard/1.0.0 \
    crate://crates.io/semver-parser/0.7.0 \
    crate://crates.io/semver/0.9.0 \
    crate://crates.io/serde/1.0.104 \
    crate://crates.io/serde_derive/1.0.104 \
    crate://crates.io/serde_json/1.0.46 \
    crate://crates.io/signal-hook-registry/1.2.0 \
    crate://crates.io/signal-hook/0.1.13 \
    crate://crates.io/smallvec/1.2.0 \
    crate://crates.io/ssh2/0.7.1 \
    crate://crates.io/strsim/0.8.0 \
    crate://crates.io/syn/1.0.14 \
    crate://crates.io/synstructure/0.12.3 \
    crate://crates.io/syntex/0.42.2 \
    crate://crates.io/syntex_errors/0.42.0 \
    crate://crates.io/syntex_pos/0.42.0 \
    crate://crates.io/syntex_syntax/0.42.0 \
    crate://crates.io/tempdir/0.3.7 \
    crate://crates.io/tempfile/3.1.0 \
    crate://crates.io/term/0.4.6 \
    crate://crates.io/textwrap/0.11.0 \
    crate://crates.io/thiserror-impl/1.0.10 \
    crate://crates.io/thiserror/1.0.10 \
    crate://crates.io/thread_local/0.3.6 \
    crate://crates.io/unicode-width/0.1.7 \
    crate://crates.io/unicode-xid/0.0.3 \
    crate://crates.io/unicode-xid/0.2.0 \
    crate://crates.io/utf8-ranges/1.0.4 \
    crate://crates.io/vcpkg/0.2.8 \
    crate://crates.io/vec_map/0.8.1 \
    crate://crates.io/virtio-bindings/0.1.0 \
    crate://crates.io/vmm-sys-util/0.4.0 \
    crate://crates.io/wasi/0.9.0+wasi-snapshot-preview1 \
    crate://crates.io/winapi-build/0.1.1 \
    crate://crates.io/winapi-i686-pc-windows-gnu/0.4.0 \
    crate://crates.io/winapi-x86_64-pc-windows-gnu/0.4.0 \
    crate://crates.io/winapi/0.2.8 \
    crate://crates.io/winapi/0.3.7 \
    crate://crates.io/ws2_32-sys/0.2.1 \
    git://github.com/firecracker-microvm/firecracker;protocol=https;nobranch=1;name=micro_http;destsuffix=micro_http \
    git://github.com/rust-vmm/linux-loader;protocol=https;nobranch=1;name=linux-loader;destsuffix=linux-loader \
    git://github.com/rust-vmm/vfio-bindings;protocol=https;nobranch=1;name=vfio-bindings;destsuffix=vfio-bindings \
    git://github.com/rust-vmm/virtio-bindings;protocol=https;nobranch=1;name=virtio-bindings;destsuffix=virtio-bindings \
    git://github.com/rust-vmm/vm-memory;protocol=https;nobranch=1;name=vm-memory;destsuffix=vm-memory \
"

SRCREV_FORMAT .= "_linux-loader"
SRCREV_linux-loader = "${AUTOREV}"
EXTRA_OECARGO_PATHS += "${WORKDIR}/linux-loader"
SRCREV_FORMAT .= "_micro_http"
SRCREV_micro_http = "${AUTOREV}"
EXTRA_OECARGO_PATHS += "${WORKDIR}/micro_http"
SRCREV_FORMAT .= "_vfio-bindings"
SRCREV_vfio-bindings = "${AUTOREV}"
EXTRA_OECARGO_PATHS += "${WORKDIR}/vfio-bindings"
SRCREV_FORMAT .= "_virtio-bindings"
SRCREV_virtio-bindings = "${AUTOREV}"
EXTRA_OECARGO_PATHS += "${WORKDIR}/virtio-bindings"
SRCREV_FORMAT .= "_vm-memory"
SRCREV_vm-memory = "${AUTOREV}"
EXTRA_OECARGO_PATHS += "${WORKDIR}/vm-memory"

LIC_FILES_CHKSUM = " \
    "

SUMMARY = "DDD"
HOMEPAGE = "https://github.com/cloud-hypervisor/cloud-hypervisor"
LICENSE = "CLOSED"

include cloud-hypervisor-${PV}.inc
include cloud-hypervisor.inc
`

jmesmon, 2020-09-02:
This looks like it might be something related to our handling of git deps in combination with `crate` deps. What repo or package is the `vmm` crate being provided by?

russell-islam, 2020-09-10:
> This looks like it might be something related to our handling of git deps in combination with `crate` deps. What repo or package is the `vmm` crate being provided by?

vmm crate is a carte inside cloud-hypervisor.

russell-islam, 2022-02-09:
Mainly patching a crate with git repo does not work. There are some dependency issues if the patch crated is used in any other dependent crates.