meta-rust/meta-rust #272: "bitbake cargo" does a git checkout of my recipes!

remleduff, 2020-04-15:
## Version(s) of meta-rust
(master) 646fecee8afb8e878da28d4cff8ca617e162aec0

## Version(s) of poky and/or oe-core
poky (master): 8702b79a3b90a06524098e8b2a10b6e6d287166a
oe-core (master): e66990ec08c4cbbb2abb42eaac9465e38d3b502b

## Expected result
"cargo bitbake" should build successfully.
However, if it fails, it should not have any effect on the checked out versions of my layers.

## Actual result
After running "bitbake cargo" (which currently fails), many of my layers have been checked out to the version which is committed to the super-project. What the heck, since when does a bitbake recipe do anything at all to the git projects of the metadata?!!

## Steps to reproduce
My project is a git repo that uses git submodules for each of my layers, ie:

project.
.
├── build
├── meta-browser
├── meta-clang
├── meta-freescale
├── meta-freescale-3rdparty
├── meta-nrl
├── meta-openembedded
├── meta-python2
├── meta-qt5
├── meta-rust
└── poky

(some layers omitted)

Most of my layers are currently checked out to the "master" branch which is a tracking branch for origin/master for each of the layers.

After "bitbake cargo" fails, several layers, including poky are no longer on the master branch. For instance, meta-freescale is checked out to revision "4af1824f3daa13bd2bceafe8de7f371b35e0dfce" in a "detached head" state, where before I ran the command it was checked out to "master". 4af18 is the version of meta-freescale that is currently recorded in my superproject, please note that the only command I have run is "bitbake cargo". I haven't run a single git command which should affect the checked out version of meta-freescale.

"git reflog" in the affected layers shows that a checkout was executed, for instance in meta-freescale: "4af1824f (HEAD) HEAD@{0}: checkout: moving from master to 4af1824f3daa13bd2bceafe8de7f371b35e0dfce"

I haven't tracked down what is doing this, but would love a suggestion or help. Is cargo using git behind the scenes, and if so, why is it recursing up to my project outside of the build directory?

--Aaron

jmesmon, 2020-09-02:
cargo does use git itself, but it shouldn't be checking out layers unless we're doing something that is confusing it about fetching from a git repo.

One area that might have some affect here is that we re-direct the location of the registry (normally a subdir in `$HOME/.cargo`) into the build tree. Plausible that something is confusing cargo about which git repo it should be updated.

chezgi, 2020-09-03:
does it related to rust-lang/rust#63476 ?

chezgi, 2020-09-03:
also i have some errors in building vector timberio/vector#3511

robbawebba, 2021-01-27:
Hey everyone, my team and I have been experiencing the same issue since we introduced meta-rust and a few custom rust tools into our a couple of our images. We have a similar hierarchy as described by @remleduff: a "super-repository" that contains our required metadata layers as submodules. During day-to-day development, developers may have checked out different revisions of one or more layers. At some point during a clean build of our images, all submodule revisions are reset to the revs commited to the super-repository, as if someone ran `git submodule update` in the super-repository.

I've tracked this down to the `cargo-native.do_compile` task, specifically the step where `libgit2_sys` is compiled. Some relevant logs from cargo-native's `log.do_compile` file:
```
Running `rustc --crate-name libgit2_sys --edition=2018 /home/rob/os/build/tmp/work/x86_64-linux/cargo-native/1.43.0-r0/rustc-1.43.0-src/vendor/libgit2-sys/lib.rs --error-format=json --json=diagnostic-rendered-ansi,artifacts --crate-type lib --emit=dep-info,metadata,link -C opt-level=3 --cfg 'feature="https"' --cfg 'feature="libssh2-sys"' --cfg 'feature="openssl-sys"' --cfg 'feature="ssh"' --cfg 'feature="ssh_key_from_memory"' -C metadata=d93524d054b5bbf8 -C extra-filename=-d93524d054b5bbf8 --out-dir /home/rob/os/build/tmp/work/x86_64-linux/cargo-native/1.43.0-r0/rustc-1.43.0-src/target/x86_64-linux/release/deps --target x86_64-linux -C linker=/home/rob/os/build/tmp/work/x86_64-linux/cargo-native/1.43.0-r0/wrapper/target-rust-ccld -L dependency=/home/rob/os/build/tmp/work/x86_64-linux/cargo-native/1.43.0-r0/rustc-1.43.0-src/target/x86_64-linux/release/deps -L dependency=/home/rob/os/build/tmp/work/x86_64-linux/cargo-native/1.43.0-r0/rustc-1.43.0-src/target/release/deps --extern libc=/home/rob/os/build/tmp/work/x86_64-linux/cargo-native/1.43.0-r0/rustc-1.43.0-src/target/x86_64-linux/release/deps/liblibc-32ef73506be1535c.rmeta --extern libssh2_sys=/home/rob/os/build/tmp/work/x86_64-linux/cargo-native/1.43.0-r0/rustc-1.43.0-src/target/x86_64-linux/release/deps/liblibssh2_sys-1e585fd7ae90666a.rmeta --extern libz_sys=/home/rob/os/build/tmp/work/x86_64-linux/cargo-native/1.43.0-r0/rustc-1.43.0-src/target/x86_64-linux/release/deps/liblibz_sys-441e650bb1514da8.rmeta --extern openssl_sys=/home/rob/os/build/tmp/work/x86_64-linux/cargo-native/1.43.0-r0/rustc-1.43.0-src/target/x86_64-linux/release/deps/libopenssl_sys-bfc71e294f8288a9.rmeta --cap-lints allow -L /home/rob/os/build/tmp/work/x86_64-linux/cargo-native/1.43.0-r0/recipe-sysroot-native/usr/lib/rust --remap-path-prefix=/home/rob/os/build/tmp/work/x86_64-linux/cargo-native/1.43.0-r0=/usr/src/debug/cargo-native/1.43.0-r0 -L native=/home/rob/os/build/tmp/work/x86_64-linux/cargo-native/1.43.0-r0/rustc-1.43.0-src/target/x86_64-linux/release/build/libgit2-sys-cefecd86ca4edadf/out/build -l static=git2 -L native=/home/rob/os/build/tmp/work/x86_64-linux/cargo-native/1.43.0-r0/recipe-sysroot-native/usr/lib/pkgconfig/../../../usr/lib -L native=/home/rob/os/build/tmp/work/x86_64-linux/cargo-native/1.43.0-r0/recipe-sysroot-native/usr/lib/pkgconfig/../../../usr/lib -L native=/home/rob/os/build/tmp/work/x86_64-linux/cargo-native/1.43.0-r0/recipe-sysroot-native/usr/lib/pkgconfig/../../../usr/lib`
```

upon closer inspection of the `libgit2-sys` source code in my tmp directory, I noticed that `tmp/work/x86_64-linux/cargo-native/1.43.0-r0/rustc-1.43.0-src/vendor/libgit2-sys/build.rs` contains the following build step:
```
    if !Path::new("libgit2/.git").exists() {
        let _ = Command::new("git")
            .args(&["submodule", "update", "--init"])
            .status();
    }
```

This confirmed my suspicion that `git submodule update` was ran during the build. I took a look at the Cargo.toml file in this same directory, and noticed that rust 1.43.0 is using version `0.12.0+0.99.0` of libgit2-sys. Next, I checked the libgit2-sys github repository to see if this issue has been reported there, and it looks like this issue has been resolved with in some more recent releases with [this commit](https://github.com/rust-lang/git2-rs/commit/627fd88f9b9f82716523d6f6ddf07b8f058eae0d).

I have not yet figured out which rust release includes this version of libgit2-sys, and whether it is available in meta-rust, but that would be my next step.

My team is using the Yocto Thud release for all metadata layers. Our meta-rust layer is at [this commit](https://github.com/meta-rust/meta-rust/commit/f56fd4ec6835d68c4e9f8a9630963acd34d172e5) from November 14th, 2020.

A couple of solutions come to mind, and I'm not sure which are possible: 

1. Use a more recent rust version that includes the proper upstream fix in the libgit2-sys crate (if it's included in meta-rust already)
2. Patch libgit2-sys in my custom metadata layer with the proper upstream fix

Any suggestions on how to move forward? In the meantime, I'll keep exploring the recent rust releases, hopefully it's already available in meta-rust :smile: Thanks in advance for your feedback!

robbawebba, 2021-01-27:
After some further investigation into Cargo's releases and the versions of its dependencies, it looks like the earliest stable release in which the fixed versions of libgit2-sys and git2 crates are available is v1.48.0, which is not currently included in meta-rust.

In the PR for introducing rust v1.47.0 (#284), the author @danc86 mentioned that there's some additional work required to add v1.48.0 to meta-rust.

If rust v1.48.0 is not going to be available for some time, then I might have to consider patching Cargo v1.47.0 to update the versions of git2 and libgit-sys to the version in which the needed fix was released.

robbawebba, 2021-01-28:
Hello! I want to share the workaround I created for my team's use case. hopefully it may be useful to some other people experiencing this issue until rust v1.48 is available. I updated our revision of meta-rust to include support for rust v1.47.0. I then created this patch for the cargo recipe to update the versions of git2 and libgit2-sys crates:
```
---
 src/tools/cargo/Cargo.toml | 4 ++--
 1 file changed, 2 insertions(+), 2 deletions(-)

diff --git a/src/tools/cargo/Cargo.toml b/src/tools/cargo/Cargo.toml
index 56211532f..838df4337 100644
--- a/src/tools/cargo/Cargo.toml
+++ b/src/tools/cargo/Cargo.toml
@@ -32,7 +32,7 @@ pretty_env_logger = { version = "0.4", optional = true }
 anyhow = "1.0"
 filetime = "0.2.9"
 flate2 = { version = "1.0.3", default-features = false, features = ["zlib"] }
-git2 = "0.13.5"
+git2 = "0.13.8"
 git2-curl = "0.14.0"
 glob = "0.3.0"
 hex = "0.4"
@@ -44,7 +44,7 @@ jobserver = "0.1.21"
 lazycell = "1.2.0"
 libc = "0.2"
 log = "0.4.6"
-libgit2-sys = "0.12.7"
+libgit2-sys = "0.12.9"
 memchr = "2.1.3"
 num_cpus = "1.0"
 opener = "0.4"
```

Lastly, I included this patch only for the `cargo-native` package using the following bbappend file:
```
FILESEXTRAPATHS_native_prepend := "${THISDIR}/${PN}-native-${PV}:"

SRC_URI_native += "file://0001-tools-cargo-update-git2-and-libgit2-sys-crate-versio.patch"
```

This should then build cargo for the host system with newer versions of git2 and libgit2-sys. Best of luck! I hope this helps :slightly_smiling_face:

jmesmon, 2021-02-09:
meta-rust currently has support for rust v1.49, which should allow avoiding this issue. We do still have rust v1.47 in tree (no v1.48 at the moment), so if one selects an older version, this issue will still appear.

Options:
 - patch our v1.47
 - add v1.48 and remove v1.47
 - wait until we add v1.50 and then drop v1.47 (v1.50 is being released this Thursday, 2021-02-11)

kraj, 2021-02-09:
I vote for waiting until 1.50

jmesmon, 2021-04-27:
We currently have v1.49 and v1.51 in master (and dropped v1.47), so this shouldn't happen anymore.

release branches probably still need need fixes. If you're using release branches, feel free to PR fixes to them (or poke `@meta-rust/maintainers` to create a release branch from a commit hash via an issue).

closed 2021-04-27 by jmesmon