meta-rust/meta-rust #282: libstd-rs build fails

cmcqueen, 2020-12-15:
## Version(s) of meta-rust

This problem occurs with f56fd4ec6835d68c4e9f8a9630963acd34d172e5.

This problem doesn't occur with 318fb46863c0df5001f937bb4ca856491a493725.

Is this problem something to do with 8a44baed440d5f41e4f2eb316b8baa8ddb615084?

## Version(s) of poky and/or oe-core

poky dunfell, eg 861cfcd52f0b769772c3726530210f2f43c3449b (but exact version doesn't seem crucial)

## Expected result

Build should succeed.

## Actual result

```
| DEBUG: Executing shell function do_compile
| NOTE: cargo = /home/craigm/yocto/poky/build/tmp/work/aarch64-poky-linux/libstd-rs/1.46.0-r0/recipe-sysroot-native/usr/bin/cargo
| NOTE: rustc =
| NOTE: cargo build -v --target aarch64-poky-linux --release --features 'panic-unwind backtrace'
| cargo: error while loading shared libraries: libgit2.so.28: cannot open shared object file: No such file or directory
| WARNING: exit code 127 from a shell command.
| ERROR: Execution of '/home/craigm/yocto/poky/build/tmp/work/aarch64-poky-linux/libstd-rs/1.46.0-r0/temp/run.do_compile.58560' failed with exit code 127:
| cargo: error while loading shared libraries: libgit2.so.28: cannot open shared object file: No such file or directory
| WARNING: exit code 127 from a shell command.
```

## Steps to reproduce

1. Yocto build environment with poky dunfell, meta-rust.
2. `bitbake rust-hello-world`

srwalter, 2020-12-15:
I wasn't able to reproduce this with an older version of dunfell (4e931b1d05018923dc145cd97f6f965f5cb6e1a5), let me try with the hash you provided...

srwalter, 2020-12-15:
I also can't reproduce with poky 861cfcd52f0b769772c3726530210f2f43c3449b and meta-rust f56fd4ec6835d68c4e9f8a9630963acd34d172e5

cmcqueen, 2020-12-16:
Hmm. I've got no ideas at this point.

cmcqueen, 2020-12-20:
Yesterday I made changes to my `readline` bbappend, which triggered a rebuild of the rust packages (I was using meta-rust 318fb46863c0df5001f937bb4ca856491a493725). I got the same sort of error building `libstd-rs` again, which surprised me, because as I said above, I didn't think that meta-rust commit had this issue.

I tried doing

    bitbake libstd-rs -c clean
    bitbake libstd-rs

And then it succeeded in building. I'm not sure why this is happening, and why cleaning libstd-rs fixed it.

cmcqueen, 2020-12-20:
Now I've tried doing a build with meta-rust f56fd4ec6835d68c4e9f8a9630963acd34d172e5, and this time it has succeeded. I didn't even have to do another `bitbake libstd-rs -c clean`.

I'm not sure what combination of circumstances led to the build error with `libstd-rs`, but the previous `bitbake libstd-rs -c clean` seems to have fixed it somehow.

cmcqueen, 2020-12-21:
I've had the same build error for `libstd-rs` when I've pushed Yocto build to a TeamCity build agent.

I tried logging in to the build agent and doing `bitbake libstd-rs -c clean` directly, but unfortunately, the build still failed. Likewise for `bitbake libstd-rs -c cleansstate`.

So now I'm back to square one.

cmcqueen, 2020-12-21:
I reverted `meta-rust` back to f56fd4ec6835d68c4e9f8a9630963acd34d172e5, and pushed that change to the TeamCity build agent. The subsequent build still failed with the same error for `libstd-rs` and `libgit2.so.28`. I logged in and did `bitbake libstd-rs -c clean` manually on the build agent, then triggered another build. That build succeeded.

cmcqueen, 2020-12-22:
I'm happy to do any experiments or provide any data that would help to diagnose this.

cmcqueen, 2020-12-28:
Perhaps you could reproduce the issue yourself by doing `bitbake libstd-rs -c clean` and then trying to rebuild it.

srwalter, 2020-12-28:
I did try that, no difference.  I'm building on Arch, so probably quite a bit newer than 18.04, however we also use this project at work where we build on 18.04, and we have not had any issues with spurious build failures.

cmcqueen, 2021-02-25:
I updated my Yocto project from `meta-rust` 318fb46863 to 029f1ea6a6. It seems to build fine so far.

cmcqueen, 2021-03-30:
I've had no further issues with building `meta-rust` for a while now. I guess I'll close this.

closed 2021-03-30 by cmcqueen