meta-rust/meta-rust #292: update to rustc 1.48/1.49

danc86, 2021-02-03:
I'm filing this issue, rather than a PR, because I have been trying to build 1.48 (later 1.49) for quite a while now but have been getting stuck. Wanted to keep some notes about my progress in case others are trying the same, or have ideas about how to get it building.

Since rustc 1.48 the bootstrapping behaviour seems to have changed substantially. I can no longer convince it to cross-bootstrap from `x86_64-unknown-linux-gnu` to `x86_64-linux`. It builds stage1 for the build triple, not the host, which seems like a bug so I have filed it: rust-lang/rust#81702

There is a CHANGELOG.md for bootstrap: https://github.com/rust-lang/rust/blob/1.49.0/src/bootstrap/CHANGELOG.md but I didn't find it helpful in understanding what the new behaviour is. It seems the two relevant commits were:
https://github.com/rust-lang/rust/commit/f7dcfcd45bd019acf8c914c204ccae519c420adc
https://github.com/rust-lang/rust/commit/3193d52a21576b8c93f00132c784cce1159f733e
although I haven't understood it well enough to spot if there is a mistake there.

Separately to that, there was also a regression in 1.49 handling of custom targets. I spent quite a while stuck on that, until I eventually found this fix which is on master already:
https://github.com/rust-lang/rust/commit/dd682cb48c8b667859dded98a4bbfbd891a1eca4
It was causing miscompilation because the `libc` crate comes out empty if `#[cfg(unix)]` is not defined.

ColinFinck, 2021-02-03:
Thanks for opening this report! We have indeed been doing duplicate work, as I have also been trying to upgrade meta-rust to 1.49.0 lately and stumbled upon the very same problems: https://github.com/ColinFinck/meta-rust/commits/update-to-rust-1.49.0

danc86, 2021-02-04:
Seems like it works by just using `x.py install` as suggested on the rust issue. PR is #293, I tested building `rust-hello-world` and `rust-hello-world-native`.

I still worry that it is building more compilers than it needs to... the `do_compile` log seems to indicate that it first builds a stage1 compiler for `x86_64-unknown-linux-gnu` and then a stage2 compiler for both `x86_64-unknown-linux-gnu` and `x86_64-linux`. I guess if it assumes that it cannot execute binaries built for `x86_64-linux` then it needs the stage2 compiler for `x86_64-unknown-linux-gnu` in order to build libstd for `x86_64-linux` ... but then what is the point of the stage1 compiler?

PR #293 (2021-02-08): add rust 1.49.0
Now it seems `./x.py install` is the only way to assemble a fully
working stage 2 cross-built compiler. See rust-lang/rust#81702.

Fixes #292.
closed 2021-02-08 by jmesmon