meta-rust/meta-rust #298: Old versions of rust-llvm no longer build, consider dropping

jmesmon, 2021-02-07:
rust-llvm-native 1.37.0 (the last version that is NCSA licensed) currently fails to build due to requiring the RISCV target be enabled via `LLVM_EXPERIMENTAL_TARGETS_TO_BUILD`. This was noted as we've recently fiddled a bit with the LICENSE handling for the NCSA versions (see #297 #296 #289).

@floion (or others): do folks still have a use case that requires the older rustc versions?

jmesmon, 2021-02-08:
We additionally seem to have broken a larger swath of old rust versions with the addition of embed-bitcode to libstd. rust 1.43 fails it's build due to this. (1.46 works still)

janderholm, 2021-02-08:
I remember reading somewhere, thought it was in the README but it isn't there, that only the latest and the version before that is necessary. I think that's a reasonable compromise. In my opinion if someone really need an older version they could fix a specific commit of meta-rust or use a branch.

ColinFinck, 2021-02-08:
I also see no reason to keep versions older than 1.48.0 as soon as https://github.com/meta-rust/meta-rust/pull/293 has been merged.

otavio, 2021-02-08:
I agree; in fact, this keeps likely unused and untested versions around. If someone needs an older release, the Git history can be used.

shr-project, 2021-02-08:
Agreed to remove older version

> I think that's a reasonable compromise. In my opinion if someone really need an older version they could fix a specific commit of meta-rust or use a branch.

I have seen this used by projects which use meta-rust (e.g. firefox in meta-browser, iotedge-* in meta-iotedge). rust_versions.inc makes it a lot easier to stay on compatible version of rust while using newer meta-rust (without the need to define many PREFERRED_VERSION_*), but still leaving only 1 or 2 most used (and tested) versions is definitely better than current 10 versions.

FWIW: current meta-iotedge recipes are compatible only with 1.41.0 rust:
https://github.com/Azure/meta-iotedge/pull/49/commits/4c0ca9f72bbff424780b292290af92b4bca0827b
I've added an work around to make it compatible with 1.47.0 (and possibly newer):
https://github.com/Azure/meta-iotedge/pull/49/commits/96bdc911f2607d60ce16d2a6143b7203e388afaf
but meta-iotedge isn't well maintained, so might take a while to get this PR merged.

Firefox from meta-browser was last tested by me with 1.39.0:
https://github.com/OSSystems/meta-browser/commit/999d36220c4a3b18084e7e13c3b9cccc1588c572
but the ESR version there is quite old and anyone serious about firefox should probably start by upgrading it anyway.

jmesmon, 2021-02-09:
Closing as we've dropped the old versions.

closed 2021-02-09 by jmesmon

floion, 2021-02-17:
@jmesmon sorry for the late reply, no, we don't have a need for older versions
Thanks