meta-rust/meta-rust #336: Yocto Project Virtual Summit Talk

srwalter, 2021-04-08:
@meta-rust/maintainers 

(Sorry for (ab)using an issue like this, but this is the most reasonable way I could think of to contact all the maintainers)

The list of topics for the Yocto Project Virtual Summit (https://pretalx.com/yocto-project-summit-2021/cfp) include "Future languages (e.g. Rust)"  I was curious if one of us had already submitted a proposal to talk about meta-rust.  If not, I gave a presentation to the Rust Belt Rust conference in 2019 (https://youtu.be/EoV94cg_Tug) that could be adapted.

Thoughts?

otavio, 2021-04-08:
Cool! I hope you guys do :-)

kraj, 2021-04-08:
I think its a fantastic ideas. Please submit a CFP Steven

weltling, 2021-04-09:
Second that! Regarding Rust itself - safety but still low level is considered really a big advantage. There are reasons to utilize `meta-rust`, even though Rust is aimed to be integrated into the core. Low level programming languages are of a high priority in general and for embedded systems in particular. It will be in support of Rust and `meta-rust` showing how tradeoffs of `C` are solvable.

One suggestion maybe - an up-to-date comparison to golang would make sense. While being already in the core, golang is in many ways different especially wrt ownership.

In any case, showing presence there is a great thing. Please let me know if any help needed.

/cc @rwmacleod 

Thanks

jmesmon, 2021-04-09:
I haven't submitted anything. I've also listened to the rust belt presentation and enjoyed it. The other recommendations here for focus seem reasonable.

rwmacleod, 2021-04-10:
Steven, it would be really good if you could submit a presentation. I could give you a slide or two about the merge to oe-core. Hopefully that will be behind us by the time of your presentation. Randy.

srwalter, 2021-04-10:
Comparison relative to Go is a good idea, I think.  Problem is I'm not very familiar with Go :-)  Do you have some talking points I could use, or point me at some articles perhaps?

weltling, 2021-04-10:
Same for me, actually :) I think publicly available materials can be used to create a slide. Perhaps also, `C`, `Rust` or `Go` would be even more expedient. For the points, sure `C` is the most popular systems programming tool, but not memory safe. Both `Rust` and `Go` are memory safe, while `Go` has garbage collection. I had a good read [here](https://bitfieldconsulting.com/golang/rust-vs-go) but there's for sure more.

It might be more effort to compare `Go` and `Rust` on the `Yocto` side. Especially when it concerns the SDK support and flexibility, there are some points `Rust` might need to catch up. However, having crates to be pre fetched and vendored works well for the offline build using `meta-rust`. There might be more to go through there.

Thanks.

rwmacleod, 2021-04-23:
Steven ( @srwalter ) did you submit something or shall we pull a submission together tomorrow?

srwalter, 2021-04-23:
I have not yet submitted my proposal, but I do intend to.  I will also accept any assistance anyone is willing to provide :smile:

kraj, 2021-04-24:
@srwalter outline could be some thing on these lines...

This session discusses using rust programming language using Yocto project. we would cover how to enable building rust cross compiler infrastructure using yocto project. At the end of the session participant will have information on building a rust based application using yocto project. Rust is on the rise and we will also cover what are advantages of using rust and how it compares to some of modern programming languages e.g. golang, participants will be equipped to know where rust will be most effective as a choice and what tradeoffs are there to use rust.

srwalter, 2021-04-24:
@kraj that's a really good description, I borrowed from it heavily :smile: 

I put up my rough draft of an outline as a wiki page: https://github.com/meta-rust/meta-rust/wiki/Yocto-Summit-2021

I'll need to submit the Abstract and Description by 11:59PM PST tomorrow.  Again, I appreciate any feedback y'all would like to provide!

kraj, 2021-04-24:
> @kraj that's a really good description, I borrowed from it heavily 😄
> 
> I put up my rough draft of an outline as a wiki page: https://github.com/meta-rust/meta-rust/wiki/Yocto-Summit-2021
> 

this looks pretty neat now.

> I'll need to submit the Abstract and Description by 11:59PM PST tomorrow. Again, I appreciate any feedback y'all would like to provide!

rwmacleod, 2021-04-25:
I think that looks like it will be a great presentation. Thanks for putting it together Steven!

srwalter, 2021-04-25:
Submitted!

closed 2021-04-25 by srwalter