meta-rust/meta-rust #339: Crate depending on `zmq` fail to compile when targeting musl

aj-bagwell, 2021-04-21:
## Version(s) of meta-rust
master (a4302679bbe2bfc5762aa47003ca8192f21a1342)

## Version(s) of poky and/or oe-core
master/hardknott (7d0988966cc1fd9e93549ca2ec21ac3bc765ceb6)

## Expected result
package compile fine

## Actual result
Complile fails with a link error

## Steps to reproduce

Compile a package that has a non trivial C dependency for example something that depends on [zmq](https://crates.io/crates/zmq) with `TCLIBC = "musl"`

aj-bagwell, 2021-04-21:
I have attached pull request #340 with the hack I used to work around this more as a point of discussion than something I expect to gt merged as is.

The basic idea behind my hack is that the linker used for `build.rs` scripts `RUST_BUILD_CCLD` should only have "native" things on its path e.g. `recipe-sysroot-native` so filter out all the non native `recipe-sysroot` included as they are for the target and will only confuse the linker.

jmesmon, 2021-04-27:
The inclusion of a `-L */recipe-sysroot/*` arg here is interesting an somewhat unexpected.

My first thought here is trying to figure out what is causing it to be included. Options include the `zmq`,`zmq-sys`,`zeromq-src`, and/or `cmake` crates doing something funny (or some other related crate here. possible zmq is a red herring).

Possible that something about how cmake treats env variables combined with how yocto uses env variables doesn't quite cooperate. [`cmake.bbclass`](https://github.com/openembedded/openembedded/blob/master/classes/cmake.bbclass) may provide some hints for us.

Resolving this might be possible by having recipies that internally call cmake do some special incantations to set the toolchain appropriately. It may be necessary to get some changes made to the `cmake` crate so that overrides for the toolchain can properly be set (so the `cmake` crate attempts at setting options don't interfere with oe's), but we might get lucky here.

Alternately, using the non-vendored zmq-sys capability might work, but it looks like it needs changes to support setting different libraries for different targets (iow: right now it can't support 2 different provided libraries for build env and target env), and also requires feature flags rather than build time env vars to pick which to link with, so you might need to patch up the `Cargo.toml` of dependents so that one can link against the non-vendored version of the library globally.