meta-rust/meta-rust #341: bitbake cargo 1.51 fails on dunfell

williewiholm, 2021-04-22:
## Version(s) of meta-rust
 master 920db7b045c1f721efcc2e1d
## Version(s) of poky and/or oe-core
Dunfell
## Expected result
cargo 1.51 builds successfully
## Actual result

87 | extern crate proc_macro;
|    | ^^^^^^^^^^^^^^^^^^^^^^^^ can't find crate
| 
| error: aborting due to previous error
| 
| For more information about this error, try `rustc --explain E0463`.
| error: could not compile `proc-macro2`
| 
| Caused by:
|   process didn't exit successfully: `rustc --crate-name proc_macro2 --edition=2018 /tmp/work/aarch64-eg-linux/cargo/1.51.0-r0/rustc-1.51.0-src/vendor/proc-macro2-1.0.19/src/lib.rs --error-format=json --json=diagnostic-rendered-ansi,artifacts --crate-type lib --emit=dep-info,metadata,link -C opt-level=3 -C embed-bitcode=no --cfg 'feature="default"' --cfg 'feature="proc-macro"' -C metadata=cb5ee8106fa0df70 -C extra-filename=-cb5ee8106fa0df70 --out-dir /tmp/work/aarch64-eg-linux/cargo/1.51.0-r0/build/target/aarch64-eg-linux/release/deps --target aarch64-eg-linux -C linker=/tmp/work/aarch64-eg-linux/cargo/1.51.0-r0/wrapper/target-rust-ccld -L dependency=/tmp/work/aarch64-eg-linux/cargo/1.51.0-r0/build/target/aarch64-eg-linux/release/deps -L dependency=/tmp/work/aarch64-eg-linux/cargo/1.51.0-r0/build/target/release/deps --extern unicode_xid=/tmp/work/aarch64-eg-linux/cargo/1.51.0-r0/build/target/aarch64-eg-linux/release/deps/libunicode_xid-1a6ef0c4b108f696.rmeta --cap-lints allow -L /tmp/work/aarch64-eg-linux/cargo/1.51.0-r0/recipe-sysroot/usr/lib/rust --remap-path-prefix=/tmp/work/aarch64-eg-linux/cargo/1.51.0-r0=/usr/src/debug/cargo/1.51.0-r0 --cfg hygiene --cfg use_proc_macro --cfg wrap_proc_macro` (exit code: 1)
| WARNING: exit code 101 from a shell command.

## Steps to reproduce

Host Ubuntu 20.04
build from scratch on dunfell

weltling, 2021-04-22:
Thanks for the report. Are you building something specific? Perhaps you could post the relevant part from the `local.conf`? Also, which Dunfell version? We have several projects based on Dunfell and things seem to work ok.

Thanks.

wbw455, 2021-04-22:
Hello, 

I'm trying to upgrade my Azure iotedge image to a newer Rust and Cargo version. Altought this error comes from simply trying to compile cargo 1.51. 

Poky commit : 13f4ddf50eccaeed96a

[local.txt.txt](https://github.com/meta-rust/meta-rust/files/6358847/local.txt.txt)

weltling, 2021-04-22:
Ah, but build for target isn't supported. Do i understand correctly, that's what you're doing?

Thanks

wbw455, 2021-04-22:
I'm sure I understand what you mean. I have a build with iotedge https://github.com/Azure/meta-iotedge that is based on Cargo 1.39. Before building my image I tried compiling Cargo 1.51 alone and it gives the error above.

Hope this clears it up.

weltling, 2021-04-22:
Thanks for more details. Yeah, you're building `cargo` for the target as the path shows things like `/tmp/work/aarch64-eg-linux/cargo/1.51.0-r0=/usr/src/debug/cargo/1.51.0-r0`. That won't work, please see also the [README.md](https://github.com/meta-rust/meta-rust#what-doesnt)

But that is also not needed, as the recipe which does `inherit cargo` will build `cargo-native` and use to cross compile the recipe. Thus, just building your image is enough.

Thanks

wbw455, 2021-04-22:
Hello, Thanks a lot for clearing this up. I will close this.

Have a nice evening!

closed 2021-04-22 by wbw455

weltling, 2021-04-23:
Great, thanks for confirming.