meta-rust/meta-rust #343: error: the crate `panic_abort` does not have the panic strategy `abort`

jordemort, 2021-05-04:
## Version(s) of meta-rust

master @ 02db606a9678bddf81169a3f2e7c6346e0876839

## Version(s) of poky and/or oe-core

poky - dunfell @ 320f059a9b7d7cf9b493778933469d5c604be42b

## Expected result

`websocat` builds without errors

## Actual result

`websocat` fails to build:

```
| error: the crate `panic_abort` does not have the panic strategy `abort`
| 
| error: aborting due to previous error
| 
| error: could not compile `websocat`
| 
| Caused by:
|   process didn't exit successfully: `rustc --crate-name websocat --edition=2018 src/main.rs --error-format=json --json=diagnostic-rendered-ansi --crate-type bin --emit=dep-info,link -C opt-level=s -C panic=abort -C lto -C codegen-units=1 --cfg 'feature="default"' --cfg 'feature="signal_handler"' --cfg 'feature="tokio-named-pipes"' --cfg 'feature="tokio-process"' --cfg 'feature="tokio-signal"' --cfg 'feature="unix_stdio"' --cfg 'feature="windows_named_pipes"' -C metadata=806f39075c0e81bf -C extra-filename=-806f39075c0e81bf --out-dir /mnt/build/tmp/work/cortexa7t2hf-neon-bantha-linux-gnueabi/websocat/1.8.0-r0/build/target/arm-bantha-linux-gnueabi/release/deps --target arm-bantha-linux-gnueabi -C linker=/mnt/build/tmp/work/cortexa7t2hf-neon-bantha-linux-gnueabi/websocat/1.8.0-r0/wrapper/target-rust-ccld -L dependency=/mnt/build/tmp/work/cortexa7t2hf-neon-bantha-linux-gnueabi/websocat/1.8.0-r0/build/target/arm-bantha-linux-gnueabi/release/deps -L dependency=/mnt/build/tmp/work/cortexa7t2hf-neon-bantha-linux-gnueabi/websocat/1.8.0-r0/build/target/release/deps --extern anymap=/mnt/build/tmp/work/cortexa7t2hf-neon-bantha-linux-gnueabi/websocat/1.8.0-r0/build/target/arm-bantha-linux-gnueabi/release/deps/libanymap-2bade36f4e6bd39f.rlib --extern atty=/mnt/build/tmp/work/cortexa7t2hf-neon-bantha-linux-gnueabi/websocat/1.8.0-r0/build/target/arm-bantha-linux-gnueabi/release/deps/libatty-287180326a573dae.rlib --extern base64=/mnt/build/tmp/work/cortexa7t2hf-neon-bantha-linux-gnueabi/websocat/1.8.0-r0/build/target/arm-bantha-linux-gnueabi/release/deps/libbase64-af4a0171192c5da3.rlib --extern derivative=/mnt/build/tmp/work/cortexa7t2hf-neon-bantha-linux-gnueabi/websocat/1.8.0-r0/build/target/release/deps/libderivative-e2b8581db0dfa063.so --extern env_logger=/mnt/build/tmp/work/cortexa7t2hf-neon-bantha-linux-gnueabi/websocat/1.8.0-r0/build/target/arm-bantha-linux-gnueabi/release/deps/libenv_logger-f3d9f38873e88f08.rlib --extern futures=/mnt/build/tmp/work/cortexa7t2hf-neon-bantha-linux-gnueabi/websocat/1.8.0-r0/build/target/arm-bantha-linux-gnueabi/release/deps/libfutures-30d4ba93a2a57b2c.rlib --extern hex=/mnt/build/tmp/work/cortexa7t2hf-neon-bantha-linux-gnueabi/websocat/1.8.0-r0/build/target/arm-bantha-linux-gnueabi/release/deps/libhex-b2bfa0acc8a4f35b.rlib --extern http_bytes=/mnt/build/tmp/work/cortexa7t2hf-neon-bantha-linux-gnueabi/websocat/1.8.0-r0/build/target/arm-bantha-linux-gnueabi/release/deps/libhttp_bytes-23171ec358f1b557.rlib --extern hyper=/mnt/build/tmp/work/cortexa7t2hf-neon-bantha-linux-gnueabi/websocat/1.8.0-r0/build/target/arm-bantha-linux-gnueabi/release/deps/libhyper-01df6e7fad0f5174.rlib --extern libc=/mnt/build/tmp/work/cortexa7t2hf-neon-bantha-linux-gnueabi/websocat/1.8.0-r0/build/target/arm-bantha-linux-gnueabi/release/deps/liblibc-f22657723912b3e6.rlib --extern log=/mnt/build/tmp/work/cortexa7t2hf-neon-bantha-linux-gnueabi/websocat/1.8.0-r0/build/target/arm-bantha-linux-gnueabi/release/deps/liblog-ec32d65f11431c2f.rlib --extern net2=/mnt/build/tmp/work/cortexa7t2hf-neon-bantha-linux-gnueabi/websocat/1.8.0-r0/build/target/arm-bantha-linux-gnueabi/release/deps/libnet2-cb89f462a6d9423f.rlib --extern slab_typesafe=/mnt/build/tmp/work/cortexa7t2hf-neon-bantha-linux-gnueabi/websocat/1.8.0-r0/build/target/arm-bantha-linux-gnueabi/release/deps/libslab_typesafe-5b432705caa07ca8.rlib --extern smart_default=/mnt/build/tmp/work/cortexa7t2hf-neon-bantha-linux-gnueabi/websocat/1.8.0-r0/build/target/release/deps/libsmart_default-60d0b83853947d06.so --extern structopt=/mnt/build/tmp/work/cortexa7t2hf-neon-bantha-linux-gnueabi/websocat/1.8.0-r0/build/target/arm-bantha-linux-gnueabi/release/deps/libstructopt-3e2b83a0d1f3c257.rlib --extern structopt_derive=/mnt/build/tmp/work/cortexa7t2hf-neon-bantha-linux-gnueabi/websocat/1.8.0-r0/build/target/release/deps/libstructopt_derive-a373d3245e8e070b.so --extern tempfile=/mnt/build/tmp/work/cortexa7t2hf-neon-bantha-linux-gnueabi/websocat/1.8.0-r0/build/target/arm-bantha-linux-gnueabi/release/deps/libtempfile-cd0d3e1663da3f89.rlib --extern tk_listen=/mnt/build/tmp/work/cortexa7t2hf-neon-bantha-linux-gnueabi/websocat/1.8.0-r0/build/target/arm-bantha-linux-gnueabi/release/deps/libtk_listen-031ebb022f28af6e.rlib --extern tokio=/mnt/build/tmp/work/cortexa7t2hf-neon-bantha-linux-gnueabi/websocat/1.8.0-r0/build/target/arm-bantha-linux-gnueabi/release/deps/libtokio-eadc116b34186b40.rlib --extern tokio_codec=/mnt/build/tmp/work/cortexa7t2hf-neon-bantha-linux-gnueabi/websocat/1.8.0-r0/build/target/arm-bantha-linux-gnueabi/release/deps/libtokio_codec-1b0a77d28140b61d.rlib --extern tokio_current_thread=/mnt/build/tmp/work/cortexa7t2hf-neon-bantha-linux-gnueabi/websocat/1.8.0-r0/build/target/arm-bantha-linux-gnueabi/release/deps/libtokio_current_thread-090693d23788c9c1.rlib --extern tokio_file_unix=/mnt/build/tmp/work/cortexa7t2hf-neon-bantha-linux-gnueabi/websocat/1.8.0-r0/build/target/arm-bantha-linux-gnueabi/release/deps/libtokio_file_unix-92ecad597716a087.rlib --extern tokio_io=/mnt/build/tmp/work/cortexa7t2hf-neon-bantha-linux-gnueabi/websocat/1.8.0-r0/build/target/arm-bantha-linux-gnueabi/release/deps/libtokio_io-a209211364773a07.rlib --extern tokio_process=/mnt/build/tmp/work/cortexa7t2hf-neon-bantha-linux-gnueabi/websocat/1.8.0-r0/build/target/arm-bantha-linux-gnueabi/release/deps/libtokio_process-d6aba11808464261.rlib --extern tokio_reactor=/mnt/build/tmp/work/cortexa7t2hf-neon-bantha-linux-gnueabi/websocat/1.8.0-r0/build/target/arm-bantha-linux-gnueabi/release/deps/libtokio_reactor-3155dcbae6c833c8.rlib --extern tokio_signal=/mnt/build/tmp/work/cortexa7t2hf-neon-bantha-linux-gnueabi/websocat/1.8.0-r0/build/target/arm-bantha-linux-gnueabi/release/deps/libtokio_signal-2e13a1d04dab0e0c.rlib --extern tokio_stdin_stdout=/mnt/build/tmp/work/cortexa7t2hf-neon-bantha-linux-gnueabi/websocat/1.8.0-r0/build/target/arm-bantha-linux-gnueabi/release/deps/libtokio_stdin_stdout-fdee656e416fd5e9.rlib --extern tokio_tcp=/mnt/build/tmp/work/cortexa7t2hf-neon-bantha-linux-gnueabi/websocat/1.8.0-r0/build/target/arm-bantha-linux-gnueabi/release/deps/libtokio_tcp-41b0e74cdc748cd6.rlib --extern tokio_timer=/mnt/build/tmp/work/cortexa7t2hf-neon-bantha-linux-gnueabi/websocat/1.8.0-r0/build/target/arm-bantha-linux-gnueabi/release/deps/libtokio_timer-d71a298625b4b10b.rlib --extern tokio_udp=/mnt/build/tmp/work/cortexa7t2hf-neon-bantha-linux-gnueabi/websocat/1.8.0-r0/build/target/arm-bantha-linux-gnueabi/release/deps/libtokio_udp-f02ff8e55ca50e77.rlib --extern tokio_uds=/mnt/build/tmp/work/cortexa7t2hf-neon-bantha-linux-gnueabi/websocat/1.8.0-r0/build/target/arm-bantha-linux-gnueabi/release/deps/libtokio_uds-cbe468c790a49586.rlib --extern url=/mnt/build/tmp/work/cortexa7t2hf-neon-bantha-linux-gnueabi/websocat/1.8.0-r0/build/target/arm-bantha-linux-gnueabi/release/deps/liburl-e814afbe6ff0e148.rlib --extern websocat=/mnt/build/tmp/work/cortexa7t2hf-neon-bantha-linux-gnueabi/websocat/1.8.0-r0/build/target/arm-bantha-linux-gnueabi/release/deps/libwebsocat-8f3615d678e16284.rlib --extern websocket=/mnt/build/tmp/work/cortexa7t2hf-neon-bantha-linux-gnueabi/websocat/1.8.0-r0/build/target/arm-bantha-linux-gnueabi/release/deps/libwebsocket-753f5a75e561a2f8.rlib --extern websocket_base=/mnt/build/tmp/work/cortexa7t2hf-neon-bantha-linux-gnueabi/websocat/1.8.0-r0/build/target/arm-bantha-linux-gnueabi/release/deps/libwebsocket_base-3f8673c1a25af904.rlib -L /mnt/build/tmp/work/cortexa7t2hf-neon-bantha-linux-gnueabi/websocat/1.8.0-r0/recipe-sysroot/usr/lib/rust --remap-path-prefix=/mnt/build/tmp/work/cortexa7t2hf-neon-bantha-linux-gnueabi/websocat/1.8.0-r0=/usr/src/debug/websocat/1.8.0-r0` (exit code: 1)
| WARNING: /mnt/build/tmp/work/cortexa7t2hf-neon-bantha-linux-gnueabi/websocat/1.8.0-r0/temp/run.do_compile.51577:1 exit 101 from '"cargo" build -v --target arm-bantha-linux-gnueabi --release --manifest-path=/mnt/build/tmp/work/cortexa7t2hf-neon-bantha-linux-gnueabi/websocat/1.8.0-r0/git//Cargo.toml "$@"'
```

## Steps to reproduce

[... cut ...]
    crate://crates.io/ws2_32-sys/0.2.1 \
"

# FIXME: update generateme with the real MD5 of the license file
LIC_FILES_CHKSUM = " \
    file://LICENSE;md5=b0249af598633524d7f0cb5b558c832c \
"

SUMMARY = "Command-line client for web sockets, like netcat/curl/socat for ws://."
HOMEPAGE = "https://github.com/vi/websocat"
LICENSE = "MIT"

# includes this file if it exists but does not fail
# this is useful for anything you may want to override from
# what cargo-bitbake generates.
include websocat-${PV}.inc
include websocat.inc
```

jwinarske, 2021-05-23:
One method to handle this is to append `RUST_PANIC_STRATEGY = "abort"` to your local.conf and rebuild the recipe.  It will re-trigger build of rust-native, cargo-native, and libstd-rs.  The default strategy is set for [`unwind`](https://github.com/meta-rust/meta-rust/blob/920db7b045c1f721efcc2e1d891516b515b7e7a8/classes/rust-common.bbclass#L12)

nateglims, 2021-09-30:
I'm seeing this if I set `RUST_PANIC_STRATEGY` to abort in local.conf on x86-64 (or if the crate sets it). But it is working on arm64.

nateglims, 2021-09-30:
ah nevermind. I tried to set up a minimal build and it worked.

frankplus, 2021-12-15:
adding `RUST_PANIC_STRATEGY = "abort"` to local.conf did not work for me

frankplus, 2021-12-17:
even with the simplest hello world with panic abort is not working
https://github.com/frankplus/rust-hello-world

jwinarske, 2021-12-17:
@frankplus I believe my suggestion was for 1.51.  what version/branch are you having trouble with?  Also honister implements rust in OE-core, so if you can replicate there Yocto list can help.

frankplus, 2021-12-18:
The problem appears on both dunfell (with meta-rust) and honister (with rust in oe-core)

frankplus, 2021-12-27:
Relate-to: https://lists.openembedded.org/g/Openembedded-issues/message/12289

nateglims, 2022-05-20:
Part of this seems to stem from the cargo features set in libstd [1] enabling panic_unwind in the std crate [2]. By setting `CARGO_FEATURES = "backtrace"` I was able to get past this but I haven't verified if the panic behavior is working.

[1] https://git.yoctoproject.org/poky/tree/meta/recipes-devtools/rust/libstd-rs.inc#n22
[2] https://github.com/rust-lang/rust/blob/master/library/std/Cargo.toml#L59

jwinarske, 2022-05-20:
Building with kirkstone I don't have to set the strategy, as the strategy defaults to `unwind`:  https://git.openembedded.org/openembedded-core/tree/meta/classes/rust-common.bbclass?h=kirkstone#n14

Changing the strategy affects all rust recipes in the image.  Using panic abort leaves no cookie crumbs in the syslog, which is why unwind is preferred.  So my point is that with kirkstone tip of tree, it should just work.

For kirkstone dynamic layers aren't used for rust recipes, as it's now part of oe-core.

nateglims, 2022-05-20:
That's a good point, my suggestion is definitely a hack. This issue is coming up when a crate sets its panic to abort which they may or may not have a good reason to set. Mozjs just patched out panics in the Cargo.toml file, but that might not work for every situation.

Websocat, from the OP, set's it's panic type: https://github.com/vi/websocat/blob/master/Cargo.toml#L112 
Firecracker, provided in meta-aws, does too: https://github.com/firecracker-microvm/firecracker/blob/main/Cargo.toml#L9

My understanding for a normal rustup setup is they ship a pre-built std library, so I'm not really clear why this cargo feature setting matters. I'm mostly bumping into the std-aware workgroup, the rustc development manual and some nightly features when I search for this.

jwinarske, 2022-05-20:
Yes I contributed the firecracker recipe to meta-aws, so I'm pretty familiar with it.

moto-timo, 2024-03-27:
> I have built a number of things against Kirkstone without issue, and no need to override anything.
> 
> Example Kirkstone recipe: https://github.com/meta-flutter/meta-flutter/blob/kirkstone/recipes-devtools/membrane/membrane-dart-example_git.bb
> 
> Requires only bbappend (proc2 workaround): https://github.com/meta-flutter/meta-flutter/blob/kirkstone/recipes-devtools/rust/libstd-rs_%25.bbappend

These are now deadlinks.

jwinarske, 2024-03-28:
@moto-timo Hi Tim.
I recently removed the membrane-dart-example due to upstream issue, but it can be found here:
https://github.com/meta-flutter/meta-flutter/blob/e08c45c6b8c961734e7be9745e0f1ea344fa5ea8/dynamic-layers/clang-layer/recipes-devtools/membrane/membrane-example_git.bb

The bbappend was removed as Proc2 issue goes away with Scott's Rust toolchain mix-in (1.69+).  So I opted to let people manage their own pain :)
https://github.com/meta-flutter/meta-flutter/blob/e08c45c6b8c961734e7be9745e0f1ea344fa5ea8/recipes-devtools/rust/libstd-rs_%25.bbappend

moto-timo, 2024-03-28:
> I recently removed the membrane-dart-example due to upstream issue, but it can be found here:
> https://github.com/meta-flutter/meta-flutter/blob/e08c45c6b8c961734e7be9745e0f1ea344fa5ea8/dynamic-layers/clang-layer/recipes-devtools/membrane/membrane-example_git.bb

This recipe can be greatly simplified by using the [cargo-update-recipe-crates.bbclass](https://git.yoctoproject.org/poky/tree/meta/classes-recipe/cargo-update-recipe-crates.bbclass). I'm also trying to find the time to add knowledge of that bbclass into devtoo/recipetool so the Auto Upgrade Helper (AUH) will be aware of it. Baby steps.

jwinarske, 2024-03-29:
@moto-timo I'll check it out.  I have a number of very complicated rust recipes.  One has over 800 crates, curious how the helper bbclass will fare.

This has been my go-to pattern (generic non-multipass repos):
* Rustup host toolchain to match Yocto version
* Delete tool chain version file, or update to match Yocto toolchain
* delete cargo.lock file (likely it may vary otherwise)
* build standalone host.  Forces correct cargo.lock
* Run tool to generate/update the src list:
https://github.com/jwinarske/root-pkg-ws

The multi-pass like this membrane recipe uses llvm and dart for native pass (code gen), then native build artifacts for target pass.  It looks like the referenced bbclass may replace my root-pkg-ws tool.  That would be nice.  I'm all for making the Yocto rust support better!

peterdelevoryas, 2025-01-06:
tbh I still don't get how to set `panic = "abort"`, I tried `CARGO_FEATURES = "backtrace"` in my recipe but that didn't change anything. If `local.conf` `RUST_PANIC_STRATEGY` is required, I guess I have to change it for the whole image? Which would be fine for me I think, just curious.

Since I saw a comment questioning why anyone would be interested in this: [Unwind considered harmful?](https://smallcultfollowing.com/babysteps/blog/2024/05/02/unwind-considered-harmful/)