meta-rust/meta-rust #355: [sdk] Proc Macro fails to link

jwinarske, 2021-07-21:
## Version(s) of meta-rust
master

## Version(s) of poky and/or oe-core
dunfell

## Expected result
proc macro links

## Actual result
```
  = note: /usr/bin/ld: cannot find /lib/libm.so.6
          /usr/bin/ld: cannot find /lib/libmvec.so.1
          collect2: error: ld returned 1 exit status
```
The compiler needs to be passed `--sysroot=`

#### Example for Ubuntu 20
/usr/lib32/libm.so.6
/usr/lib/x86_64-linux-gnu/libm.so.6
/opt/poky/3.1.9/sysroots/x86_64-pokysdk-linux/lib/libm.so.6

## Steps to reproduce
create rust cross canadian sdk
set environment `. /opt/poky/3.1.9/environment-setup-corei7-64-poky-linux`
build project containing a proc macro crate appending `--target x86_64-poky-linux`

Given proc macros only build via HOST, how do I pass `--sysroot=${OECORE_NATIVE_SYSROOT}` to HOST?

@weltling

weltling, 2021-07-22:
Thanks for the report, @jwinarske.

As a quick tip - once the environment is setup, RUSTFLAGS will contain `--sysroot=` and other necessary flags that will be passed to `rustc`. I'd suggest you to play with that variable. In general however - the SDK supplies the libraries for both host and target and a number of other Rust projects have build scripts that need to run on host, that usually seems to work fine.

Perhaps you could also link to an existing project having this issue?

Thanks

jwinarske, 2021-07-22:
I see.
```
env |grep RUSTFLAGS
RUSTFLAGS=--sysroot=/opt/poky/3.1.9/sysroots/x86_64-pokysdk-linux/usr -C link-arg=--sysroot=/opt/poky/3.1.9/sysroots/corei7-64-poky-linux -L/opt/poky/3.1.9/sysroots/x86_64-pokysdk-linux/usr/lib/x86_64-poky-linux/rustlib/x86_64-poky-linux/lib
```
This makes sense why building without --target behaves different.

Not using "--target" does work for case of x86_64 target, but will fail for aarc64 target.  The rustc auto generates some compiler flags.  Passing x86_64 triplet to aarch64 will fail as aarch64 compiler does not appreciate the "-m64" option.  So to set correct compiler flags one should "--target <triplet>".

Regarding use of "--target" I found this documentation informative:
https://doc.rust-lang.org/cargo/reference/config.html
build.rustflags:  "If the --target flag (or build.target) is used, then the flags will only be passed to the compiler for the target. Things being built for the host, such as build scripts or proc macros, will not receive the args."

The exception I hit using "--target <triplet>" is with proc macros.  Proc macros are only built with HOST.  In which case it uses the host triplet and does not pickup RUSTFLAGS (by design?!), so it tries to find libraries relative to the default host sysroot, and one potentially ends up with this:
```
  = note: /usr/bin/ld: cannot find /lib/libm.so.6
          /usr/bin/ld: cannot find /lib/libmvec.so.1
```
I believe a valid solution would be to pass a correct --sysroot when host triplet is active.  My attempt using .cargo/config is here:
https://github.com/rust-lang/cargo/issues/9713

I'll look at creating a MVP repro case if that helps, but I think I've pretty much narrowed it down.

weltling, 2021-07-22:
Thanks for the further check. Yeah, i think the issue gets more clear now. Using `--target` will trigger some different logic in cargo/rust, that's the issue. The tricky point on the Yocto side is also, that targets like `x86_64-poky-linux` are non not standard build triplets that Rust would understand. But Rust has a mechanism making possible to handle these custom target names by accepting definitions in json, so that's fine. The target name will depend on the name of the `DISTRO` user set in their `local.conf`, etc.

> The exception I hit using "--target " is with proc macros. 
The `--target` invoked for `build.rs` will be the host arch and will result in a binary able to run on host. It is something done by `cargo` in course of processing the build configuration of the project.

I guess the conlusion is this - using an extra `--target` is neither necessary nor appropriate given the nature of Yocto SDK.  When using just `cargo build` in the SDK, the current host arch and target arch, as well as any possible custom triplets, will be taken into account. Even if `x86_64-poky-linux` would be indeed same as `x86_64-unknown-linux-gnu`, the way things are layered out in the SDK to support the custom triplet would prevent Rust realizing that.

Similarly - say if  the SDK is optimized for corei7, there won't be a chance to use another (for example generic x86_64). It doesn't only regard Rust, it is about any related toolchains. Similarly where say gcc and binutils will by default produce binaries for the target arch, the Rust in the SDK will only contain `std` and accompanying tools only in two flavors - the host flavor so it can possibly produce build scripts and in target flavor to generate artifacts for target. Note, that not only Rust, but pretty much anything will go by this scheme in most case (well, perhaps golang?). The whole GNU toolchain, any DSOs and else, etc. The Yocto SDK will produce the artifacts with just one capacity - run on host and generate for target. Say to build for `aarch64`, the way is to produce the dedicated SDK that will contain all the suitable toolchains and dependencies. I currently don't see much opportunity to offer more flexibility.

Probably it's a good time to start documenting this, good points are being discovered.

Thanks!

jwinarske, 2021-07-22:
In order to build a target where the machine architecture is different than the host it requires using "--target <triplet>".  There's no way around this.  Not using it for same machine architecture cross compiles will end up using non-optimal build flags, as the triplet is consumed by the compiler to set proper flags, as well as used for path/lib/plugin resolution.  For best runtime performance you want to be as specific with your triplet as possible when cross compiling, especially with a toolchain targeting a specific SoC/CPU.  Already by creating the SDK it's a step towards optimizing, and in fact the target-list could/should be trimmed to reflect what is available with the unique toolchain.

When I build a SDK targeting MACHINE=intel-corei7-64, install on host, and build using "--target x86_64-poky-linux" it works, and this triplet is passed to all the crates.  If I build the same for MACHINE=raspberrypi4-64, it works with "--target aarch64-poky-linux", same behavior as with x86_64 SDK.  This is an outcome of building the custom toolchain.

The problem remains - How to pass a sysroot to HOST (proc macro) when using "--target <triplet>" using the meta-rust Cross Canadian Toolchain?

jwinarske, 2021-07-22:
https://github.com/rust-lang/cargo/issues/9713

weltling, 2021-07-22:
> In order to build a target where the machine architecture is different than the host it requires using "--target ". 

This is not correct with regard to the SDK. The target and its optimizations chosen during the SDK creation will be used automatically when `cargo build` is used without any args. All the other parameters like target CFLAGS, CXXFLAGS and optimizations will be the same as if an image for the target would have been built directly. The whole SDK is optimized using flags chosen at the creation time, there might be issues if those parameters are modified at runtime.

Say if SDK is produced to target ARMv8 optimizations, adding flags to target an older version might work but is not guaranteed. Overriding `RUSTFLAGS` could sure work by chance, by default anything in the SDK is to reflect what was configured at the creation time.

Thanks

weltling, 2021-07-22:
Perhaps to articulate another way - `cargo build` in the SDK will automatically produce binaries for the defined target. This is matching same with what GCC does.

Thanks

jwinarske, 2021-07-22:
Okay thanks that helps.  I'll keep digging on my side.

jwinarske, 2021-07-22:
Well I have made progress with x86_64 SDK not using --target, but building the same thing with the aarch64 SDK I'm see the -m64 break:
```
  running: "aarch64-poky-linux-g++" "-mcpu=cortex-a72+crc+crypto" "-fstack-protector-strong" "-D_FORTIFY_SOURCE=2" "-Wformat" "-Wformat-security" "-Werror=format-security" "--sysroot=/opt/poky/3.1.9/rpi4/sysroots/aarch64-poky-linux" "-O0" "-ffunction-sections" "-fdata-sections" "-fPIC" "-g" "-fno-omit-frame-pointer" "-m64" "-O2" "-pipe" "-g" "-feliminate-unused-debug-types" "-o" "/home/joel/Downloads/dc-flutter/vehicle_interface/target/debug/build/cxx-765cfbfa42ca602f/out/src/cxx.o" "-c" "src/cxx.cc"
  cargo:warning=aarch64-poky-linux-g++: error: unrecognized command line option ‘-m64’

```
Building a different architect than host.  Which leads back to triplets.  Has anyone tested building different target architect than host?

weltling, 2021-07-27:
I don't think the FFI scenario is explicitly tested. The `-m64` might come from the build script. Pure Rust crates should be fine.

Probably best would be still to check on some repro project and probably add it as a test for both image build and SDK build. Would it be possible to compose some minimal repro please?

Thanks

jwinarske, 2021-09-08:
Still hitting this.  It's related to using cxx v1.0.49.  Only happens on aarch64 SDK.  Only when I build cxx demo project it's fine for both target architecture.  Some nasty interaction that is buried like a needle in the haystack.  I really am not a fan of cross compiling with cargo 1.51.0

jwinarske, 2021-09-09:
Found fix to magically appearing `-m64`.  Evil.  cc-rs crate was automatically adding it.

The fix is to set CRATE_CC_NO_DEFAULTS env variable

https://github.com/alexcrichton/cc-rs#external-configuration-via-environment-variables

closed 2021-09-09 by jwinarske

jwinarske, 2021-09-30:
The Rust guy is stating building via cross candian is not cross compilation.  Interesting.

https://github.com/alexcrichton/cc-rs/issues/624

jmesmon, 2021-10-01:
We might want to just set this magic env var by default. We already do this for a pkg-config related magic env var.

jwinarske, 2021-10-05:
That might be a good idea, as It took quite a long time to sort out.