meta-rust/meta-rust #360: ERROR: ParseError at /.../rust_security_flags.inc:2: unparsed line

etam, 2021-08-13:
## Version(s) of meta-rust
448047c7e4746eac8f0a96a2e3966b2219e2a4ca

## Version(s) of poky and/or oe-core
poky: hardknott-3.3.2 (31c639eb8664059eb4ed711be9173c223b4cc940)
oe-core: 5a4b2ab29d38c02535f24d5308cc40615739f557

## Expected result
No parsing errors.

## Actual result
```
ERROR: ParseError at /work/meta-rust/conf/distro/include/rust_security_flags.inc:2: unparsed line: 'SECURITY_CFLAGS:pn-rust-native = "${SECURITY_NO_PIE_CFLAGS}"'
```

## Steps to reproduce

Using [kas-container 2.5](https://github.com/siemens/kas/tree/2.5)

`kas-project-test.yml`:
```yml
header:
  version: 10
distro: poky
target: core-image-full-cmdline
machine: qemuarm64
repos:
  poky:
    url: "git://git.yoctoproject.org/poky.git"
    refspec: hardknott-3.3.2
    layers:
      meta:
      meta-poky:
      meta-yocto-bsp:
  meta-oe:
    url: "git://github.com/openembedded/meta-openembedded.git"
    refspec: hardknott
    layers:
      meta-oe:
  meta-rust:
    url: "git://github.com/meta-rust/meta-rust.git"
    refspec: master
```

```
$ kas-container build kas-project-test.yml
2021-08-13 13:44:26 - INFO     - kas 2.5 started
2021-08-13 13:44:26 - INFO     - /repo$ git rev-parse --show-toplevel
2021-08-13 13:44:26 - INFO     - /work/poky$ git remote set-url origin git://git.yoctoproject.org/poky.git
2021-08-13 13:44:26 - INFO     - /work/meta-oe$ git remote set-url origin git://github.com/openembedded/meta-openembedded.git
2021-08-13 13:44:26 - INFO     - /work/meta-rust$ git remote set-url origin git://github.com/meta-rust/meta-rust.git
2021-08-13 13:44:26 - INFO     - /work/poky$ git cat-file -t hardknott-3.3.2
2021-08-13 13:44:26 - INFO     - /work/meta-oe$ git cat-file -t hardknott
2021-08-13 13:44:26 - INFO     - /work/meta-rust$ git cat-file -t master
2021-08-13 13:44:26 - INFO     - Repository poky already contains hardknott-3.3.2 as tag
2021-08-13 13:44:26 - INFO     - Repository meta-oe already contains hardknott as commit
2021-08-13 13:44:26 - INFO     - Repository meta-rust already contains master as commit
2021-08-13 13:44:26 - INFO     - /work/poky$ git status -s
2021-08-13 13:44:26 - INFO     - /work/poky$ git rev-parse --verify -q origin/hardknott-3.3.2
2021-08-13 13:44:26 - INFO     - /work/poky$ git checkout -q hardknott-3.3.2
2021-08-13 13:44:26 - INFO     - /work/meta-oe$ git status -s
2021-08-13 13:44:26 - INFO     - /work/meta-oe$ git rev-parse --verify -q origin/hardknott
2021-08-13 13:44:26 - INFO     - 5a4b2ab29d38c02535f24d5308cc40615739f557
2021-08-13 13:44:26 - INFO     - /work/meta-oe$ git checkout -q 5a4b2ab29d38c02535f24d5308cc40615739f557 -B hardknott
2021-08-13 13:44:26 - INFO     - /work/meta-rust$ git status -s
2021-08-13 13:44:26 - INFO     - /work/meta-rust$ git rev-parse --verify -q origin/master
2021-08-13 13:44:26 - INFO     - 448047c7e4746eac8f0a96a2e3966b2219e2a4ca
2021-08-13 13:44:26 - INFO     - /work/meta-rust$ git checkout -q 448047c7e4746eac8f0a96a2e3966b2219e2a4ca -B master
2021-08-13 13:44:26 - INFO     - /work/poky$ /tmp/tmp33e1a7wj/get_bb_env /build
2021-08-13 13:44:26 - INFO     - /build$ /work/poky/bitbake/bin/bitbake -c build core-image-full-cmdline
ERROR: ParseError at /work/meta-rust/conf/distro/include/rust_security_flags.inc:2: unparsed line: 'SECURITY_CFLAGS:pn-rust-native = "${SECURITY_NO_PIE_CFLAGS}"'
2021-08-13 13:44:27 - ERROR    - Command returned non-zero exit status 1
```

etam, 2021-08-13:
According to https://github.com/meta-rust/meta-rust/pull/357#issuecomment-890819779 a newer bitbake version is required, than what's included in yocto git repo.

avrabe, 2021-08-15:
You need at least hardknott commit [666d3421389cc3c01c56e92199cc32bddf1922de](https://git.yoctoproject.org/cgit/cgit.cgi/poky/log/?h=666d3421389cc3c01c56e92199cc32bddf1922de) (The commit after the hardknott-3.3.2 release).

Below is the kas configuration I used. Mainly I'm also in poky on hardknott.

```yaml
header:
  version: 8
machine: qemuriscv64
target: rust-hello-world
repos:
  meta-custom:
  poky:
    url: "https://git.yoctoproject.org/git/poky"
    refspec: hardknott
    layers:
      meta:
      meta-poky:
  meta-openembedded:
    url: "https://github.com/openembedded/meta-openembedded.git"
    refspec: hardknott 
    layers:
      meta-oe:
local_conf_header:
  meta-custom: |
    TOOLCHAIN_HOST_TASK_append = " packagegroup-rust-cross-canadian-${MACHINE}"
    PREFERRED_PROVIDER_virtual/kernel = "linux-dummy"
```

etam, 2021-08-16:
Thanks, that works well.

closed 2021-08-16 by etam