meta-rust/meta-rust #361: No package openssl found for rust-openssl

IniterWorker, 2021-08-18:
## Version(s) of meta-rust
meta-rust            = "master:448047c7e4746eac8f0a96a2e3966b2219e2a4ca"

## Version(s) of poky and/or oe-core
meta-poky            = "dunfell:08acba82d160815549a3c237b8c4850f96534da6"

## Actual result
```
|   run pkg_config fail: "`\"pkg-config\" \"--libs\" \"--cflags\" \"openssl\"` did not exit successfully: exit status: 1\n--- stderr\nPackage openssl was not found in the pkg-config search path.\nPerhaps you should add the directory containing `openssl.pc'\nto the PKG_CONFIG_PATH environment variable\nNo package 'openssl' found\n"
| 
|   --- stderr
|   thread 'main' panicked at '
| 
|   Could not find directory of OpenSSL installation, and this `-sys` crate cannot
|   proceed without this knowledge. If OpenSSL is installed and this crate had
|   trouble finding it,  you can set the `OPENSSL_DIR` environment variable for the
|   compilation process.
| 
|   Make sure you also have the development packages of openssl installed.
|   For example, `libssl-dev` on Ubuntu or `openssl-devel` on Fedora.
| 
|   If you're in a situation where you think the directory *should* be found
|   automatically, please open a bug at https://github.com/sfackler/rust-openssl
|   and include information about your system as well as this message.
| 
|   $HOST = x86_64-linux
|   $TARGET = arm-poky-linux-gnueabi
|   openssl-sys = 0.9.65
| 
|   ', /workdir/build/tmp/work/cortexa8hf-neon-poky-linux-gnueabi/actix-hello-world/0.1.0.AUTOINC+4df39ad410-r0/cargo_home/bitbake/openssl-sys-0.9.65/build/find_normal.rs:174:5
|   stack backtrace:
|      0: rust_begin_unwind
|      1: std::panicking::begin_panic_fmt
|      2: build_script_main::find_normal::find_openssl_dir
|      3: build_script_main::find_normal::get_openssl::{{closure}}
|      4: core::option::Option<T>::unwrap_or_else
|      5: build_script_main::find_normal::get_openssl
|      6: build_script_main::find_openssl
|      7: build_script_main::main
|      8: core::ops::function::FnOnce::call_once
|   note: Some details are omitted, run with `RUST_BACKTRACE=full` for a verbose backtrace.
| warning: build failed, waiting for other jobs to finish...
|     Building [=======================> ] 179/185: actix-http
| error: build failed
| WARNING: /workdir/build/tmp/work/cortexa8hf-neon-poky-linux-gnueabi/actix-hello-world/0.1.0.AUTOINC+4df39ad410-r0/temp/run.do_compile.11816:1 exit 101 from '"cargo" build -v --target arm-poky-linux-gnueabi --release --manifest-path=/workdir/build/tmp/work/cortexa8hf-neon-poky-linux-gnueabi/actix-hello-world/0.1.0.AUTOINC+4df39ad410-r0/git//Cargo.toml "$@"'
| 
ERROR: Task (/workdir/meta-custom/recipes-core/ums/actix-hello-world_0.1.0.bb:do_compile) failed with exit code '1'
```

## Steps to reproduce

### Custom Recipe

```
RDEPENDS_${PN} = "openssl"
# please note if you have entries that do not begin with crate://
# you must change them to how that package can be fetched
SRC_URI += " \
    crate://crates.io/autocfg/1.0.1 \
    crate://crates.io/bitflags/1.3.2 \
    crate://crates.io/cc/1.0.69 \
    crate://crates.io/cfg-if/1.0.0 \
    crate://crates.io/foreign-types-shared/0.1.1 \
    crate://crates.io/foreign-types/0.3.2 \
    crate://crates.io/libc/0.2.99 \
    crate://crates.io/once_cell/1.8.0 \
    crate://crates.io/openssl-sys/0.9.65 \
    crate://crates.io/openssl/0.10.35 \
    crate://crates.io/pkg-config/0.3.19 \
    crate://crates.io/vcpkg/0.2.15 \
"
```

### Cargo.toml
```
[dependencies]
openssl = "0.10.35"
```

## Questions

- In README.md is mandatory to use `-native or are you referring to native target (a.k.a the normal package)? 

```
On the host:

Any -sys packages your project might need must have RDEPENDs for the native library.
On the target:

Any -sys packages your project might need must have RDEPENDs for the native library.
```

- Do I need more environment variables to build or do I need another dependency?

avrabe, 2021-08-18:
You might want to use `DEPENDS_append = " openssl ` " instead of the `RDEPENDS_${PN} = "openssl"`.

IniterWorker, 2021-08-20:
You're right. After many attempts, I didn't add it. My OpenSSL hello-world works. 

But, I still have an issue with openssl-sys when I am using it in a bigger recipe. I need to dig into it a bit more.

Thanks,

```
| The following warnings were emitted during compilation:
| 
| warning: In file included from /workdir/build/tmp/work/cortexa8hf-neon-poky-linux-gnueabi/ums-common/0.1.0.AUTOINC+1e26f6f039-r0/recipe-sysroot/usr/include/openssl/opensslconf.h:19:0,
| warning:                  from build/expando.c:2:
| warning: /workdir/build/tmp/work/cortexa8hf-neon-poky-linux-gnueabi/ums-common/0.1.0.AUTOINC+1e26f6f039-r0/recipe-sysroot/usr/include/bits/wordsize.h:23:2: error: #error "__WORDSIZE is not defined"
| warning:  #error "__WORDSIZE is not defined"
| warning:   ^~~~~
| warning: /workdir/build/tmp/work/cortexa8hf-neon-poky-linux-gnueabi/ums-common/0.1.0.AUTOINC+1e26f6f039-r0/recipe-sysroot/usr/include/bits/wordsize.h:46:2: error: #error "Unknown __WORDSIZE detected"
| warning:  #error "Unknown __WORDSIZE detected"
| warning:   ^~~~~
| warning: In file included from build/expando.c:2:0:
| warning: /workdir/build/tmp/work/cortexa8hf-neon-poky-linux-gnueabi/ums-common/0.1.0.AUTOINC+1e26f6f039-r0/recipe-sysroot/usr/include/openssl/opensslconf.h:23:2: error: #error "__WORDSIZE is not defined"
| warning:  #error "__WORDSIZE is not defined"
| warning:   ^~~~~
| warning: /workdir/build/tmp/work/cortexa8hf-neon-poky-linux-gnueabi/ums-common/0.1.0.AUTOINC+1e26f6f039-r0/recipe-sysroot/usr/include/openssl/opensslconf.h:46:2: error: #error "Unknown __WORDSIZE detected"
| warning:  #error "Unknown __WORDSIZE detected"
| warning:   ^~~~~
| 
| error: failed to run custom build command for `openssl-sys v0.9.65`
```

kraj, 2021-08-20:
yocto provides wrappers for `wordsize.h`  which defined it based on compiler internal define fo arch target e.g. for x86_64 this file looks like

```
/* Determine the wordsize from the preprocessor defines.  */

#if defined __x86_64__ && !defined __ILP32__
# define __WORDSIZE     64
#else
# define __WORDSIZE     32
#define __WORDSIZE32_SIZE_ULONG         0
#define __WORDSIZE32_PTRDIFF_LONG       0
#endif

#ifdef __x86_64__
# define __WORDSIZE_TIME64_COMPAT32     1
/* Both x86-64 and x32 use the 64-bit system call interface.  */
# define __SYSCALL_WORDSIZE             64
#else
# define __WORDSIZE_TIME64_COMPAT32     0
#endif
```

So if the target arch define is missing from compiler then it wont find __WORDSIZE

IniterWorker, 2021-08-20:
I isolated the packages related to the rust compiler issue. When I will have more time I will check crates' dependencies. I'm guessing that is related related "runtime-actix-native-tls".

```toml
[dependencies]
actix-web = "4.0.0-beta.8"
sqlx = { version = "0.5.5", features = [ "runtime-actix-native-tls" ] }
```

```
| The following warnings were emitted during compilation:
| 
| warning: In file included from /workdir/build/tmp/work/cortexa8hf-neon-poky-linux-gnueabi/actix-hello-world/0.1.0.AUTOINC+cdd49fb843-r0/recipe-sysroot/usr/include/openssl/opensslconf.h:19:0,
| warning:                  from build/expando.c:2:
| warning: /workdir/build/tmp/work/cortexa8hf-neon-poky-linux-gnueabi/actix-hello-world/0.1.0.AUTOINC+cdd49fb843-r0/recipe-sysroot/usr/include/bits/wordsize.h:23:2: error: #error "__WORDSIZE is not defined"
| warning:  #error "__WORDSIZE is not defined"
| warning:   ^~~~~
| warning: /workdir/build/tmp/work/cortexa8hf-neon-poky-linux-gnueabi/actix-hello-world/0.1.0.AUTOINC+cdd49fb843-r0/recipe-sysroot/usr/include/bits/wordsize.h:46:2: error: #error "Unknown __WORDSIZE detected"
| warning:  #error "Unknown __WORDSIZE detected"
| warning:   ^~~~~
| warning: In file included from build/expando.c:2:0:
| warning: /workdir/build/tmp/work/cortexa8hf-neon-poky-linux-gnueabi/actix-hello-world/0.1.0.AUTOINC+cdd49fb843-r0/recipe-sysroot/usr/include/openssl/opensslconf.h:23:2: error: #error "__WORDSIZE is not defined"
| warning:  #error "__WORDSIZE is not defined"
| warning:   ^~~~~
| warning: /workdir/build/tmp/work/cortexa8hf-neon-poky-linux-gnueabi/actix-hello-world/0.1.0.AUTOINC+cdd49fb843-r0/recipe-sysroot/usr/include/openssl/opensslconf.h:46:2: error: #error "Unknown __WORDSIZE detected"
| warning:  #error "Unknown __WORDSIZE detected"
| warning:   ^~~~~
| 
| error: failed to run custom build command for `openssl-sys v0.9.66`
| 
| Caused by:
|   process didn't exit successfully: `/workdir/build/tmp/work/cortexa8hf-neon-poky-linux-gnueabi/actix-hello-world/0.1.0.AUTOINC+cdd49fb843-r0/build/target/release/build/openssl-sys-ca8a378065f06e63/build-script-main` (exit status: 101)
|   --- stdout
|   cargo:rustc-cfg=const_fn
|   cargo:rerun-if-env-changed=X86_64_LINUX_OPENSSL_LIB_DIR
|   X86_64_LINUX_OPENSSL_LIB_DIR unset
|   cargo:rerun-if-env-changed=OPENSSL_LIB_DIR
|   OPENSSL_LIB_DIR unset
|   cargo:rerun-if-env-changed=X86_64_LINUX_OPENSSL_INCLUDE_DIR
|   X86_64_LINUX_OPENSSL_INCLUDE_DIR unset
|   cargo:rerun-if-env-changed=OPENSSL_INCLUDE_DIR
|   OPENSSL_INCLUDE_DIR unset
|   cargo:rerun-if-env-changed=X86_64_LINUX_OPENSSL_DIR
|   X86_64_LINUX_OPENSSL_DIR unset
|   cargo:rerun-if-env-changed=OPENSSL_DIR
|   OPENSSL_DIR unset
|   cargo:rerun-if-env-changed=OPENSSL_NO_PKG_CONFIG
|   cargo:rerun-if-env-changed=PKG_CONFIG
|   cargo:rerun-if-env-changed=OPENSSL_STATIC
|   cargo:rerun-if-env-changed=OPENSSL_DYNAMIC
|   cargo:rerun-if-env-changed=PKG_CONFIG_ALL_STATIC
|   cargo:rerun-if-env-changed=PKG_CONFIG_ALL_DYNAMIC
|   cargo:rerun-if-env-changed=PKG_CONFIG_PATH_x86_64-linux
|   cargo:rerun-if-env-changed=PKG_CONFIG_PATH_x86_64_linux
|   cargo:rerun-if-env-changed=HOST_PKG_CONFIG_PATH
|   cargo:rerun-if-env-changed=PKG_CONFIG_PATH
|   cargo:rerun-if-env-changed=PKG_CONFIG_LIBDIR_x86_64-linux
|   cargo:rerun-if-env-changed=PKG_CONFIG_LIBDIR_x86_64_linux
|   cargo:rerun-if-env-changed=HOST_PKG_CONFIG_LIBDIR
|   cargo:rerun-if-env-changed=PKG_CONFIG_LIBDIR
|   cargo:rerun-if-env-changed=PKG_CONFIG_SYSROOT_DIR_x86_64-linux
|   cargo:rerun-if-env-changed=PKG_CONFIG_SYSROOT_DIR_x86_64_linux
|   cargo:rerun-if-env-changed=HOST_PKG_CONFIG_SYSROOT_DIR
|   cargo:rerun-if-env-changed=PKG_CONFIG_SYSROOT_DIR
|   cargo:rerun-if-env-changed=PKG_CONFIG_SYSROOT_DIR
|   cargo:rerun-if-env-changed=OPENSSL_STATIC
|   cargo:rerun-if-env-changed=OPENSSL_DYNAMIC
|   cargo:rerun-if-env-changed=PKG_CONFIG_ALL_STATIC
|   cargo:rerun-if-env-changed=PKG_CONFIG_ALL_DYNAMIC
|   cargo:rustc-link-lib=ssl
|   cargo:rustc-link-lib=crypto
[... 76 lines cut ...]
|      6: core::option::Option<T>::unwrap_or_else
|      7: build_script_main::find_normal::get_openssl
|      8: build_script_main::find_openssl
|      9: build_script_main::main
|     10: core::ops::function::FnOnce::call_once
|   note: Some details are omitted, run with `RUST_BACKTRACE=full` for a verbose backtrace.
| warning: build failed, waiting for other jobs to finish...
|     Building [==================>      ] 290/365: sqlformat, encoding_rs, r...
|     Building [==================>      ] 291/365: sqlformat, encoding_rs, r...
|     Building [===================>     ] 292/365: sqlformat, encoding_rs, r...
|     Building [===================>     ] 293/365: sqlformat, rand, tokio, u...
|     Building [===================>     ] 294/365: sqlformat, rand, tokio, u...
|     Building [===================>     ] 295/365: sqlformat, rand, tokio, b...
|     Building [===================>     ] 296/365: sqlformat, tokio, brotli-...
|     Building [===================>     ] 297/365: sqlformat, tokio, brotli-...
|     Building [===================>     ] 298/365: sqlformat, tokio, brotli-...
|     Building [===================>     ] 299/365: sqlformat, tokio, brotli-...
|     Building [===================>     ] 300/365: sqlformat, tokio, brotli-...
|     Building [===================>     ] 301/365: tokio, brotli-sys(build),...
|     Building [===================>     ] 302/365: tokio, zstd-sys(build)
|     Building [===================>     ] 303/365: zstd-sys(build)
| error: build failed
| WARNING: /workdir/build/tmp/work/cortexa8hf-neon-poky-linux-gnueabi/actix-hello-world/0.1.0.AUTOINC+cdd49fb843-r0/temp/run.do_compile.79687:1 exit 101 from '"cargo" build -v --target arm-poky-linux-gnueabi --release --manifest-path=/workdir/build/tmp/work/cortexa8hf-neon-poky-linux-gnueabi/actix-hello-world/0.1.0.AUTOINC+cdd49fb843-r0/git//Cargo.toml "$@"'
| 
```

IniterWorker, 2021-08-21:
Ah yeah, I made some progress.

I use the vendored feature for OpenSSL. But, still have an issue with the compilation. I don't know what's happening seems related to pkg-config trying to build x86.

- [https://docs.rs/openssl/0.10.36/openssl/#vendored](https://docs.rs/openssl/0.10.36/openssl/#vendored)
 
```toml
[dependencies]
actix-web = "4.0.0-beta.8"
openssl = { version = "0.10", features = ["vendored"] }
sqlx = { version = "0.5.5", features = [ "runtime-actix-native-tls" ] }
```

```
|     Building [==================>      ] 288/364: tokio, rand, encoding_rs,...
|      Running `/workdir/build/tmp/work/cortexa8hf-neon-poky-linux-gnueabi/actix-hello-world/0.1.0.AUTOINC+8fb3aeba16-r0/build/target/release/build/openssl-sys-6e2e02f83a532717/build-script-main`
|      Running `/workdir/build/tmp/work/cortexa8hf-neon-poky-linux-gnueabi/actix-hello-world/0.1.0.AUTOINC+8fb3aeba16-r0/build/target/release/build/openssl-sys-6e2e02f83a532717/build-script-main`
|     Building [==================>      ] 289/364: tokio, encoding_rs, regex...
| error: failed to run custom build command for `openssl-sys v0.9.66`
| 
| Caused by:
|   process didn't exit successfully: `/workdir/build/tmp/work/cortexa8hf-neon-poky-linux-gnueabi/actix-hello-world/0.1.0.AUTOINC+8fb3aeba16-r0/build/target/release/build/openssl-sys-6e2e02f83a532717/build-script-main` (exit status: 101)
|   --- stdout
|   cargo:rustc-cfg=const_fn
|   cargo:rerun-if-env-changed=X86_64_LINUX_OPENSSL_NO_VENDOR
|   X86_64_LINUX_OPENSSL_NO_VENDOR unset
|   cargo:rerun-if-env-changed=OPENSSL_NO_VENDOR
|   OPENSSL_NO_VENDOR unset
| 
|   --- stderr
|   thread 'main' panicked at 'don't know how to configure OpenSSL for x86_64-linux', /workdir/build/tmp/work/cortexa8hf-neon-poky-linux-gnueabi/actix-hello-world/0.1.0.AUTOINC+8fb3aeba16-r0/cargo_home/bitbake/openssl-src-111.15.0+1.1.1k/src/lib.rs:275:18
|   stack backtrace:
|      0: rust_begin_unwind
|      1: std::panicking::begin_panic_fmt
|      2: openssl_src::Build::build
|      3: build_script_main::find_vendored::get_openssl
|      4: build_script_main::find_openssl
|      5: build_script_main::main
|      6: core::ops::function::FnOnce::call_once
|   note: Some details are omitted, run with `RUST_BACKTRACE=full` for a verbose backtrace.
| warning: build failed, waiting for other jobs to finish...
|     Building [==================>      ] 290/364: tokio, encoding_rs, regex...
|     Building [==================>      ] 291/364: tokio, encoding_rs, regex...
|     Building [===================>     ] 292/364: tokio, encoding_rs, regex...
|     Building [===================>     ] 293/364: tokio, regex-syntax, nom,...
|     Building [===================>     ] 294/364: tokio, regex-syntax, nom,...
|     Building [===================>     ] 295/364: tokio, regex-syntax, sqlf...
|     Building [===================>     ] 296/364: tokio, regex-syntax, sqlf...
|     Building [===================>     ] 297/364: tokio, sqlformat, regex, ...
|     Building [===================>     ] 298/364: tokio, sqlformat, regex, ...
|     Building [===================>     ] 299/364: tokio, sqlformat, regex, ...
|     Building [===================>     ] 300/364: tokio, regex, brotli-sys(...
|     Building [===================>     ] 301/364: tokio, brotli-sys(build),...
|     Building [===================>     ] 302/364: tokio, zstd-sys(build)
|     Building [===================>     ] 303/364: zstd-sys(build)
| error: build failed
| WARNING: /workdir/build/tmp/work/cortexa8hf-neon-poky-linux-gnueabi/actix-hello-world/0.1.0.AUTOINC+8fb3aeba16-r0/temp/run.do_compile.102824:1 exit 101 from '"cargo" build -v --target arm-poky-linux-gnueabi --release --manifest-path=/workdir/build/tmp/work/cortexa8hf-neon-poky-linux-gnueabi/actix-hello-world/0.1.0.AUTOINC+8fb3aeba16-r0/git//Cargo.toml "$@"'
```

IniterWorker, 2023-03-28:
Nope, issue seems to come from an unsupported target.
**See:** https://github.com/alexcrichton/openssl-sys/issues/4#issuecomment-571189377

Awarty, 2023-03-30:
Thanks for your quick answer. I have kept investigating my problem and have found a solution that works for me. I thought I would share it here if anyone else has the same issue. I will give some context on what I was trying to do for a more clear description.

I tried to build a console image for a Raspberry Pi Compute Module 4 device where I also added a rust program by compiling it from a remote git repository. Using meta-rust I could compile my rust program hosted on GitHub with the .bb file generated from running 'cargo bitbake'. My problem was when I tried to compile a rust program that uses OpenSSL I got an error saying failed to build 'openssl-sys'. The output was very similar to @IniterWorker , different target system.

<details>
<summary>output</summary>

```
| error: failed to run custom build command for openssl-sys v0.9.83
|
| Caused by:
|   process didn't exit successfully: /home/jacob/dev/gaia/yocto/rust-yocto/build/tmp/work/cortexa72-poky-linux/gpt-hello-world-vendor/1.0.AUTOINC+df79f2cd79-r0/build/target/release/build/openssl-sys-b44ce5b10c5aa4ef/build-script-main (exit status: 101)
|   --- stdout
|   cargo:rustc-cfg=const_fn
|   cargo:rerun-if-env-changed=AARCH64_POKY_LINUX_OPENSSL_NO_VENDOR
|   AARCH64_POKY_LINUX_OPENSSL_NO_VENDOR unset
|   cargo:rerun-if-env-changed=OPENSSL_NO_VENDOR
|   OPENSSL_NO_VENDOR unset
|
|   --- stderr
|   warning: target json file contains unused fields: has-elf-tls
|
|   warning: target json file contains unused fields: has-elf-tls
|
|   thread 'main' panicked at 'don't know how to configure OpenSSL for aarch64-poky-linux', /home/jacob/dev/gaia/yocto/rust-yocto/build/tmp/work/cortexa72-poky-linux/gpt-hello-world-vendor/1.0.AUTOINC+df79f2cd79-r0/cargo_home/bitbake/openssl-src-111.25.2+1.1.1t/src/lib.rs:305:18
|   stack backtrace:
|      0: rust_begin_unwind
|      1: core::panicking::panic_fmt
|      2: openssl_src::Build::build
|      3: build_script_main::find_vendored::get_openssl
|      4: build_script_main::find_openssl
|      5: build_script_main::main
|      6: core::ops::function::FnOnce::call_once
|   note: Some details are omitted, run with RUST_BACKTRACE=full for a verbose backtrace.
```
</details>

You linked me to an issue about unsupported targets. In my case, the target is 'aarch64-poky-linux' and should work. After looking around on a lot of different threads I found that I didn't need to specify 'vendored' for cross-compiling rust using Yocto. It can use the OpenSSL library generated by Yocto and don't need to use the crate 'openssl-src' (which is used when 'vendored' is specified). So in my case, it was easier to just remove 'vendored' from my features. That did not fix everything you also need to add:
DEPENDS = "openssl pkgconfig-native"
RDEPENDS_${PN} += "openssl"
This makes sure that OpenSSL is built and added for this program using 'pkgconfig-native' for finding the built OpenSSL library in Yocto.

In summary, I fixied my issue with OpenSSL by removing 'vendored' in my Cargo.toml file and by adding 'openssl' and 'pkgconfig-native' to the .bb file. If anyone has the same issue this will hopefully help you out as it did me.

srwalter, 2023-03-30:
@Awarty Thanks for documenting that!  I agree that in your case, that's the correct solution