meta-rust/meta-rust #362: [dunfell] ERROR: ParseError at /.../rust_security_flags.inc:2: unparsed line: 'SECURITY_CFLAGS:pn-rust-native = "${SECURITY_NO_PIE_CFLAGS}"'

jwinarske, 2021-08-19:
## Version(s) of meta-rust
test-dunfell

## Version(s) of poky and/or oe-core
dunfell

## Expected result
compile

## Actual result
ERROR: ParseError at /.../meta-rust/conf/distro/include/rust_security_flags.inc:2: unparsed line: 'SECURITY_CFLAGS:pn-rust-native = "${SECURITY_NO_PIE_CFLAGS}"'

## Steps to reproduce
master worked fine about a month ago for dunfell

jwinarske, 2021-08-20:
Dunfell was broken with https://github.com/meta-rust/meta-rust/commit/bf6705fb7f41b16c3277cbf84af0b9757dcd3a01

kraj, 2021-08-20:
Are you on latest dunfell ?

jwinarske, 2021-08-20:
@kraj Dunfell as of Jul 05 2021 (QC release)

kraj, 2021-08-20:
> @kraj Dunfell as of Jul 05 2021 (QC release)

Please use the latest as of today and report back

jwinarske, 2021-08-20:
What in particular is required from tip of tree?

kraj, 2021-08-20:
https://git.yoctoproject.org/cgit/cgit.cgi/poky/commit/?h=dunfell&id=8f65434827840a71966798a35b090ce5e31ae3c0
https://git.yoctoproject.org/cgit/cgit.cgi/poky/commit/?h=dunfell&id=9c2186108e2e8fcf5837c0ffe9dbdeeaea4b7398

are needed

jwinarske, 2021-08-20:
@kraj Awesome thanks!  I will patch in the near term.

jwinarske, 2021-08-21:
With both of those patches, I'm still seeing the parse error.

```
$ git apply --verbose poky_variable_expansion_regex-9c2186108e2e8fcf5837c0ffe9dbdeeaea4b7398.patch
Checking patch bitbake/lib/bb/data_smart.py...
Checking patch bitbake/lib/bb/tests/codeparser.py...
Applied patch bitbake/lib/bb/data_smart.py cleanly.
Applied patch bitbake/lib/bb/tests/codeparser.py cleanly.
$ git apply --verbose poky_bbhandler-8f65434827840a71966798a35b090ce5e31ae3c0.patch
Checking patch bitbake/lib/bb/parse/parse_py/BBHandler.py...
Applied patch bitbake/lib/bb/parse/parse_py/BBHandler.py cleanly.
```

...

```
ERROR: ParseError at /.../meta-rust/conf/distro/include/rust_security_flags.inc:2: unparsed line: 'SECURITY_CFLAGS:pn-rust-native = "${SECURITY_NO_PIE_CFLAGS```
}"'
```

jwinarske, 2021-09-25:
I think a branch/tag scheme for meta-rust would be great, as a minimum on LTS releases.  Tag for the release version.  Maybe even a build matrix to validate.

russell-islam, 2022-02-15:
I am also hitting this issue. Is this still a bug? Any workaround?

jwinarske, 2022-02-17:
@russell-islam this is what I am using for this issue:
```
git reset --hard 1b59fd45906082c978d0a0a6e4e51a0ea4aa32c7
```