meta-rust/meta-rust #375: Not able to build with mixed version in depedent crate

russell-islam, 2022-02-08:
## Version(s) of meta-rust
master
## Version(s) of poky and/or oe-core
dunfel
## Expected result
Should be able to use different source for depedent crate
## Actual result
Not able to build
## Steps to reproduce
1.  Main project has a dependency of a crate named ABC" sourced from git repo.
2. Main project has a dependent package ABCD sourced from a git repo
3. ABCD has a depedent package ABC that is referenced from crates.io
meta rust is not able to build the main package. Vanilla cargo in ubuntu can build the above scenario easily.

russell-islam, 2022-02-08:
Any suggestions??

russell-islam, 2022-02-08:
Detailed error: 
muislam@muislam-desktop:~/yocto/poky/build$ bitbake cloud-hypervisor 
Loading cache: 100% |#######################################################################################################################################################################| Time: 0:00:00
Loaded 2364 entries from dependency cache.
Removing 1 recipes from the core2-64 sysroot: 100% |########################################################################################################################################| Time: 0:00:00
NOTE: Resolving any missing task queue dependencies

Build Configuration:
BB_VERSION           = "1.46.0"
BUILD_SYS            = "x86_64-linux"
NATIVELSBSTRING      = "universal"
TARGET_SYS           = "x86_64-poky-linux"
MACHINE              = "qemux86-64"
DISTRO               = "poky"
DISTRO_VERSION       = "3.1.14"
TUNE_FEATURES        = "m64 core2"
TARGET_FPU           = ""
meta                 
meta-poky            = "dunfell:bba323389749ec3e306509f8fb12649f031be152"
meta-rust            = "master:d4384db8c3f0aa01732627f92fc3d446359bf743"
meta-cloud-hypervisor = "master:fc23beefa14f6646d05a5a7e1c8df3d6122163a2"
meta-oe              = "dunfell:ab9fca485e13f6f2f9761e1d2810f87c2e4f060a"

Initialising tasks: 100% |##################################################################################################################################################################| Time: 0:00:00
Sstate summary: Wanted 7 Found 2 Missed 5 Current 153 (28% match, 96% complete)
NOTE: Executing Tasks
ERROR: cloud-hypervisor-20.2.0-r0 do_compile: Execution of '/home/muislam/yocto/poky/build/tmp/work/core2-64-poky-linux/cloud-hypervisor/20.2.0-r0/temp/run.do_compile.35737' failed with exit code 101
ERROR: Logfile of failure stored in: /home/muislam/yocto/poky/build/tmp/work/core2-64-poky-linux/cloud-hypervisor/20.2.0-r0/temp/log.do_compile.35737
Log data follows:
| DEBUG: Executing shell function do_compile
| NOTE: cargo = /home/muislam/yocto/poky/build/tmp/work/core2-64-poky-linux/cloud-hypervisor/20.2.0-r0/recipe-sysroot-native/usr/bin/cargo
| NOTE: rustc =
| NOTE: cargo build -v --target x86_64-poky-linux --release --manifest-path=/home/muislam/yocto/poky/build/tmp/work/core2-64-poky-linux/cloud-hypervisor/20.2.0-r0/git//Cargo.toml
| warning: path override for crate `versionize_derive` has altered the original list of
| dependencies; the dependency on `proc-macro2` was either added or
| modified to not match the previously resolved version
| 
| This is currently allowed but is known to produce buggy behavior with spurious
| recompiles and changes to the crate graph. Path overrides unfortunately were
| never intended to support this feature, so for now this message is just a
| warning. In the future, however, this message will become a hard error.
| 
| To change the dependency graph via an override it's recommended to use the
| `[patch]` feature of Cargo instead of the path override feature. This is
| documented online at the url below for more information.
| 
| https://doc.rust-lang.org/cargo/reference/overriding-dependencies.html
| 
| warning: path override for crate `kvm-ioctls` has altered the original list of
| dependencies; the dependency on `kvm-bindings` was either added or
| modified to not match the previously resolved version
| 
| This is currently allowed but is known to produce buggy behavior with spurious
| recompiles and changes to the crate graph. Path overrides unfortunately were
| never intended to support this feature, so for now this message is just a
| warning. In the future, however, this message will become a hard error.
| 
| To change the dependency graph via an override it's recommended to use the
| `[patch]` feature of Cargo instead of the path override feature. This is
| documented online at the url below for more information.
| 
| https://doc.rust-lang.org/cargo/reference/overriding-dependencies.html
| 
| warning: path override for crate `vfio-bindings` has altered the original list of
| dependencies; the dependency on `vmm-sys-util` was either added or
| modified to not match the previously resolved version
| 
| This is currently allowed but is known to produce buggy behavior with spurious
| recompiles and changes to the crate graph. Path overrides unfortunately were
| never intended to support this feature, so for now this message is just a
| warning. In the future, however, this message will become a hard error.
| 
| To change the dependency graph via an override it's recommended to use the
| `[patch]` feature of Cargo instead of the path override feature. This is
| documented online at the url below for more information.
| 
| https://doc.rust-lang.org/cargo/reference/overriding-dependencies.html
| 
| error: failed to get `kvm-bindings` as a dependency of package `vfio-ioctls v0.1.0 (/home/muislam/yocto/poky/build/tmp/work/core2-64-poky-linux/cloud-hypervisor/20.2.0-r0/vfio-ioctls)`
|     ... which is depended on by `vmm v0.1.0 (/home/muislam/yocto/poky/build/tmp/work/core2-64-poky-linux/cloud-hypervisor/20.2.0-r0/git/vmm)`
|     ... which is depended on by `cloud-hypervisor v20.2.0 (/home/muislam/yocto/poky/build/tmp/work/core2-64-poky-linux/cloud-hypervisor/20.2.0-r0/git)`
| 
| Caused by:
|   found patches and a path override
| WARNING: exit code 101 from a shell command.
| ERROR: Execution of '/home/muislam/yocto/poky/build/tmp/work/core2-64-poky-linux/cloud-hypervisor/20.2.0-r0/temp/run.do_compile.35737' failed with exit code 101
ERROR: Task (/home/muislam/yocto/meta-cloud-hypervisor/recipes-virtualization/cloud-hypervisor/cloud-hypervisor_20.2.0.bb:do_compile) failed with exit code '1'
NOTE: Tasks Summary: Attempted 648 tasks of which 641 didn't need to be rerun and 1 failed.

Summary: 1 task failed:
  /home/muislam/yocto/meta-cloud-hypervisor/recipes-virtualization/cloud-hypervisor/cloud-hypervisor_20.2.0.bb:do_compile
Summary: There was 1 ERROR message shown, returning a non-zero exit code.
muislam@muislam-desktop:~/yocto/poky/build$

cardoe, 2022-02-08:
Unfortunately your summary of the issue isn't correct. Mixed versions of dependent crates are supported and have always worked since day one as that's extremely common in Rust land. In fact your detailed error message is pointing you to that here: https://doc.rust-lang.org/cargo/reference/overriding-dependencies.html Further we can see the crates with an issue appear here https://github.com/cloud-hypervisor/cloud-hypervisor/blob/9f61c97e2b8aa41f8aa20b6a3e1edb7e0a0404eb/Cargo.toml#L37-L40 which should be the version you're using. Given you've got #271 still opened for not handling git references as dependencies against this package, I'm inclined to believe its an example of that same error just with a newer version of Cargo being driven by the tooling. It's also possible in your local dev environment you're attempting to override the dependencies as documented in the Cargo manual and running into issues with that as well. Either way I do not believe this bug is functionally any different than #271 so I'll close this as a duplicate.

closed 2022-02-08 by cardoe