meta-rust/meta-rust #380: When will the official kirkstone support be released?

coveritytest, 2022-05-04:
## General question

Can you guys say, when the official kirkstone support will be released?

snke, 2022-05-19:
(Not an active member of this project or anything, I am just playing around with this)

As far as I can tell from #376 and #364 and the the [oe-core git](https://git.openembedded.org/openembedded-core/log/meta/recipes-devtools/rust?h=kirkstone), rust support is now on oe-core (starting with honistor even).

I don't know if this has been announced somewhere or more "official" info can be found.
Would be nice if somebody could post some in here.

jaskij, 2022-11-09:
`oe-core` indeed has Rust, but the ecosystem doesn't care for maintaining older versions, so it can be a pain.

So, I got it to build (didn't test yet), with Kirkstone. There were two things needed:

Update `LAYERSERIES_COMPAT`

```diff
diff --git a/conf/layer.conf b/conf/layer.conf
index 53b2368..48becaf 100644
--- a/conf/layer.conf
+++ b/conf/layer.conf
@@ -9,7 +9,7 @@ BBFILE_PATTERN_rust-layer := "^${LAYERDIR}/"
 BBFILE_PRIORITY_rust-layer = "7"
 
 LAYERDEPENDS_rust-layer = "core openembedded-layer"
-LAYERSERIES_COMPAT_rust-layer = "dunfell gatesgarth hardknott honister"
+LAYERSERIES_COMPAT_rust-layer = "dunfell gatesgarth hardknott honister kirkstone"
 
 # Override security flags
 require conf/distro/include/rust_security_flags.inc
```

And add `PREFERRED_VERSION` - I added it to my `distro.conf`, but `local.conf` will work as well.

```
PREFERRED_VERSION_cargo = "1.64.0"
PREFERRED_VERSION_libstd-rs = "1.64.0"
PREFERRED_VERSION_rust = "1.64.0"
PREFERRED_VERSION_rust-cross = "1.64.0"
PREFERRED_VERSION_rust-llvm = "1.64.0"
PREFERRED_VERSION_rust-native = "1.64.0"
PREFERRED_VERSION_rust-source = "1.64.0"
PREFERRED_VERSION_rust-snapshot = "1.64.0"
```

hommeabeil, 2023-04-28:
I test it a little bit and it seems to work with the PREFERED_VERSION set. However, the meta-rust must be include after `meta` in the `bblayers.conf` file. Otherwise the bbclass used will not work for other receipe, should be fixed by #414 . 

```
 BBLAYERS ?= " \
/poky/meta \
...

/meta-rust \
 "

```

PR #414 (2023-04-28): export RUST_TARGET_PATH
In the kirkstone branch and later rust is now available as a dependency for other packages. Among other thing the `python3-cryptography` use some rust binding. Thus it needs to run `rustc --print target-list`. If `RUST_TARGET_PATH` is not exported, then all the `rustc --print` invocation fail with: 

`Error loading target specification: Could not find specification for target "x86_64-linux". Run rustc --print target-list for a list of built-in targets`

This seems like those issue where the fix is to export the `RUST_TARGET_PATH`

https://github.com/rust-lang/cargo/issu [... cut]
closed 2023-04-28 by srwalter