meta-rust/meta-rust #396: Anyway to integrate cargo bitbake into recipe file to autogenerate dependencies?

cornelius-span, 2022-11-08:
Currently am unable to compile cargo bitbake within recipe file for automatic dependency listing using bitbake recipe file, does anyone have a good method for doing this or any tips on how that might be achieved? Currently trying to run cargo-native install cargo-bitbake inside of do_fetch_append() but getting dependency errors

srwalter, 2022-11-08:
That's not really how it's designed to work.  Can you say more about what you're trying to do?  Why doesn't the documented working of generating the .bb file with cargo-bitbake work?

cornelius-span, 2022-11-08:
It works well I think the problem is just in terms of automation. Instead of having to generate a new bb file each time you update your rust dependencies/code I was imagining automating the process so that the cargo bitbake command is called from within the yocto build system thus avoiding the need to manually run the command each time. Idk if that makes sense but thanks for the reply!

michalfita, 2023-07-27:
That's not actually how `bitbake` works. While anything is running it's too late to generate dependencies list to download them. Then you wouldn't have `cargo` in place to download, install and use `cargo bitbake` at `do_fetch()` stage of any recipe dependent on `meta-rust`.
And probably that's not how recipes in Yocto should work - the generics for development testing are allowed, but discouraged from use in production.