meta-rust/meta-rust #398: Code generation issues when using meta-rust on 32 bit ARM targets with no/limited vector instruction support

galmasi, 2022-11-30:
## Version(s) of meta-rust
Tested with versions building rust-1.54, but the related code in meta-rust is unchanged with today's master.

## Version(s) of poky and/or oe-core
As used by current versions of openbmc (but most likely not relevant)

## Expected result
arm7 32 bit targets should not generate code with `neon` instructions when the `neon` feature set is not enabled

## Actual result
libstd-rs is generated with `neon` instructions; any user applications built with the rust compiler are not.

## Steps to reproduce
* We reproduced this problem within openbmc
* created a `qemuarma7` target with arch-arm7a target
* created a "rust hello world" target with its own cargo.toml and a helloworld.rs file
* built all targets and ran the hello world application with qemu as well as an AST2600 EVB

## Analysis
* The AST2600 EVB has partial support for vector instructions, i.e. only the first 16 vector registers are implemented. This is the vfp4d16 target.
* Rust (64 bit) supports arm vfp4d16; rust (32 bit) does not (it's not listed as a legitimate target). 
* Rust (32 bit targets) can either have neon instructions *on* or *off*. 
* rust-common.inc allows `neon` to be added explicitly to the target, but does not disable it explicitly otherwise.
* the rust compiler behaves in an ambivalent manner. To wit, compare the rust flags when compiling just a hello world target binary vs. when compiling the rust library itself:

```
>> rustc --target arm-openbmc-linux-gnueabi --print cfg
debug_assertions      
target_arch="arm"    
target_endian="little" 
target_env="gnu"                      
target_family="unix"                  
target_os="linux"                     
target_pointer_width="32"            
target_vendor="unknown"                
unix                             
```

```
>> RUSTC_BOOTSTRAP=1 rustc --target arm-openbmc-linux-gnueabi --print cfg
debug_assertions                                                                                                                                                    
panic="unwind"                                                                                                                                                      
target_arch="arm"                                                                                                                                                   
target_endian="little"                                                                                                                                              
target_env="gnu"                                                                                                                                                    
target_family="unix"                                                                                                                                                
target_feature="aclass"                                                                                                                                             
target_feature="dsp"                                                                                                                                                
target_feature="neon"    <------------------------- trouble                                                                                                                                           
target_feature="v5te"                                                                                                                                               
target_feature="v6"                                                                                                                                                 
target_feature="v6k"                                                                                                                                                
target_feature="v6t2"                                                                                                                                               
target_feature="v7"                  
target_feature="vfp2"                                                                                                                                               
target_feature="vfp3"                                                                                                                                               
target_has_atomic="16"           
target_has_atomic="32"                                                            
target_has_atomic="64"                                                                                                                                              
target_has_atomic="8"             
target_has_atomic="ptr"                                                           
target_has_atomic_equal_alignment="16"                                                                                                                              
target_has_atomic_equal_alignment="32"                                            
target_has_atomic_equal_alignment="64"   
target_has_atomic_equal_alignment="8"                                                                                                                               
target_has_atomic_equal_alignment="ptr"                                                                                                                             
target_has_atomic_load_store="16"                                                                                                                                   
target_has_atomic_load_store="32"                                                                                                                                   
target_has_atomic_load_store="64"
target_has_atomic_load_store="8"                                                                                                                                    
target_has_atomic_load_store="ptr"                                                
target_os="linux"                                                                                                                                                   
target_pointer_width="32"         
target_thread_local                                                               
target_vendor="unknown"                                                                                                                                             
unix                         
```

Conclusion
==

We don't know whether this could be fixable in the rust compiler itself, and whether meta-rust should have to deal with this issue. But we'll try you guys first :) E.g. we could explicitly disable `neon` when the user does not want it. But we are open to other suggestions. 

```
--- a/poky/meta/recipes-devtools/rust/rust-common.inc
+++ b/poky/meta/recipes-devtools/rust/rust-common.inc
@@ -26,6 +26,8 @@ def llvm_features_from_tune(d):
 
     if 'neon' in feat:
         f.append("+neon")
+    else:
+        f.append("-neon")
 
```

jaskij, 2023-02-08:
Just saw this issue and, at least according to GCC, NEON shouldn't be used by default on ARMv7-A regardless of the support. To quote the GCC manual:

> If the selected floating-point hardware includes the NEON extension (e.g. -mfpu=neon), note that floating-point operations are not generated by GCC’s auto-vectorization pass unless -funsafe-math-optimizations is also specified. This is because NEON hardware does not fully implement the IEEE 754 standard for floating-point arithmetic (in particular denormal values are treated as zero), so the use of NEON instructions may lead to a loss of precision. 

I do not have a more authoritative source at hand though.

galmasi, 2023-02-08:
I read that as "agreement" that the neon instruction set should be disabled by default :) any chance this can go upstream?

galmasi, 2023-02-08:
Would you like me top open a PR on this topic?

jaskij, 2023-02-08:
@galmasi I'm not one of the maintainers, just a person also using `meta-rust`. But yes, I do agree with you that disabling it by default is probably the right choice.

I want to dig in a little and ask around how `rustc` actually handles this before proposing an actual solution.

Do note that, from what I remember of looking into it, at the very least stage 1 is built with an older compiler version (one minor lower than current, so for 1.66 it would be 1.65), which may exhibit different behavior. The older version binaries are downloaded in `rust-snapshot-M.mm.inc`.

jaskij, 2023-02-08:
Found the authoritative source:

> Additionally, NEON instructions always treats denormals as zero.

From [ARM Cortex-A Series Programmer's Guide for ARMv7-A](https://developer.arm.com/documentation/den0013/d/Introducing-NEON/NEON-architecture-overview/Data-types), at the very bottom.

jaskij, 2023-02-08:
Doing some more reading, it seems that LLVM will handle at least the rounding situation correctly.

galmasi, 2023-02-08:
I was wondering about the activity after 2 months of being ignored. We will test with more up-to-date versions of openbmc, but the fact is that a lot of the vendor specific code tie us to specific old versions of openbmc. In the end, if newer versions of llvm or rustc handle this correctly, so much the better.

cuviper, 2023-02-22:
Hi - I'm unfamiliar with how meta-rust works, but I was pointed here to see if I could help...

>     rustc --target arm-openbmc-linux-gnueabi

That's not a builtin target upstream -- how is it defined?

A lot of the builtin arm targets have explicitly disabled neon, e.g.:
https://github.com/rust-lang/rust/blob/b869e84e581612f4a30a4bca63bd9e90e9a17003/compiler/rustc_target/src/spec/armv7_unknown_linux_gnueabihf.rs#L15

cuviper, 2023-02-22:
@galmasi I think your suggestion was already applied in commit 20d0e6f5e23516bcda097dfb96349798f44b5cc7.

jaskij, 2023-02-22:
@cuviper the targets are custom targets, generated by [`cargo_common.bbclass`, starting in line 32](https://github.com/meta-rust/meta-rust/blob/1ed669c464a113cddba8222b419565273d3410f2/classes/cargo_common.bbclass#L32)