meta-rust/meta-rust #406: rustc-native: Can't get target-list

mhuangwm, 2023-03-01:
## Version(s) of meta-rust
[master](https://github.com/meta-rust/meta-rust/commit/838d3aab62fe8360ed046b2523fa311fa412c125)

## Version(s) of poky and/or oe-core
dunfell

## Expected result
running `rustc --print target-list` should provide a list of targets

## Actual result
running above command provides:
```
error: Error loading target specification: Could not find specification for target "x86_64-linux". Run `rustc --print target-list` for a list of built-in targets
```

## Steps to reproduce
1. Build rustc-native `bitbake rust-native`. Ensure work is not removed
2. from rustc work directory, find the to-be installed vesrion of `rustc`. This is usually in `image/<something>/x86_64-linux/rust-native/<more stuff>/recipe-sysroot-native/usr/bin/rustc`
3. Run the command  `rustc --print target-list` from OUTSIDE of the directory of where the binary is located.

Funnily enough, if I run it inside the directory where rustc is located, I don't get any errors.

I'm trying to get python3-cryptography version 36, 37, 39 running on our dunfell builds. These have a dependency on rust and somewhere in the setuptools, it's running `['rustc', '--print', ''-target-list"]`

mhuangwm, 2023-03-01:
top google search hinted at exporting `RUST_TARGET_PATH` to location of `rustlib` which helps resolve the issue.

microhobby, 2023-06-26:
@mhuangwm I was trying to add the `RUST_TARGET_PATH`, this appears to fix the native case, but now I'm getting the error message for the aarch64 target. Did you get any fixes for this?