meta-rust/meta-rust #408: Update to at least Rust 1.66.1, preferably 1.68.0

jaskij, 2023-03-14:
There's a list of changes to Cargo since 1.66.0 which suggest updating sooner rather than later.

Listing only the most relevant things:

**1.66.1**:
- fixes [CVE-2022-46176](https://github.com/advisories/GHSA-r5w3-xm58-jv6j)
**1.68.0**:
- fixes an issue with `cargo package` and `publish` not respecting a workspace `Cargo.lock`, not sure if it is used internally. See https://github.com/rust-lang/cargo/issues/11148
- introduces the new Sparse protocol for remote registries, which will speed up downloads - if not for Yocto builds, then for developers who want to use the same version when developing

be-neth, 2023-04-07:
I will try to do it myself and proposed a PR.

But before do we know what files needs to be change ? And what point need to be checked ?
Maybe for future rust update, we should create a wiki page to update Rust version.

jaskij, 2023-04-07:
I have an internal fork, which I'm sadly not at a liberty to share, but the changed files are all in `recipes-devtools`:

Files to add:

- `cargo/cargo-cross-canadian_1.68.0bb`
- `cargo/cargo_1.68.0,bb`
- `rust/libstd-rs_16.0.bb`
- `rust/rust-cross-canadian_16.8.0.bb`
- `rust/rust-cross_16.8.0.bb`
- `rust/rust-llvm_16.8.0.bb`
- `rust/rust-snapshot-16.8.0.inc`
- `rust/rust-source-16.8.0.inc`
- `rust/rust_16.8.0.bb`

Since `meta-rust` doesn't carry any patches to Rust itself, the changes are relatively straightforward - just update the variable and checksums.

be-neth, 2023-04-07:
Thanks for the hint.
Comments in recipe are very useful.

I think we need to update `rust-llvm` recipe for rust version 1.68.2 and rust stage0.json for stable version as some change also.

- rust-llvm_1.68.2.bb : Needs to change LLVM RELEASE to 15.0.6 as seen [here](https://github.com/rust-lang/llvm-project/blob/9ad24035fea8d309753f5e39e6eb53d1d0eb39ce/llvm/CMakeLists.txt).
- rust-snapshot-1.68.2.inc : According to [this](https://github.com/rust-lang/rust/blob/stable/src/stage0.json).
RS_VERSION = "1.67.1"
CARGO_VERSION = "1.67.1"

I will try all this, and if it works, submit a PR.

Thanks!

be-neth, 2023-04-11:
Hum,

I have a build failure on rust-native 1.68.2, complaining that std crate cannot be found.
I will check in depth what is going on, but I think there is some change starting with rust 1.67 regarding rustlib directory.

I have found this ref : https://github.com/rust-lang/rust/issues/109994 but this is not enough.

@jaskij : Have you seen this on your side ?

BKDaugherty, 2023-05-31:
@be-neth / @jaskij - I'm hoping to use 1.68 primarily for the reason listed above

> introduces the new Sparse protocol for remote registries, which will speed up downloads - if not for Yocto builds, then for developers who want to use the same Rust version locally

Do either of you have an in-progress PR that you're willing to share that I could work off of?

jaskij, 2023-05-31:
@BKDaugherty not at the moment, but I could probably work up something over the weekend.

@be-neth are you working on this, or should I take over?

jaskij, 2023-06-02:
Since 1.70 just dropped, I will try to include it as well.

I know I mentioned the sparse Cargo protocol, which is the default for crates.io as of 1.70, but I now worry something weird might fail somewhere in the fetcher. We'll see.

be-neth, 2023-06-02:
> @be-neth are you working on this, or should I take over?

Hi, You can take over, I have started to integrate rust 1.69 version but with the same issue mention previously, and no time to track the issue.

be-neth, 2023-06-22:
Hi,
Is there any update from someone to integrate a rust version 1.69 or 1.70 ?
Maybe we can start a draft PR to converge on the right solution ?

be-neth, 2023-07-25:
I think we can close this issue, as rust 1.71.0 has just been merged by : #419