meta-rust/meta-rust #410: Error building sdk: package is trying to install files into a shared area

tronical, 2023-04-14:
## Version(s) of meta-rust

Master branch at `816a860f2b7a40722a87aafc6b74e7a7bc4ad3eb`

## Version(s) of poky and/or oe-core

Kirkstone branch at `407c3e0237d947ec003bdd1af89a226121c7939c`

## Expected result

Creation of an SDK that includes cargo/rust.

## Actual result

```sh
ERROR: rust-cross-canadian-aarch64-1.66.0-r0 do_package_write_rpm_setscene: The recipe rust-cross-canadian-aarch64 is trying to install files into a shared area when those files already exist. Those files and their manifest location are:
  /home/simon/src/yocto/poky/build/tmp/deploy/rpm/x86_64_nativesdk/rust-cross-canadian-src-1.66.0-r0.x86_64_nativesdk.rpm
    (matched in manifest-x86_64_x86_64-nativesdk-rust-cross-canadian-x86-64.package_write_rpm)
Please verify which recipe should provide the above files.

The build has stopped, as continuing in this scenario WILL break things - if not now, possibly in the future (we've seen builds fail several months later). If the system knew how to recover from this automatically it would, however there are several different scenarios which can result in this and we don't know which one this is. It may be you have switched providers of something like virtual/kernel (e.g. from linux-yocto to linux-yocto-dev), in that case you need to execute the clean task for both recipes and it will resolve this error. It may be you changed DISTRO_FEATURES from systemd to udev or vice versa. Cleaning those recipes should again resolve this error, however switching DISTRO_FEATURES on an existing build directory is not supported - you should really clean out tmp and rebuild (reusing sstate should be safe). It could be the overlapping files detected are harmless in which case adding them to SSTATE_ALLOW_OVERLAP_FILES may be the correct solution. It could also be your build is including two different conflicting versions of things (e.g. bluez 4 and bluez 5 and the correct solution for that would be to resolve the conflict. If in doubt, please ask on the mailing list, sharing the error and filelist above.
ERROR: rust-cross-canadian-aarch64-1.66.0-r0 do_package_write_rpm_setscene: If the above message is too much, the simpler version is you're advised to wipe out tmp and rebuild (reusing sstate is fine). That will likely fix things in most (but not all) cases.
```

## Steps to reproduce

```sh
git clone https://github.com/slint-ui/meta-slint.git meta-slint
git clone -b kirkstone --single-branch git://git.yoctoproject.org/poky
git clone -b kirkstone --single-branch git://git.openembedded.org/meta-openembedded
git clone -b master --single-branch https://github.com/meta-rust/meta-rust.git
git clone -b kirkstone --single-branch https://github.com/kraj/meta-clang.git
cd poky
. oe-init-build-env
bitbake-layers add-layer ../../meta-openembedded/meta-oe
bitbake-layers add-layer ../../meta-rust
bitbake-layers add-layer ../../meta-clang
bitbake-layers add-layer ../../meta-slint
echo 'PREFERRED_PROVIDER_virtual/kernel = "linux-dummy"' >> conf/local.conf
echo 'MACHINE = "qemuarm64"' >> conf/local.conf
echo 'TOOLCHAIN_HOST_TASK:append = " packagegroup-rust-cross-canadian-${MACHINE} "' >> conf/local.conf
bitbake core-image-weston -c populate_sdk
```

BedRockJie, 2023-05-14:
Thanks for the program，Wouldn't it be better to use a switch for a PR.

tronical, 2023-05-14:
I’m not sure I understand. What’s a switch for a PR?:)