meta-rust/meta-rust #420: Building 1.71.0 on Honister misses `-lzstd` and symbols are unresolved

michalfita, 2023-07-27:
## Version(s) of meta-rust

e7b5acb2a87a5bf1544c7f9375bb79668745a8e0

## Version(s) of poky and/or oe-core

PetaLinux 2022.2 / Honister

## Expected result

I wanted to build 1.71.0 with this layer replacing core's original recipes. Managed to get this working by putting it on top of `bblayers.conf` and adding `BBMASK` for existing recipes. I expected this to build the compiler for use with my recipes, as I have code requiring minimum 1.66.0

## Actual result

One of the linking attempts doesn't seem to add `-lzstd` even if symbols are used:
```
|   = note: LC_ALL="C" PATH="/home/user/project/build/tmp/work/x86_64-linux/rust-native/1.71.0-r0/rustc-1.71.0-src/build/x86_64-unknown-linux-gnu/stage1/lib/rustlib/x86_64-unknown-linux-gnu/bin:/home/user/project/components/yocto/layers/core/scripts/native-intercept:/home/user/project/components/yocto/layers/core/scripts:/home/user/project/build/tmp/work/x86_64-linux/rust-native/1.71.0-r0/recipe-sysroot-native/usr/bin/x86_64-linux:/home/user/project/build/tmp/work/x86_64-linux/rust-native/1.71.0-r0/recipe-sysroot-native/usr/bin:/home/user/project/build/tmp/work/x86_64-linux/rust-native/1.71.0-r0/recipe-sysroot-native/usr/sbin:/home/user/project/build/tmp/work/x86_64-linux/rust-native/1.71.0-r0/recipe-sysroot-native/usr/bin:/home/user/project/build/tmp/work/x86_64-linux/rust-native/1.71.0-r0/recipe-sysroot-native/sbin:/home/user/project/build/tmp/work/x86_64-linux/rust-native/1.71.0-r0/recipe-sysroot-native/bin:/home/user/project/components/yocto/layers/core/bitbake/bin:/home/user/project/build/tmp/hosttools" VSLANG="1033" "/home/user/project/build/tmp/work/x86_64-linux/rust-native/1.71.0-r0/wrapper/target-rust-cc" "/tmp/rustcQNkyeO/symbols.o" "/home/user/project/build/tmp/work/x86_64-linux/rust-native/1.71.0-r0/rustc-1.71.0-src/build/x86_64-unknown-linux-gnu/stage1-rustc/x86_64-linux/release/deps/rustc_main-a37f09723ee10275.rustc_main.6b68437c8bf693ae-cgu.0.rcgu.o" "/home/user/project/build/tmp/work/x86_64-linux/rust-native/1.71.0-r0/rustc-1.71.0-src/build/x86_64-unknown-linux-gnu/stage1-rustc/x86_64-linux/release/deps/rustc_main-a37f09723ee10275.rustc_main.6b68437c8bf693ae-cgu.1.rcgu.o" "/home/user/project/build/tmp/work/x86_64-linux/rust-native/1.71.0-r0/rustc-1.71.0-src/build/x86_64-unknown-linux-gnu/stage1-rustc/x86_64-linux/release/deps/rustc_main-a37f09723ee10275.rustc_main.6b68437c8bf693ae-cgu.2.rcgu.o" "-Wl,--as-needed" "-L" "/home/user/project/build/tmp/work/x86_64-linux/rust-native/1.71.0-r0/rustc-1.71.0-src/build/x86_64-unknown-linux-gnu/stage1-rustc/x86_64-linux/release/deps" "-L" "/home/user/project/build/tmp/work/x86_64-linux/rust-native/1.71.0-r0/rustc-1.71.0-src/build/x86_64-unknown-linux-gnu/stage1-rustc/release/deps" "-L" "/home/user/project/build/tmp/work/x86_64-linux/rust-native/1.71.0-r0/rustc-1.71.0-src/build/x86_64-unknown-linux-gnu/stage1-rustc/x86_64-linux/release/build/psm-17a4d2dc15f6a915/out" "-L" "/home/user/project/build/tmp/work/x86_64-linux/rust-native/1.71.0-r0/rustc-1.71.0-src/build/x86_64-unknown-linux-gnu/stage1-rustc/x86_64-linux/release/build/rustc_llvm-c722294965ce521b/out" "-L" "/home/user/project/build/tmp/work/x86_64-linux/rust-native/1.71.0-r0/recipe-sysroot-native/usr/lib/llvm-rust/lib" "-L" "/home/user/project/build/tmp/work/x86_64-linux/rust-llvm-native/1.71.0-r0/recipe-sysroot-native/usr/lib" "-L" "/home/user/project/build/tmp/work/x86_64-linux/rust-llvm-native/1.71.0-r0/recipe-sysroot-native/lib" "-L" "/home/user/project/build/tmp/work/x86_64-linux/rust-native/1.71.0-r0/rustc-1.71.0-src/build/x86_64-unknown-linux-gnu/stage1/lib/rustlib/x86_64-linux/lib" "-L" "/home/user/project/build/tmp/work/x86_64-linux/rust-native/1.71.0-r0/rustc-1.71.0-src/build/x86_64-unknown-linux-gnu/stage1-rustc/x86_64-linux/release/deps" "-Wl,-Bdynamic" "-lrustc_driver-c711a4b20a07db32" "-L" "/home/user/project/build/tmp/work/x86_64-linux/rust-native/1.71.0-r0/rustc-1.71.0-src/build/x86_64-unknown-linux-gnu/stage1/lib/rustlib/x86_64-linux/lib" "-lstd-5f0f2ca95fa72397" "-Wl,-Bstatic" "/home/user/project/build/tmp/work/x86_64-linux/rust-native/1.71.0-r0/rustc-1.71.0-src/build/x86_64-unknown-linux-gnu/stage1/lib/rustlib/x86_64-linux/lib/libcompiler_builtins-a02a69f41c6f2690.rlib" "-Wl,-Bdynamic" "-lstdc++" "-ldl" "-lgcc_s" "-lutil" "-lrt" "-lpthread" "-lm" "-ldl" "-lc" "-Wl,--eh-frame-hdr" "-Wl,-z,noexecstack" "-L" "/home/user/project/build/tmp/work/x86_64-linux/rust-native/1.71.0-r0/rustc-1.71.0-src/build/x86_64-unknown-linux-gnu/stage1/lib/rustlib/x86_64-linux/lib" "-o" "/home/user/project/build/tmp/work/x86_64-linux/rust-native/1.71.0-r0/rustc-1.71.0-src/build/x86_64-unknown-linux-gnu/stage1-rustc/x86_64-linux/release/deps/rustc_main-a37f09723ee10275" "-Wl,--gc-sections" "-pie" "-Wl,-O1" "-nodefaultlibs" "-Wl,-z,origin" "-Wl,-rpath,$ORIGIN/../lib"
|   = note: /home/user/project/build/tmp/hosttools/ld: /home/user/project/build/tmp/work/x86_64-linux/rust-native/1.71.0-r0/rustc-1.71.0-src/build/x86_64-unknown-linux-gnu/stage1-rustc/x86_64-linux/release/deps/librustc_driver-c711a4b20a07db32.so: undefined reference to `ZSTD_isError'
|           /home/user/project/build/tmp/hosttools/ld: /home/user/project/build/tmp/work/x86_64-linux/rust-native/1.71.0-r0/rustc-1.71.0-src/build/x86_64-unknown-linux-gnu/stage1-rustc/x86_64-linux/release/deps/librustc_driver-c711a4b20a07db32.so: undefined reference to `ZSTD_getErrorName'
|           /home/user/project/build/tmp/hosttools/ld: /home/user/project/build/tmp/work/x86_64-linux/rust-native/1.71.0-r0/rustc-1.71.0-src/build/x86_64-unknown-linux-gnu/stage1-rustc/x86_64-linux/release/deps/librustc_driver-c711a4b20a07db32.so: undefined reference to `ZSTD_compressBound'
|           /home/user/project/build/tmp/hosttools/ld: /home/user/project/build/tmp/work/x86_64-linux/rust-native/1.71.0-r0/rustc-1.71.0-src/build/x86_64-unknown-linux-gnu/stage1-rustc/x86_64-linux/release/deps/librustc_driver-c711a4b20a07db32.so: undefined reference to `ZSTD_decompress'
|           /home/user/project/build/tmp/hosttools/ld: /home/user/project/build/tmp/work/x86_64-linux/rust-native/1.71.0-r0/rustc-1.71.0-src/build/x86_64-unknown-linux-gnu/stage1-rustc/x86_64-linux/release/deps/librustc_driver-c711a4b20a07db32.so: undefined reference to `ZSTD_compress'
|           collect2: error: ld returned 1 exit status
| 
|   = note: some `extern` functions couldn't be found; some native libraries may need to be installed or have their path specified
|   = note: use the `-l` flag to specify native libraries to link
|   = note: use the `cargo:rustc-link-lib` directive to specify the native libraries to link with Cargo (see https://doc.rust-lang.org/cargo/reference/build-scripts.html#cargorustc-link-libkindname)
| 
| 
| Did not run successfully: exit status: 1
```

## Steps to reproduce

```sh
petalinux-build -c rust-native
```
that actually runs `bitbake rust-native`

be-neth, 2023-07-27:
Hi,

I confirm the same issue with minimal Yocto environment on Yocto version kirkstone.

I am trying to understand the error, because I have another Yocto (same version) with same meta-rust integration which do not trigger this issue. (And log are pretty same between both).

philippderdiedas, 2023-07-27:
i added  ` -DLLVM_ENABLE_ZSTD=OFF \` to EXTRA_OECMAKE in meta-rust/recipes-devtools/rust/rust-llvm.inc based on [openembedded-core](https://git.openembedded.org/openembedded-core/tree/meta/recipes-devtools/rust/rust-llvm_1.70.0.bb#n39)

ColinFinck, 2023-07-27:
@philippderdiedas Thank you for quickly researching this!

Could you submit a PR adding this line? (and possibly other important lines that we currently miss from the openembedded-core version)
I would then do my best to review it tomorrow.