meta-rust/meta-rust #425: Building cargo 1.72 on Kirkstone fails: could not compile `proc-macro2` (lib)

erebks, 2023-08-31:
## Version(s) of meta-rust
`8af044c0edabf1b7e1aac4762af677e058900f92`

## Version(s) of poky and/or oe-core
poky: `f53ab3a2ff206a130cdc843839dd0ea5ec4ad02f`
oe-core: `9e539aaed608a03b1128a7361426703d8f9ba705`

## Expected result
Running `bitbake cargo` builds cargo (and all dependencies)

## Actual result
Build fails with the following error:

```
error: could not compile `proc-macro2` (lib) due to previous error; 2 warnings emitted

Caused by:
  process didn't exit successfully: `rustc --crate-name proc_macro2 --edition=2018 /home/user/yocto/build/tmp/work/armv8a-poky-linux/cargo/1.72.0-r0/rustc-1.72.0-src/vendor/proc-macro2-1.0.60/src/lib.rs --error-format=json --json=diagnostic-rendered-ansi,artifacts,future-incompat --crate-type lib --emit=dep-info,metadata,link -C opt-level=3 -C embed-bitcode=no --cfg 'feature="proc-macro"' -C metadata=6a2107cbb96f0acc -C extra-filename=-6a2107cbb96f0acc --out-dir /home/user/yocto/build/tmp/work/armv8a-poky-linux/cargo/1.72.0-r0/build/target/aarch64-poky-linux/release/deps --target aarch64-poky-linux -C linker=/home/user/yocto/build/tmp/work/armv8a-poky-linux/cargo/1.72.0-r0/wrapper/target-rust-ccld -L dependency=/home/user/yocto/build/tmp/work/armv8a-poky-linux/cargo/1.72.0-r0/build/target/aarch64-poky-linux/release/deps -L dependency=/home/user/yocto/build/tmp/work/armv8a-poky-linux/cargo/1.72.0-r0/build/target/release/deps --extern unicode_ident=/home/user/yocto/build/tmp/work/armv8a-poky-linux/cargo/1.72.0-r0/build/target/aarch64-poky-linux/release/deps/libunicode_ident-e239b614f7e8ae95.rmeta --cap-lints allow -L /home/user/yocto/build/tmp/work/armv8a-poky-linux/cargo/1.72.0-r0/recipe-sysroot/usr/lib/rust --remap-path-prefix=/home/user/yocto/build/tmp/work/armv8a-poky-linux/cargo/1.72.0-r0=/usr/src/debug/cargo/1.72.0-r0 --cfg wrap_proc_macro` (exit status: 1)
```

## Steps to reproduce
* Add `meta-rust` to existing kirkstone setup. 
* Set `RUSTVERSION = "1.72%"` in local.conf so it uses the newest version provided by `meta-rust`.

As a workaround I've reverted back to version 1.71 by setting `RUSTVERSION = "1.71%"`, here the build succeeds.

DAMEK86, 2023-10-06:
hey, we had the same issue since 1.71  
This can be fixed by changing `S` of libstd to library/sysroot

many thx to @theswiftfox. She already created a fix but isn't upstreamed until know :)

try to add a bbappend file in your layer. this should fix your issue until it's upstreamed
```
diff --git a/meta-rust/recipes-devtools/rust/libstd-rs_%.bbappend b/meta-rust/recipes-devtools/rust/libstd-rs_%.bbappend
new file mode 100644
index 0000000..06c7054
--- /dev/null
+++ b/meta-rust/recipes-devtools/rust/libstd-rs_%.bbappend
@@ -0,0 +1 @@
+S = "${RUSTSRC}/library/sysroot"
--

jaskij, 2024-04-10:
Just tested the workaround with 1.75, and it seems to work. Granted, my build is from x86-64 to x86-64, so there may have been some contamination Yocto didn't catch.

FelixSchladt, 2025-06-20:
I am experiencing the same issue, though if I try to apply the patch suggested it seems cargo cannot find any external crates used in my project:

```sh
[...]
| error[E0463]: can't find crate for `rand`
|   --> acsctl/src/trunking/trunk/mod.rs:19:5
|    |
| 19 | use rand;
|    |     ^^^^ can't find crate
| 
| error[E0463]: can't find crate for `nix`
|  --> acsctl/src/trunking/tun.rs:9:5
|   |
| 9 | use nix::fcntl::{F_SETFL, OFlag, fcntl};
|   |     ^^^ can't find crate
| 
| error[E0463]: can't find crate for `nix`
|   --> acsctl/src/trunking/tun.rs:10:5
|    |
| 10 | use nix::ioctl_write_int;
|    |     ^^^ can't find crate
| 
[...]
```

Does anyone know how to solve this issue? It seems my recipe is incapable of finding any crates after i applied the patch.
 Is this patch still valid? 
 I am running on version rustc version 1.82.0.

FelixSchladt, 2025-06-23:
> Hey [@FelixSchladt](https://github.com/FelixSchladt), IIRC the patch given fixed our problem back then.
> 
> One thing to check that comes to my mind: Are you sure that those crates are in the `SRC_URI`s? https://docs.yoctoproject.org/ref-manual/classes.html#cargo

Thanks for the tip, it was indeed some missing `SRC_URI` and a leftover yocto cache...

I can confirm the patch still works with Rust 1.82. Why is this not upstreamed yet?

commit 5807084b (2025-06-23): Fix "could not compile proc-macro2"

Fixes #425
PR #466 (2025-06-24): Fix "could not compile proc-macro2"
Fixes #425