meta-rust/meta-rust #427: rust-native fails in do_compile with poky mickeldore

mischief, 2023-09-11:
it seems there is an issue with the templating of the cargo cross config.

my ultimate goal is to be able to build a statically linked rust binary with TCLIBC="musl", but i've gotten stuck trying to build rust itself in a default configuration.

## Version(s) of meta-rust
8af044c0edabf1b7e1aac4762af677e058900f92

## Version(s) of poky and/or oe-core
yoctoproject/poky@aa63b25cbe25d89ab07ca11ee72c17cab68df8de
openembedded/meta-openembedded@57c516973978ca2667339bf1aae7dc722b5bcac7

## Expected result
build rust-native

## Actual result
```
DEBUG: Executing shell function do_compile
COMPILE rust-native build --stage 2
Building bootstrap
running: /home/mischief/src/poky/build/tmp/work/x86_64-linux/rust-native/1.72.0-r0/rust-snapshot/bin/cargo build --manifest-path /home/mischief/src/poky/build/tmp/work/x86_64-linux/rust-native/1.72.0-r0/rustc-1.72.0-src/src/bootstrap/Cargo.toml --verbose --frozen
error: failed to parse manifest at `/home/mischief/src/poky/build/tmp/work/x86_64-linux/rust-native/1.72.0-r0/rustc-1.72.0-src/src/bootstrap/Cargo.toml`

Caused by:
  could not load Cargo configuration

Caused by:
  could not parse TOML configuration in `/home/mischief/src/poky/build/tmp/work/x86_64-linux/rust-native/1.72.0-r0/cargo_home/config`

Caused by:
  could not parse input as TOML

Caused by:
  TOML parse error at line 26, column 1
     |
  26 | [target.x86_64-unknown-linux-gnu]
     | ^
  invalid table header
  duplicate key `x86_64-unknown-linux-gnu` in table `target`
Traceback (most recent call last):
  File "/home/mischief/src/poky/build/tmp/work/x86_64-linux/rust-native/1.72.0-r0/rustc-1.72.0-src/src/bootstrap/bootstrap.py", line 1129, in <module>
    main()
  File "/home/mischief/src/poky/build/tmp/work/x86_64-linux/rust-native/1.72.0-r0/rustc-1.72.0-src/src/bootstrap/bootstrap.py", line 1113, in main
    bootstrap(args)
  File "/home/mischief/src/poky/build/tmp/work/x86_64-linux/rust-native/1.72.0-r0/rustc-1.72.0-src/src/bootstrap/bootstrap.py", line 1079, in bootstrap
    build.build_bootstrap()
  File "/home/mischief/src/poky/build/tmp/work/x86_64-linux/rust-native/1.72.0-r0/rustc-1.72.0-src/src/bootstrap/bootstrap.py", line 875, in build_bootstrap
    run(args, env=env, verbose=self.verbose, cwd=self.rust_root)
  File "/home/mischief/src/poky/build/tmp/work/x86_64-linux/rust-native/1.72.0-r0/rustc-1.72.0-src/src/bootstrap/bootstrap.py", line 184, in run
    raise RuntimeError(err)
RuntimeError: failed to run: /home/mischief/src/poky/build/tmp/work/x86_64-linux/rust-native/1.72.0-r0/rust-snapshot/bin/cargo build --manifest-path /home/mischief/src/poky/build/tmp/work/x86_64-linux/rust-native/1.72.0-r0/rustc-1.72.0-src/src/bootstrap/Cargo.toml --verbose --frozen
WARNING: /home/mischief/src/poky/build/tmp/work/x86_64-linux/rust-native/1.72.0-r0/temp/run.do_compile.677494:193 exit 1 from 'python3 src/bootstrap/bootstrap.py -j 32 "$@" --verbose'
WARNING: Backtrace (BB generated script): 
	#1: rust_runx, /home/mischief/src/poky/build/tmp/work/x86_64-linux/rust-native/1.72.0-r0/temp/run.do_compile.677494, line 193
	#2: do_compile, /home/mischief/src/poky/build/tmp/work/x86_64-linux/rust-native/1.72.0-r0/temp/run.do_compile.677494, line 157
	#3: main, /home/mischief/src/poky/build/tmp/work/x86_64-linux/rust-native/1.72.0-r0/temp/run.do_compile.677494, line 197

```

## Steps to reproduce
```
bitbake-layers add-layer /home/mischief/src/meta-openembedded/meta-oe /home/mischief/src/meta-rust
```
in build/conf/local.conf:
```
RUST_VERSION = "1.72%"
require conf/distro/include/rust_versions.inc
```

mrtzcspr, 2023-09-14:
I'm seeing this as well. 
The line `[target.x86_64-unknown-linux-gnu]` exists twice:
[image]

tronical, 2023-10-02:
This seems to be caused by `setup_cargo_environment()` in `rust.inc` appending the `[target...]` section once more after cargo_common.bbclass has already done it:

```
setup_cargo_environment () {
    # The first step is to build bootstrap and some early stage tools,
    # these are build for the same target as the snapshot, e.g.
    # x86_64-unknown-linux-gnu.
    # Later stages are build for the native target (i.e. target.x86_64-linux)
    cargo_common_do_configure

    printf '[target.%s]\n' "${SNAPSHOT_BUILD_SYS}" >> ${CARGO_HOME}/config
    printf "linker = '%s'\n" "${RUST_BUILD_CCLD}" >> ${CARGO_HOME}/config
}
```

I wonder why it breaks with mickeldore...