meta-rust/meta-rust #439: Support for nanbield ?

francoiscolas, 2024-01-11:
## Version(s) of meta-rust

Latest master 

## Version(s) of poky and/or oe-core

Latest master

## Expected result

Build for `nanbield` is working.

## Actual result

`meta-rust` doesn't support latest `poky` contrary to what the README.md says :

> The master branch supports the latest master of poky. When poky creates releases, we will create a branch with the same name as the poky release. This release branch should always work with that poky release. Note that these release branches will typically be less tested than the master branch.

## Steps to reproduce

1. Update `poky` to latest master.
2. Update `meta-rust` to latest master.
3. It doesn't build because `meta-rust` doesn't support `nanbield`.

rwmacleod, 2024-01-12:
Do you know that the rust toolchain was merged into poky/oe-core in honister.
http://layers.openembedded.org/layerindex/branch/honister/recipes/?q=rust

ColinFinck, 2024-01-25:
@rwmacleod That's true, but @francoiscolas is right: meta-rust should support nanbield and future Yocto releases, because OpenEmbedded has no interest in backporting the latest Rust version to all supported releases. See https://github.com/meta-rust/meta-rust/pull/364#issuecomment-1016643903

meta-rust has provided the latest Rust version for all Yocto releases in the past and should continue to do so.

rwmacleod, 2024-01-25:
@ColinFinck Ross is correct of course.

I tried to convince the Yocto  technical steering committee members that the only supported version of Rust is the 1.XX release and that we should be updating all supported branches. The response was that Rust updates add features and while they may have a humongous test suite to test for regressions the YP TSC wanted evidence from members of the YP/OE community that updates were safe to do.

Some of the discussion happened in this thread:
https://lore.kernel.org/openembedded-core/49909f0e-9fc9-cb74-66f4-e6feec95f8d0@windriver.com/

Do we have these meta-rust updates that send to be working for people but it's not clear what evidence is available to show that no user code changes are required.

We also have the rust mixin for kirkstone.

If anyone is interested in changing this policy, please let me know or just follow-up by email on that thread or on the oe-arch list.

Thanks.

Hollingsworth-martin-bl, 2024-03-06:
It took me some time to figure out that - for most standard uses cases around rust and cargo - meta-rust is no longer required since ~Kirkstone (thanks to your communities great work bringing it upstream into oe-core!).

How about a short note in the README on the default branch that this layer is no longer required for standard rust based software packages?