meta-rust/meta-rust #443: "Hello World" build fails with a full TTY of errors

voloder, 2024-06-03:
## Version(s) of meta-rust
git

## Version(s) of poky and/or oe-core
Kirkstone

## Expected result
Hello World compiles

## Actual result
I get a full TTY of errors, here's the tail:
~~~
| error[E0412]: cannot find type `Error` in this scope
|   --> src/tools/cargo/src/cargo/lib.rs:69:56
|    |
| 69 | pub fn display_warning_with_error(warning: &str, err: &Error, shell: &mut Shell) {
|    |                                                        ^^^^^ not found in this scope
|    |
| help: consider importing one of these items
|    |
| 11 | use core::fmt::Error;
|    |
| 11 | use std::error::Error;
|    |
| 11 | use std::fmt::Error;
|    |
| 11 | use std::io::Error;
|    |
|
| error[E0412]: cannot find type `Error` in this scope
|   --> src/tools/cargo/src/cargo/lib.rs:75:25
|    |
| 75 | fn _display_error(err: &Error, shell: &mut Shell, as_err: bool) -> bool {
|    |                         ^^^^^ not found in this scope
|    |
| help: consider importing one of these items
|    |
| 11 | use core::fmt::Error;
|    |
| 11 | use std::error::Error;
|    |
| 11 | use std::fmt::Error;
|    |
| 11 | use std::io::Error;
|    |
|
| warning: unused import: `FromStr`
|   --> src/tools/cargo/src/cargo/core/compiler/build_context/target_info.rs:15:22
|    |
| 15 | use std::str::{self, FromStr};
|    |                      ^^^^^^^
|    |
|    = note: `#[warn(unused_imports)]` on by default
|
| warning: unused import: `crate::core::resolver::errors::describe_path`
|  --> src/tools/cargo/src/cargo/core/compiler/links.rs:2:5
|   |
| 2 | use crate::core::resolver::errors::describe_path;
|   |     ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
|
| warning: unused import: `iter_join`
|    --> src/tools/cargo/src/cargo/core/features.rs:104:35
|     |
| 104 | use crate::util::{indented_lines, iter_join};
|     |                                   ^^^^^^^^^
|
| warning: unused import: `std::collections::HashSet`
|  --> src/tools/cargo/src/cargo/core/package_id.rs:1:5
|   |
| 1 | use std::collections::HashSet;
|   |     ^^^^^^^^^^^^^^^^^^^^^^^^^
|
| warning: unused import: `std::sync::Mutex`
|  --> src/tools/cargo/src/cargo/core/package_id.rs:7:5
|   |
| 7 | use std::sync::Mutex;
|   |     ^^^^^^^^^^^^^^^^
|
| warning: unused import: `std::sync::Mutex`
|   --> src/tools/cargo/src/cargo/core/source/source_id.rs:14:5
|    |
| 14 | use std::sync::Mutex;
|    |     ^^^^^^^^^^^^^^^^
|
| warning: unused import: `VersionReqExt`
|   --> src/tools/cargo/src/cargo/ops/cargo_install.rs:11:56
|    |
| 11 | use crate::util::{Config, Filesystem, Rustc, ToSemver, VersionReqExt};
|    |                                                        ^^^^^^^^^^^^^
|
| warning: unused import: `from_utf8`
|   --> src/tools/cargo/src/cargo/ops/cargo_new.rs:14:16
|    |
| 14 | use std::str::{from_utf8, FromStr};
|    |                ^^^^^^^^^
|
| warning: unused import: `std::collections::HashSet`
|  --> src/tools/cargo/src/cargo/util/interning.rs:4:5
|   |
| 4 | use std::collections::HashSet;
|   |     ^^^^^^^^^^^^^^^^^^^^^^^^^
|
| warning: unused import: `std::sync::Mutex`
|   --> src/tools/cargo/src/cargo/util/interning.rs:12:5
|    |
| 12 | use std::sync::Mutex;
|    |     ^^^^^^^^^^^^^^^^
|
| warning: unused import: `VersionReqExt`
|   --> src/tools/cargo/src/cargo/util/toml/mod.rs:29:79
|    |
| 29 |     self, config::ConfigRelativePath, validate_package_name, Config, IntoUrl, VersionReqExt,
|    |                                                                               ^^^^^^^^^^^^^
|
| Some errors have detailed explanations: E0405, E0412, E0422, E0425, E0432, E0433.
| For more information about an error, try `rustc --explain E0405`.
| warning: `cargo` (lib) generated 13 warnings (2 duplicates)
| error: could not compile `cargo` due to 2268 previous errors; 13 warnings emitted
~~~

## Steps to reproduce

- Clone meta-rust to `sources`
- Add meta-rust to `bblayers.conf`
- Add `IMAGE_INSTALL:append = "rust-hello-world"` to `local.conf`
- `bitbake core-image-base`

srwalter, 2024-06-03:
This is passing in our CI, any idea what's different about your configuration versus what our CI builds?

voloder, 2024-06-03:
I am using DART-6UL BSP from Variscite if it helps, really not sure what could be the issue though... I have had to add `CARGO_BUILD_FLAGS:append = " -Z namespaced-features "` to local.conf because a lot of cargo deps were failing to build without it.

cedrickkukela-cd, 2024-10-29:
Hey i just want to second this ticket. I had the same issue and Authors suggestion seems to helpl.