meta-rust/meta-rust #444: Build fails for STM32/ARMv7 target

tronje, 2024-06-12:
## Version(s) of meta-rust

Current master, 81d0fbfc023e987068ea1c035e33d6f0ad175cd1

## Version(s) of poky and/or oe-core

kirkstone-4.0.17, commit 6d1a878bbf24c66f7186b270f823fcdf82e35383

## Expected result

The `rust-hello-world` recipe builds successfully

## Actual result

Trying to build `rust-hello-world` does not actually reach trying to build that recipe. Instead, building `libstd-rs` fails.
Abbreviated error output:

```
|      Running `rustc --crate-name libc --edition=2015 /workdir/build/tmp/work/cortexa7t2hf-neon-vfpv4-poky-linux-gnueabi/libstd-rs/1.78.0-r0/rustc-1.78.0-src/vendor/libc/src/lib.rs --error-format=json --json=diagnostic-rendered-ansi,artifacts,future-incompat --crate-type lib --emit=dep-info,metadata,link -C opt-level=3 -C embed-bitcode=no --cfg 'feature="align"' --cfg 'feature="rustc-dep-of-std"' --cfg 'feature="rustc-std-workspace-core"' -C metadata=ffe5355565292cd6 -C extra-filename=-ffe5355565292cd6 --out-dir /workdir/build/tmp/work/cortexa7t2hf-neon-vfpv4-poky-linux-gnueabi/libstd-rs/1.78.0-r0/build/arm-poky-linux-gnueabi/release/deps --target arm-poky-linux-gnueabi -C linker=/workdir/build/tmp/work/cortexa7t2hf-neon-vfpv4-poky-linux-gnueabi/libstd-rs/1.78.0-r0/wrapper/target-rust-ccld -L dependency=/workdir/build/tmp/work/cortexa7t2hf-neon-vfpv4-poky-linux-gnueabi/libstd-rs/1.78.0-r0/build/arm-poky-linux-gnueabi/release/deps -L dependency=/workdir/build/tmp/work/cortexa7t2hf-neon-vfpv4-poky-linux-gnueabi/libstd-rs/1.78.0-r0/build/release/deps --extern rustc_std_workspace_core=/workdir/build/tmp/work/cortexa7t2hf-neon-vfpv4-poky-linux-gnueabi/libstd-rs/1.78.0-r0/build/arm-poky-linux-gnueabi/release/deps/librustc_std_workspace_core-7df3a077613c532b.rmeta --cap-lints allow -L /workdir/build/tmp/work/cortexa7t2hf-neon-vfpv4-poky-linux-gnueabi/libstd-rs/1.78.0-r0/recipe-sysroot/usr/lib/rust --remap-path-prefix=/workdir/build/tmp/work/cortexa7t2hf-neon-vfpv4-poky-linux-gnueabi/libstd-rs/1.78.0-r0=/usr/src/debug/libstd-rs/1.78.0-r0 -Cembed-bitcode=yes -L /workdir/build/tmp/work/cortexa7t2hf-neon-vfpv4-poky-linux-gnueabi/libstd-rs/1.78.0-r0/recipe-sysroot/usr/lib -C link-arg=-Wl,-soname,libstd.so --cfg freebsd11 --cfg libc_priv_mod_use --cfg libc_union --cfg libc_const_size_of --cfg libc_align --cfg libc_int128 --cfg libc_core_cvoid --cfg libc_packedN --cfg libc_cfg_target_vendor --cfg libc_non_exhaustive --cfg libc_long_array --cfg libc_ptr_addr_of --cfg libc_underscore_const_names --cfg libc_thread_local --cfg libc_const_extern_fn`
| error: data-layout for target `arm-poky-linux-gnueabi`, `e-m:e-p:32:32-i64:64-v128:64:128-a:0:32-n32-S64`, differs from LLVM target's `armv7-unknown-linux-gnueabihf` default layout, `e-m:e-p:32:32-Fi8-i64:64-v128:64:128-a:0:32-n32-S64`
```

Basically the data-layout is different between Rust's and LLVM's, I guess...?

## Steps to reproduce

I'm targeting a machine that uses an STM32MP1. Board support, etc. is from here: https://github.com/karo-electronics/meta-karo
I guess try to build `rust-hello-world` with current `meta-rust` for a machine that uses the armv7-eabi data layout...

## Additional information

Building the `rust-hello-world` recipe works fine with my Poky's version of Rust, and that, curiously, has the same `DATA_LAYOUT` set as `meta-rust`. I thought maybe the difference in version of `rust-llvm` might be the issue, but as far as I can tell, both versions of `rust-llvm` use the same data layout, too. I tried simply pasting the LLVM data layout from the error message into `rust-common.inc`, but to no avail. I did a `bitbake -c cleanall rust-native rust-llvm-native`, but that did not help, either..

I did notice b93853cb12e5e9cacb76c439d36eb6b96415026d which fixes basically the same issue but for x86.

Any help is appreciated, thanks!

compenguy, 2024-06-14:
I have also started running into the error that @tronje reports in their update, but on an imx6 build of rust-native 1.73.  I can provide additional details if helpful - I just wanted to help clarify that this isn't limited to a niche build setting or target.

karlri, 2024-06-20:
I can confirm both of the above issues in imx8. 1.78 builds and works if you do a manual edit of the file usr/lib/rustlib/arm-poky-linux-gnueabi.json and rebuild libstd-rs. Changes to DATA_LAYOUT[armv7-eabi] = in rust-common.inc do not seem to take. libstd-rs builds on 1.78 after manually editing usr/lib/rustlib/arm-poky-linux-gnueabi.json in accordance with the data layout mismatch error though.

tronje, 2024-06-20:
> I can confirm both of the above issues in imx8. 1.78 builds and works if you do a manual edit of the file usr/lib/rustlib/arm-poky-linux-gnueabi.json and rebuild libstd-rs. Changes to [...]

That's interesting. There are two of these json files in my build directory, do you know which one is the one I need to change? Or simply both of them?

```console
$ fd arm-poky-linux-gnueabi.json build
build/tmp/work/x86_64-linux/rust-cross-cortexa7t2hf-neon-vfpv4-glibc/1.78.0-r0/targets/arm-poky-linux-gnueabi.json
build/tmp/work/cortexa7t2hf-neon-vfpv4-poky-linux-gnueabi/libstd-rs/1.78.0-r0/recipe-sysroot-native/usr/lib/rustlib/arm-poky-linux-gnueabi.json
```

karlri, 2024-06-20:
Both probably would not hurt but the latter would be my guess. It's just a problem when targeting arm. The first one would probably be a native build of the rust compiler that should run on x86_64 and produce code trgeting arm. The latter would be a build targeting arm so the file will be used in that case and cause issues for libstd-rs to build.

karlri, 2024-06-20:
Now how and why changes to DATA_LAYOUT in rust-common.inc do not take and propagate to the file arm-poky-linux-gnueabi.json I do not understand. It seems to be generated by rust_gen_target in rust-common.inc which takes the string from DATA_LAYOUT.

karlri, 2024-06-20:
Modifying the json file like this is only a temporary fix and if you delete the tmp directory and rebuild the error is back. I did lik this:
1. modify DATA_LAYOUT for armv7 in rust-common.inc
2. bitbake libstd-rs, wait for it to fail
3. edit "data-layout" the arm-poky-linux-gnueabi.json file (which I think should have been generated in accordance with DATA_LAYOUT in rust-common.inc but for some reason that is not the case)
4. bitbake libstd-rs again, success

tronje, 2024-06-20:
Yeah, I figured any change to some generated file would only be a temporary fix, but at least it would build for the time being. I'll give it a shot later today or tomorrow, probably. Thanks a lot @karlri :slightly_smiling_face:

tronje, 2024-06-21:
I added some simple logging to `rust_gen_target` in `rust-common.inc`, and that seemed to trigger a rebuild of the Rust toolchain. Now everything built fine (with the fixed data layout). Maybe the reason that changing the data layout in `rust-common.inc` had no effect was some Yocto magic. I don't quite understand why my `-c cleanall` invocations did not have that effect, though.

tronje, 2024-06-21:
I'm guessing this is the fix:

```diff
diff --git a/recipes-devtools/rust/rust-common.inc b/recipes-devtools/rust/rust-common.inc
index 47a6d68..5ec9763 100644
--- a/recipes-devtools/rust/rust-common.inc
+++ b/recipes-devtools/rust/rust-common.inc
@@ -378,4 +378,11 @@ python do_rust_gen_targets () {

 addtask rust_gen_targets after do_patch before do_compile
 do_rust_gen_targets[dirs] += "${WORKDIR}/targets"
-
+do_rust_gen_targets[vardeps] += "\
+    LLVM_TARGET \
+    DATA_LAYOUT \
+    MAX_ATOMIC_WIDTH \
+    TARGET_POINTER_WIDTH \
+    TARGET_C_INT_WIDTH \
+    TARGET_ENDIAN \
+"
```

Testing this now...

karlri, 2024-06-21:
That totally makes sense, great analysis. Could it be that the same change should be made to arm-eabi aswell?

tronje, 2024-06-21:
> That totally makes sense, great analysis.

Thanks :slightly_smiling_face: 

> Could it be that the same change should be made to arm-eabi aswell?

`rustup run nightly rustc -Z unstable-options --print target-spec-json --target arm-unknown-linux-gnueabihf` says:

```
  "data-layout": "e-m:e-p:32:32-Fi8-i64:64-v128:64:128-a:0:32-n32-S64",
```

...so I guess so. Will add it to #446.

PR #446 (2024-06-21): rust-common: Update data layout for armv7-unknown-linux-gnueabihf
Akin to b93853c, update the data layout to fix the following build error:

error: data-layout for target `arm-poky-linux-gnueabi`, `e-m:e-p:32:32-i64:64-v128:64:128-a:0:32-n32-S64`, differs from LLVM target's `armv7-unknown-linux-gnueabihf` default layout, `e-m:e-p:32:32-Fi8-i64:64-v128:64:128-a:0:32-n32-S64`

Fixes #444.

~Note that even a rebuild of the Rust toolchain doesn't seem to make this change take effect in an existing Yocto setup. My assumption is that the rust_gen_target function is cached somehow. To fix a build tha [... cut]
commit 18ae2be1 (2024-06-21): rust-common: Update data layouts for 32-bit arm targets

Akin to b93853c, update the data layout to fix the following build
error:

error: data-layout for target `arm-poky-linux-gnueabi`, `e-m:e-p:32:32-i64:64-v128:64:128-a:0:32-n32-S64`, differs from LLVM target's `armv7-unknown-linux-gnueabihf` default layout, `e-m:e-p:32:32-Fi8-i64:64-v128:64:128-a:0:32-n32-S64`

Fixes #444.
closed 2024-06-21 by srwalter